espressif/esp-idf #13015: when start nimble broadcast data,os_msys_get_pkthdr  function return NULL,trigger assert check

luacerbb, 2024-01-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

idf 5.1.2

### Espressif SoC revision.

esp32-s3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

custom board

### Power Supply used.

USB

### What is the expected behavior?

expected ble can broadcast success

### What is the actual behavior?

This is an occasional problem,when I start ble broadcast, the os_msys_get_pkthdr function return null.

### Steps to reproduce.

1、start  ble broadcast
2、ble connected
3、ble disconnected,start ble broadcast,os_msys_get_pkthdr return NULL

### Debug Logs.

```plain
assert failed: bd_ble_adv_start bd_ble_drv.c:1028 (data)

Backtrace: 0x40375e36:0x3fce4a70 0x40381ead:0x3fce4a90 0x403890ba:0x3fce4ab0 0x4204d522:0x3fce4bd0 0x42014935:0x3fce4c20 0x420149d1:0x3fce4c80 0x4204c68e:0x3fce4db0 0x4204cad9:0x3fce4dd0 0x42053c09:0x3fce4e50 0x42055f5a:0x3fce4e70 0x42055faa:0x3fce4ee0 0x4204ff0b:0x3fce4f40 0x42050561:0x3fce4f60 0x4204f841:0x3fce4f80 0x4037f767:0x3fce4fa0 0x40378896:0x3fce4fc0 0x4204c792:0x3fce4fe0 0x40384acd:0x3fce5000
0x40375e36: panic_abort at /home/workspace/esp-idf-v5.1.2/components/esp_system/panic.c:452

0x40381ead: esp_system_abort at /home/workspace/esp-idf-v5.1.2/components/esp_system/port/esp_system_chip.c:84

0x403890ba: __assert_func at /home/workspace/esp-idf-v5.1.2/components/newlib/assert.c:81

0x4204d522: bd_ble_adv_start at /home/workspace/wifi_gate_malloc_monitor/main/HAL/bd_ble_drv.c:1028 (discriminator 1)

0x42014935: _bd_ble_adv_start at /home/workspace/wifi_gate_malloc_monitor/main/bd_ble.c:2192

0x420149d1: _bd_ble_slave_gap_callback at /home/workspace/wifi_gate_malloc_monitor/main/bd_ble.c:2116

0x4204c68e: _ble_slave_snd_evt_notify at /home/workspace/wifi_gate_malloc_monitor/main/HAL/bd_ble_drv.c:719

0x4204cad9: _ble_slave_gap_event at /home/workspace/wifi_gate_malloc_monitor/main/HAL/bd_ble_drv.c:777

0x42053c09: ble_gap_call_event_cb at /home/workspace/esp-idf-v5.1.2/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:773

0x42055f5a: ble_gap_conn_broken at /home/workspace/esp-idf-v5.1.2/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:1344

0x42055faa: ble_gap_rx_disconn_complete at /home/workspace/esp-idf-v5.1.2/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:1372 (discriminator 4)

0x4204ff0b: ble_hs_hci_evt_disconn_complete at /home/workspace/esp-idf-v5.1.2/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:294

0x42050561: ble_hs_hci_evt_process at /home/workspace/esp-idf-v5.1.2/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:1051

0x4204f841: ble_hs_event_rx_hci_ev at /home/workspace/esp-idf-v5.1.2/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:526

0x4037f767: npl_freertos_event_run at /home/workspace/esp-idf-v5.1.2/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:451

0x40378896: ble_npl_event_run at /home/workspace/esp-idf-v5.1.2/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:185
 (inlined by) nimble_port_run at /home/workspace/esp-idf-v5.1.2/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:248

0x4204c792: _ble_host_task at /home/workspace/wifi_gate_malloc_monitor/main/HAL/bd_ble_drv.c:857

0x40384acd: vPortTaskWrapper at /home/workspace/esp-idf-v5.1.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

### More Information.

	if(adv_para->adv_data.length > 0) {
	  
	  /* get mbuf for adv data */
	  data = os_msys_get_pkthdr(adv_para->adv_data.length, 0);
	  assert(data);

	  /* fill mbuf with adv data */
	  rc = os_mbuf_append(data, adv_para->adv_data.data_buf, adv_para->adv_data.length);
	  assert(rc == 0);

	  rc = ble_gap_ext_adv_set_data(0, data);
	  assert (rc == 0);

	  /* get mbuf for scan rsp data */
	  data = os_msys_get_pkthdr(31, 0);
	  assert(data);
	}

darshan7patel, 2024-01-24:
Hii @luacerbb , 
Would you kindly share the application's SDK config?

luacerbb, 2024-01-26:
[sdconfig.txt](https://github.com/espressif/esp-idf/files/14060653/sdconfig.txt)

luacerbb, 2024-01-28:
@darshan7patel sdkcofijg has uploaded, please assist analysis,thanks

darshan7patel, 2024-01-29:
Sure, let me check this

darshan7patel, 2024-01-29:
Hii @luacerbb ,

Used the same sdkconfig you gave along with bleprph ( extended advertise ) example and do not observe any issue (Kindly check attached log)
So , it may be tied up with environment , or any other setting. 
 
1.  Can you please let me know if there are any modifications done ?  
2. If yes, can you  share your application ? 
3. If application cannot be shared, then can you compare it with the example in IDF that is similar , so as to use the same at our end 
4. Do you observe same issue if you use IDF's exmaple ?

[app_log.txt](https://github.com/espressif/esp-idf/files/14082763/app_log.txt)

darshan7patel, 2024-02-09:
Hii @luacerbb ,
Any Update ?

darshan7patel, 2024-03-29:
No more updates found. If you have any further problems or issues. Feel free to reopen or create a new issue.

closed 2024-03-29 by rahult-github