espressif/esp-idf #13016: TWAI RX delay after installing driver

theodem, 2024-01-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

After installing TWAI driver and creating a simple Freertos task that simply does `twai_receive(&twai_message, portMAX_DELAY)` the first twai_messages are received 100ms after driver install occurred (connected on an already loaded bus).

`(12150) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0`
`(12150) gpio: GPIO[10]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0`
`(12150) CAN: Starting RX task`
`(12250) CAN: RX id: 0x25`

So if I send a CAN frame to and ECU and get a CAN response before that 100ms delay, I never receive that response CAN frame.

After lot of trial an error i found that I don't have this behavior if I disable all console output in menuconfig (UART and USB-Serial).

For now the workaround i am using is a taskDelay of 250ms just after installing the driver before getting the success install feedback to my application (communicating to ESP32-S3 using BLE).

wanckl, 2024-01-22:
@theodem ,,How about start your esp32 earlier?    

and what this payload time for your ECU?    
If one day esp32 can receive msg within 50ms, will you complain about you can't receive any response before that 50ms delay?

theodem, 2024-01-22:
@wanckl I don't understand your comment.

This issue was not created to "complain" but to understand if the fact that TWAI driver receive message 100ms after starting the driver is a normal behavior or a bug.

In case it is a normal behavior, that is not a concern as I found a  workaround. But if this is not the expected behavior something might not work as expected.

And that might be a time related bug to me as it doesn't append when all console output are disabled.

wanckl, 2024-01-23:
@theodem Sorry for my rudness, ,,    

But for my side, I think this 100ms is reasonable as follow reasons:    
1. Using console need initialize many components like newlib, uart and even vfs
2. How busy the `already loaded bus` be, when is the first frame esp32 should receive
3. In TWAI driver, there need some memory copy, it also need time    

Any way, it shouldn't a big issue I think, feel free to workaround,,,

closed 2025-05-19 by Alvin1Zhang