espressif/esp-idf #13018: FAILED: gptimer_etm_capture.elf

AirNeon, 2024-01-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

想要例程gptimer_capture_hc_sr04构建成功

### What is the actual behavior?

例程gptimer_capture_hc_sr04构建失败

### Steps to reproduce.

1.Set Espressif device target：选择本工程文件，选择esp32s3芯片
2.Select port to use：选择COM4端口
3.Build your project

### Build or installation Logs.

```plain
[1/6] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D D:\MusicLab\Engine_test\ArduinoCode\esp\gptimer_capture_hc_sr04\build\bootloader\esp-idf\esptool_py && C:\Users\19724\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe C:/Users/19724/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 D:/MusicLab/Engine_test/ArduinoCode/esp/gptimer_capture_hc_sr04/build/bootloader/bootloader.bin"
Bootloader binary size 0x51c0 bytes. 0x2e40 bytes (36%) free.
[2/4] Linking CXX executable gptimer_etm_capture.elf
FAILED: gptimer_etm_capture.elf 
cmd.exe /C "cd . && C:\Users\19724\.espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-g++.exe -mlongcalls -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32S3=0 -Wl,--Map=D:/MusicLab/Engine_test/ArduinoCode/esp/gptimer_capture_hc_sr04/build/gptimer_etm_capture.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32s3.peripherals.ld -T esp32s3.rom.ld -T esp32s3.rom.api.ld -T esp32s3.rom.libgcc.ld -T esp32s3.rom.newlib.ld -T esp32s3.rom.version.ld -T memory.ld -T sections.ld @CMakeFiles\gptimer_etm_capture.elf.rsp -o gptimer_etm_capture.elf  && cd ."
c:/users/19724/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/main/libmain.a(gptimer_capture_hc_sr04.c.obj):(.literal.app_main+0x80): undefined reference to `gpio_new_etm_event'
c:/users/19724/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/main/libmain.a(gptimer_capture_hc_sr04.c.obj):(.literal.app_main+0x84): undefined reference to `gpio_etm_event_bind_gpio'
c:/users/19724/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/main/libmain.a(gptimer_capture_hc_sr04.c.obj):(.literal.app_main+0x88): undefined reference to `gptimer_new_etm_task'
c:/users/19724/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/main/libmain.a(gptimer_capture_hc_sr04.c.obj):(.literal.app_main+0x8c): undefined reference to `esp_etm_new_channel'
c:/users/19724/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/main/libmain.a(gptimer_capture_hc_sr04.c.obj):(.literal.app_main+0x90): undefined reference to `esp_etm_channel_connect'
c:/users/19724/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/main/libmain.a(gptimer_capture_hc_sr04.c.obj):(.literal.app_main+0x98): undefined reference to `esp_etm_channel_enable'
c:/users/19724/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/main/libmain.a(gptimer_capture_hc_sr04.c.obj):(.literal.app_main+0xa0): undefined reference to `esp_etm_dump'
c:/users/19724/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/main/libmain.a(gptimer_capture_hc_sr04.c.obj): in function `app_main':
D:/MusicLab/Engine_test/ArduinoCode/esp/gptimer_capture_hc_sr04/main/gptimer_capture_hc_sr04.c:89: undefined reference to `gpio_new_etm_event'
c:/users/19724/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: D:/MusicLab/Engine_test/ArduinoCode/esp/gptimer_capture_hc_sr04/main/gptimer_capture_hc_sr04.c:93: undefined reference to `gpio_etm_event_bind_gpio'
c:/users/19724/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: D:/MusicLab/Engine_test/ArduinoCode/esp/gptimer_capture_hc_sr04/main/gptimer_capture_hc_sr04.c:106: undefined reference to `gptimer_new_etm_task'
c:/users/19724/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: D:/MusicLab/Engine_test/ArduinoCode/esp/gptimer_capture_hc_sr04/main/gptimer_capture_hc_sr04.c:112: undefined reference to `esp_etm_new_channel'
c:/users/19724/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: D:/MusicLab/Engine_test/ArduinoCode/esp/gptimer_capture_hc_sr04/main/gptimer_capture_hc_sr04.c:115: undefined reference to `esp_etm_channel_connect'
c:/users/19724/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: D:/MusicLab/Engine_test/ArduinoCode/esp/gptimer_capture_hc_sr04/main/gptimer_capture_hc_sr04.c:128: undefined reference to `esp_etm_channel_enable'
c:/users/19724/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: D:/MusicLab/Engine_test/ArduinoCode/esp/gptimer_capture_hc_sr04/main/gptimer_capture_hc_sr04.c:130: undefined reference to `esp_etm_dump' 
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

 *  The terminal process "C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja " terminated with exit code: 1.
```

### More Information.

历程没有做任何改动

suda-morris, 2024-01-20:
The example's README file shows it doesn't support esp32s3.

closed 2024-01-21 by suda-morris

GravityandSupport, 2024-07-14:
Thanks for sharing