espressif/esp-idf #13019: Hang and Reset with PSRAM ( may not correctly ) configured

KurehaTian, 2024-01-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.2-beta1-263-ge49823f10c-dirty

### Espressif SoC revision.

ESP32-S3-WROOM-1,N16R8

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-WROOM-1 Module | Custom Board

### Power Supply used.

USB

### What is the expected behavior?

The app_main can run and allocatable.
Actually I am trying ESP32-Camera who make use of PSRAM.

### What is the actual behavior?

As the log, run to "cpu_start: App cpu up.."(v5.1)/"esp_psram: Speed: 80MHz"(v5.2),Then hangs for about eight secs, then reset for watchdog.

### Steps to reproduce.

In example get-started/hello_world:
1.set-target to esp32s3
2.menuconfig:
    Serial Flasher Config->Flash SPI Mode = QIO
                                         Flash SPI Speed = 80MHz
                                         Flash Size = 16MB
    Component Config-> ESP PSRAM -> Support for external, SPI-connected RAM=enabled
                                         SPI RAM config->Mode of SPI RAM chip in use = Octal Mode PSRAM
                                                                      Set RAM Clock Speed = 80MHz Clock Speed
                                                                      SPI RAM access method = using heap_caps_malloc()...
  Save.
3.build, flash and monitor.

### Debug Logs.

```plain
### in IDF v5.1，the log is :
I (0) cpu_start: AESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x1870
load:0x403c9700,len:0x4
load:0x403c9704,len:0xda4
load:0x403cc700,len:0x3130
entry 0x403c9934
I (27) boot: ESP-IDF v5.1-dirty 2nd stage bootloader
I (28) boot: compile time Jan 20 2024 15:34:15
I (28) boot: Multicore bootloader
I (31) boot: chip revision: v0.1
I (35) qio_mode: Enabling default flash chip QIO
I (40) boot.esp32s3: Boot SPI Speed : 80MHz
I (45) boot.esp32s3: SPI Mode       : QIO
I (50) boot.esp32s3: SPI Flash Size : 16MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0a180h ( 41344) map
I (112) esp_image: segment 1: paddr=0001a1a8 vaddr=3fc92500 size=02c04h ( 11268) load
I (116) esp_image: segment 2: paddr=0001cdb4 vaddr=40374000 size=03264h ( 12900) load
I (125) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=18114h ( 98580) map
I (146) esp_image: segment 4: paddr=0003813c vaddr=40377264 size=0b268h ( 45672) load
I (161) boot: Loaded app from partition at offset 0x10000
I (161) boot: Disabling RNG early entropy source...
I (173) cpu_start: Multicore app
I (173) octal_psram: vendor id    : 0x0d (AP)
I (173) octal_psram: dev id       : 0x02 (generation 3)
I (176) octal_psram: density      : 0x03 (64 Mbit)
I (182) octal_psram: good-die     : 0x01 (Pass)
I (187) octal_psram: Latency      : 0x01 (Fixed)
I (192) octal_psram: VCC          : 0x01 (3V)
I (197) octal_psram: SRF          : 0x01 (Fast Refresh)
I (203) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (209) octal_psram: BurstLen     : 0x01 (32 Byte)
I (215) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (221) octal_psram: DriveStrength: 0x00 (1/1)
W (227) MSPI Timing: tuning fail, best point is fallen back to index 5
I (233) esp_psram: Found 8MB PSRAM device
I (238) esp_psram: Speed: 80MHz
I (242) cpu_start: Pro cpu up.
I (245) cpu_start: Starting app cpu, entry point is 0x40375378
0x40375378: call_start_cpu1 at /home/ubuntu/esp32s3/esp-idf/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.

### in IDF v5.2, the log is :

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x1918
load:0x403c9700,len:0x4
load:0x403c9704,len:0xe5c
load:0x403cc700,len:0x301c
entry 0x403c993c
I (27) boot: ESP-IDF v5.2-beta1-263-ge49823f10c-dirt 2nd stage bootloader
I (28) boot: compile time Jan 20 2024 16:20:04
I (29) boot: Multicore bootloader
I (33) boot: chip revision: v0.1
I (37) qio_mode: Enabling default flash chip QIO
I (42) boot.esp32s3: Boot SPI Speed : 80MHz
I (47) boot.esp32s3: SPI Mode       : QIO
I (51) boot.esp32s3: SPI Flash Size : 16MB
I (56) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (80) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  2 factory          factory app      00 00 00010000 00177000
I (95) boot: End of partition table
I (99) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=11130h ( 69936) map
I (118) esp_image: segment 1: paddr=00021158 vaddr=3fc94f00 size=03e3ch ( 15932) load
I (121) esp_image: segment 2: paddr=00024f9c vaddr=40374000 size=0b07ch ( 45180) load
I (133) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=25750h (153424) map
I (157) esp_image: segment 4: paddr=00055778 vaddr=4037f07c size=05d90h ( 23952) load
I (169) boot: Loaded app from partition at offset 0x10000
I (169) boot: Disabling RNG early entropy source...
I (180) cpu_start: Multicore app
I (181) octal_psram: vendor id    : 0x0d (AP)
I (181) octal_psram: dev id       : 0x02 (generation 3)
I (184) octal_psram: density      : 0x03 (64 Mbit)
I (189) octal_psram: good-die     : 0x01 (Pass)
I (195) octal_psram: Latency      : 0x01 (Fixed)
I (200) octal_psram: VCC          : 0x01 (3V)
I (205) octal_psram: SRF          : 0x01 (Fast Refresh)
I (211) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (217) octal_psram: BurstLen     : 0x01 (32 Byte)
I (222) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (228) octal_psram: DriveStrength: 0x00 (1/1)
W (234) MSPI Timing: tuning fail, best point is fallen back to index 5
I (241) MSPI Timing: PSRAM timing tuning index: 5
I (246) esp_psram: Found 8MB PSRAM device
I (251) esp_psram: Speed: 80MHz
```

### More Information.

I wonder if any misunderstanding to the docs:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/flash_psram_config.html
or any hardware fault.

KurehaTian, 2024-01-20:
IO35、35、37 are not connected for other use.

AxelLin, 2024-02-20:
Test on v5.2-166-ge0944287f3, I hit the same issue on ESP32-S3-WROOM-1,N4R8.
I'm also testing esp32-camera.

ps.
If I set SPIRAM_USE_MEMMAP=y, it won't hang.
However it will reset (rst:0x8 (TG1WDT_SYS_RST)) if I set SPIRAM_USE_CAPS_ALLOC=y or SPIRAM_USE_MALLOC=y.

KurehaTian, 2025-02-22:
It seems that the promblem is from hardware ( PSRAM ) . This problem does not happen on another ESP32-S3-EYE board.

closed 2025-02-22 by KurehaTian

electronicsguy, 2025-06-20:
> It seems that the promblem is from hardware ( PSRAM ) . This problem does not happen on another ESP32-S3-EYE board.

what is the problem, exactly?

BitsForPeople, 2025-06-20:
> These GPIO should be available for use as per the spec: [schematic](https://files.waveshare.com/wiki/ESP32-S3-AMOLED-1.91/ESP32-S3-AMOLED-1.91.pdf)
...
> I (248) octal_psram: vendor id    : 0x0d (AP)
> I (249) octal_psram: dev id       : 0x02 (generation 3)
> I (249) octal_psram: density      : 0x03 (64 Mbit)
...

The schematic doesn't say the pins are "available for use", and the ESP32-S3 datasheet says that those pins are used for communication with the PSRAM in Octal SPI (OPI) mode. Don't use those pins when you have PSRAM enabled in OPI mode.