espressif/esp-idf #13024: Unnecessary vTaskDelay() on touch_pad_config

DarkZeros, 2024-01-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am developing an ultra low power device (energy harvest) with ESP32, that uses a touch sensor as wake up from deep sleep. In this project the power usage is critical, and saving any time in processing is essential.

I noticed that the touch config takes 28ms to run per input (using 1mAs), that is around 30% of my total energy usage having 4 inputs.
Since in my project the touch is setup is done before entering deep sleep, there is no need to add a vTaskDelay() after the setup, since I am not going to read from it immediately. I tested removing it and see no difference in the touch response, just a reduction in power usage (108ms less active time)

I think there should be a way to disable the delay in FSM mode on touch_pad_config(), or it should be the job of the application to handle the delay if reading manually.

DarkZeros, 2024-01-21:
Note: This is the same for v4.4, v5.1 and master branches as far as I can see.

DarkZeros, 2024-01-21:
The issue can be reproduced like this:
```
  touch_pad_init();
  esp_sleep_enable_touchpad_wakeup();
  touch_pad_set_fsm_mode(TOUCH_FSM_MODE_TIMER);   // returns ESP_OK
  touch_pad_config((touch_pad_t)0, THRESHOLD);
```

But like this is bypassed, keeping the FSM off until the pins are set, then set the mask manually.
```
  touch_pad_init();
  esp_sleep_enable_touchpad_wakeup();
  touch_pad_config((touch_pad_t)0, THRESHOLD);
  touch_pad_set_fsm_mode(TOUCH_FSM_MODE_TIMER);   // returns ESP_OK
  uint16_t mask = (1 << 0);
  touch_pad_set_group_mask(mask, mask, mask);`
```

L-KAYA, 2024-01-23:
Hi @DarkZeros , thanks for reporting this issue! 

It's indeed a shortcoming of the current driver, i.e., no oneshot scanning supported and no stage judgement in driver. 

As you know, a touch pad needs an initial measurement to initialize the original data.  The bad logic of the current driver is that wait for one round scanning every time when configure the pad. 

The good news is that, the driver-NG for the touch sensor is under developing, it has stage judgement and supports oneshot scanning, which mean the initial scanning can be triggered manually after all pad configured, instead of waiting for one round scanning every time configuring the pad.

Therefore, the current driver is going to be a legacy driver soon, we don't plan to do a big refactor to solve that bad logic in the legacy driver, sorry for the inconvenience and please wait for the touch sensor driver-NG.

DarkZeros, 2024-01-24:
Thanks, In understand the logic behind the decision.

I will use the workaround for the time being (use manual reading, configure, and finally enable FSM), it works as intended and doesn't need any driver changes.

If anyone find this same problem and issue, I would recommend to use the solution above.

closed 2024-01-24 by DarkZeros