espressif/esp-idf #13026: Error with "Store BLE Mesh Configuration Persistently" Option in ESP32S3 Vendor Server Example

lessely, 2024-01-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am currently working with the ESP32S3 and using the vendor_server example as a starting point for my project. When I enable the "Store BLE Mesh configuration persistently" option in the menuconfig, I encounter the following error:
idf version:v4.4.6-323-g979a56386d

```
I (516) phy_init: phy_version 620,ec7ec30,Sep  5 2023,13:49:13
E (676) BLE_MESH: Invalid SIG model index 1
E (676) BLE_MESH: sig model not found, elem_idx 0, model_idx 1
W (676) BLE_MESH: bt_mesh_proxy_server_prov_disable, Already
W (696) VENDOR_SERVER: example_ble_mesh_provisioning_cb
I (696) VENDOR_SERVER: ESP_BLE_MESH_NODE_PROV_COMPLETE_EVT
I (696) VENDOR_SERVER: net_idx 0x000, addr 0x000b
I (696) VENDOR_SERVER: flags 0x00, iv_index 0x00000000
W (706) VENDOR_SERVER: example_ble_mesh_provisioning_cb
I (716) VENDOR_SERVER: ESP_BLE_MESH_PROV_REGISTER_COMP_EVT, err_code 0
W (716) BLE_MESH: bt_mesh_prov_enable, Already
W (726) VENDOR_SERVER: example_ble_mesh_provisioning_cb
I (726) VENDOR_SERVER: ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT, err_code -120
I (736) VENDOR_SERVER: BLE Mesh Node initialized
```
[main.txt](https://github.com/espressif/esp-idf/files/14002017/main.txt)
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/14002018/sdkconfig.txt)

lessely, 2024-01-22:
The issue has been temporarily resolved. When replacing a completely new device that has not been programmed with BLE Mesh before, no errors are reported. It is speculated that there may be a conflict between the old configuration and the new one.

ruslanslobozhenuk, 2024-01-22:
I have a similar problem. When updating a device via OTA, the device's BLE mesh configuration is not loaded. The device becomes unprovisioned. (Old device IDF 5.2, new OTA firmware from IDF master branch)

lessely, 2024-01-24:
For older devices that burned ble mesh, running "idf.py erase-flash" before burning new code can solve this problem

closed 2024-01-24 by lessely