espressif/esp-idf #13027: #error "This header should only be included when building for ESP32"

AchimPieters, 2024-01-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

When compiling my code with ESP-IDF `v5.3-dev-1353-gb3f7e2c8a4` like this:

- idf.py set-target esp32
- idf.py menuconfig
- idf.py build

my code compiles without any errors, and works, but when I want to compile for let's say an ESP32C2:

- idf.py set-target esp32c2
- idf.py menuconfig
- idf.py build

then I get this error:

```
/opt/esp/idf/components/esp_rom/include/esp32/rom/ets_sys.h:15:2: error: #error "This header should only be included when building for ESP32"
   15 | #error "This header should only be included when building for ESP32"
      |  ^~~~~
ninja: build stopped: subcommand failed.
```

another example is, when I want to compile for let's say an ESP32S2:

- idf.py set-target esp32s2
- idf.py menuconfig
- idf.py build

then I get this error:

```
/opt/esp/idf/components/esp_rom/include/esp32/rom/ets_sys.h:15:2: error: #error "This header should only be included when building for ESP32"
   15 | #error "This header should only be included when building for ESP32"
      |  ^~~~~
[779/916] Building C object esp-idf/fatfs/CMakeFiles/__idf_fatfs.dir/src/ff.c.obj
ninja: build stopped: subcommand failed.
```

How to solve this issue?

For more information and reproduction: https://github.com/AchimPieters/esp32-homekit-demo I used the simple LED example.

igrr, 2024-01-21:
Can you please attach the complete build log (preferably as a text file)?

KonstantinKondrashov, 2024-01-24:
Hi @AchimPieters!
Could you provide the whole build log it can help to understand the issue.
Thanks.

AchimPieters, 2024-01-24:
@igrr and @KonstantinKondrashov My sincere apologies, I have found out that the cause is wolfssl

KonstantinKondrashov, 2024-01-30:
If there is no issue then I close it. Thanks.

closed 2024-01-30 by KonstantinKondrashov