espressif/esp-idf #13028: ot_rcp example not compiling

hertugweile, 2024-01-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Compile without errors on an ESP32-H2.

### What is the actual behavior?

It produces an error, when the "Config the Thread radio type" is set to "Connect via SPI" / OPENTHREAD_RADIO_SPINEL_SPI as mentioned in the documentation here - https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html - under 2.1.3.4

### Steps to reproduce.

1. Open menuconfig, either via commandline or VS Code
2. Search after "SPINEL"
3. Select "Connect via SPI"
4. Save
5. Build

### Build or installation Logs.

```plain
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[805/1083] Building CXX object esp-idf/openthread/CMakeFiles/__idf_openthread.dir/src/port/esp_openthread_radio_spinel.cpp.obj
FAILED: esp-idf/openthread/CMakeFiles/__idf_openthread.dir/src/port/esp_openthread_radio_spinel.cpp.obj
ccache C:\Espressif\tools\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\bin\riscv32-esp-elf-g++.exe -DESP_PLATFORM -DIDF_VER=\"v5.2-beta1-263-ge49823f10c\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DOPENTHREAD_BUILD_DATETIME="\" 2024-01-21 17:25:02 UTC\"" -DOPENTHREAD_CONFIG_FILE=\"openthread-core-esp32x-radio-config.h\" -DOPENTHREAD_RADIO=1 -DPACKAGE_VERSION=\"e49823f10c-af5938e38\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -IC:/Users/udenr/esp/esp-idf/examples/openthread/ot_rcp/build/config -IC:/Users/udenr/esp/esp-idf/components/openthread/include -IC:/Users/udenr/esp/esp-idf/components/openthread/openthread/include -IC:/Users/udenr/esp/esp-idf/components/openthread/openthread/examples/platforms -IC:/Users/udenr/esp/esp-idf/components/openthread/openthread/include/openthread -IC:/Users/udenr/esp/esp-idf/components/openthread/openthread/src -IC:/Users/udenr/esp/esp-idf/components/openthread/openthread/src/core -IC:/Users/udenr/esp/esp-idf/components/openthread/openthread/src/lib/hdlc -IC:/Users/udenr/esp/esp-idf/components/openthread/openthread/src/lib/spinel -IC:/Users/udenr/esp/esp-idf/components/openthread/openthread/src/ncp -IC:/Users/udenr/esp/esp-idf/components/openthread/private_include -IC:/Users/udenr/esp/esp-idf/components/newlib/platform_include -IC:/Users/udenr/esp/esp-idf/components/freertos/config/include -IC:/Users/udenr/esp/esp-idf/components/freertos/config/include/freertos -IC:/Users/udenr/esp/esp-idf/components/freertos/config/riscv/include -IC:/Users/udenr/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -IC:/Users/udenr/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include -IC:/Users/udenr/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include/freertos -IC:/Users/udenr/esp/esp-idf/components/freertos/esp_additions/include -IC:/Users/udenr/esp/esp-idf/components/esp_hw_support/include -IC:/Users/udenr/esp/esp-idf/components/esp_hw_support/include/soc -IC:/Users/udenr/esp/esp-idf/components/esp_hw_support/include/soc/esp32h2 -IC:/Users/udenr/esp/esp-idf/components/esp_hw_support/port/esp32h2/. -IC:/Users/udenr/esp/esp-idf/components/esp_hw_support/port/esp32h2/private_include -IC:/Users/udenr/esp/esp-idf/components/heap/include -IC:/Users/udenr/esp/esp-idf/components/log/include -IC:/Users/udenr/esp/esp-idf/components/soc/include -IC:/Users/udenr/esp/esp-idf/components/soc/esp32h2 -IC:/Users/udenr/esp/esp-idf/components/soc/esp32h2/include -IC:/Users/udenr/esp/esp-idf/components/hal/platform_port/include -IC:/Users/udenr/esp/esp-idf/components/hal/esp32h2/include -IC:/Users/udenr/esp/esp-idf/components/hal/include -IC:/Users/udenr/esp/esp-idf/components/esp_rom/include -IC:/Users/udenr/esp/esp-idf/components/esp_rom/include/esp32h2 -IC:/Users/udenr/esp/esp-idf/components/esp_rom/esp32h2 -IC:/Users/udenr/esp/esp-idf/components/esp_common/include -IC:/Users/udenr/esp/esp-idf/components/esp_system/include -IC:/Users/udenr/esp/esp-idf/components/esp_system/port/soc -IC:/Users/udenr/esp/esp-idf/components/esp_system/port/include/riscv -IC:/Users/udenr/esp/esp-idf/components/esp_system/port/include/private -IC:/Users/udenr/esp/esp-idf/components/riscv/include -IC:/Users/udenr/esp/esp-idf/components/lwip/include -IC:/Users/udenr/esp/esp-idf/components/lwip/include/apps -IC:/Users/udenr/esp/esp-idf/components/lwip/include/apps/sntp -IC:/Users/udenr/esp/esp-idf/components/lwip/lwip/src/include -IC:/Users/udenr/esp/esp-idf/components/lwip/port/include -IC:/Users/udenr/esp/esp-idf/components/lwip/port/freertos/include -IC:/Users/udenr/esp/esp-idf/components/lwip/port/esp32xx/include -IC:/Users/udenr/esp/esp-idf/components/lwip/port/esp32xx/include/arch -IC:/Users/udenr/esp/esp-idf/components/lwip/port/esp32xx/include/sys -IC:/Users/udenr/esp/esp-idf/components/esp_netif/include -IC:/Users/udenr/esp/esp-idf/components/esp_event/include -IC:/Users/udenr/esp/esp-idf/components/driver/include -IC:/Users/udenr/esp/esp-idf/components/driver/deprecated -IC:/Users/udenr/esp/esp-idf/components/driver/analog_comparator/include -IC:/Users/udenr/esp/esp-idf/components/driver/dac/include -IC:/Users/udenr/esp/esp-idf/components/driver/gpio/include -IC:/Users/udenr/esp/esp-idf/components/driver/gptimer/include -IC:/Users/udenr/esp/esp-idf/components/driver/i2c/include -IC:/Users/udenr/esp/esp-idf/components/driver/i2s/include -IC:/Users/udenr/esp/esp-idf/components/driver/ledc/include -IC:/Users/udenr/esp/esp-idf/components/driver/mcpwm/include -IC:/Users/udenr/esp/esp-idf/components/driver/parlio/include -IC:/Users/udenr/esp/esp-idf/components/driver/pcnt/include -IC:/Users/udenr/esp/esp-idf/components/driver/rmt/include -IC:/Users/udenr/esp/esp-idf/components/driver/sdio_slave/include -IC:/Users/udenr/esp/esp-idf/components/driver/sdmmc/include -IC:/Users/udenr/esp/esp-idf/components/driver/sigma_delta/include -IC:/Users/udenr/esp/esp-idf/components/driver/spi/include -IC:/Users/udenr/esp/esp-idf/components/driver/temperature_sensor/include -IC:/Users/udenr/esp/esp-idf/components/driver/touch_sensor/include -IC:/Users/udenr/esp/esp-idf/components/driver/twai/include -IC:/Users/udenr/esp/esp-idf/components/driver/uart/include -IC:/Users/udenr/esp/esp-idf/components/driver/usb_serial_jtag/include -IC:/Users/udenr/esp/esp-idf/components/esp_pm/include -IC:/Users/udenr/esp/esp-idf/components/esp_ringbuf/include -IC:/Users/udenr/esp/esp-idf/components/console -IC:/Users/udenr/esp/esp-idf/components/vfs/include -IC:/Users/udenr/esp/esp-idf/components/esp_partition/include -IC:/Users/udenr/esp/esp-idf/components/esp_timer/include -IC:/Users/udenr/esp/esp-idf/components/ieee802154/include -IC:/Users/udenr/esp/esp-idf/components/mbedtls/port/include -IC:/Users/udenr/esp/esp-idf/components/mbedtls/mbedtls/include -IC:/Users/udenr/esp/esp-idf/components/mbedtls/mbedtls/library -IC:/Users/udenr/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -IC:/Users/udenr/esp/esp-idf/components/mbedtls/mbedtls/3rdparty/everest/include -IC:/Users/udenr/esp/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m -IC:/Users/udenr/esp/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m/p256-m -IC:/Users/udenr/esp/esp-idf/components/nvs_flash/include -IC:/Users/udenr/esp/esp-idf/components/spi_flash/include -march=rv32imac_zicsr_zifencei  -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -nostartfiles -Os -freorder-blocks -msave-restore -fmacro-prefix-map=C:/Users/udenr/esp/esp-idf/examp [... cut ...] esp-idf/components/bootloader/subproject/components/micro-ecc C:/Users/udenr/esp/esp-idf/components/newlib C:/Users/udenr/esp/esp-idf/components/partition_table C:/Users/udenr/esp/esp-idf/components/riscv C:/Users/udenr/esp/esp-idf/components/soc C:/Users/udenr/esp/esp-idf/components/spi_flash
-- Configuring done (5.0s)
-- Generating done (0.0s)
-- Build files have been written to: C:/Users/udenr/esp/esp-idf/examples/openthread/ot_rcp/build/bootloader
ninja: build stopped: subcommand failed.
```

zwx1995esp, 2024-01-24:
Hi, @hertugweile  Could you please show me which commit did you use? Thanks!

zwx1995esp, 2024-01-24:
You need to enable the config `OPENTHREAD_RCP_SPI` via menuconfig
```
(Top) → Component config → OpenThread → The RCP transport type → (X) SPI RCP
```

For RCP, it must be a radio native device. The configure `Connect via SPI` is used for host, not for RCP.

Indastri, 2024-01-24:
Hi, as the problem was solved I am closing this issue. If any other problems occur please feel free to reopen this issue or create a new one.

Cheers!

closed 2024-01-24 by Indastri