espressif/esp-idf #13029: ot_rcp does not work correctly on esp32h2

KEEPHEADU, 2024-01-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-beta1-263-ge49823f10c

### Espressif SoC revision.

ESP32-H2

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-H2-DevKitM-1 V1.2

### Power Supply used.

USB

### What is the expected behavior?

ot_rcp and ot_br form the border router.

### What is the actual behavior?

ot_rcp could not run successfully on esp32h2. Below is all the logs, it doesn't seem to start executing the main function

[log.txt](https://github.com/espressif/esp-idf/files/14005936/log.txt)

### Steps to reproduce.

idf.py set-target esp32h2

idf.py flash monitor -p /dev/ttyUSB1

KEEPHEADU, 2024-01-25:
This is the log of ot-br, and the log shows that the rcp cannot be reset
```
W(11567) OPENTHREAD:[W] Platform------: Wait for response timeout
W(13577) OPENTHREAD:[W] Platform------: Wait for response timeout
E(13577) OPENTHREAD:[C] Platform------: Failed to reset RCP!
E(13587) OPENTHREAD:[C] Platform------: ResetRcp() at radio_spinel_impl.hpp:281: Failure
```
[log.txt](https://github.com/espressif/esp-idf/files/14045766/log.txt)

zwx1995esp, 2024-01-25:
Hi @KEEPHEADU  Could you please show me more details about your env? 
1. Which type of the interface did you use for connecting BR with RCP, UART or SPI?
2. Did you check the connection pin between BR and RCP is correct?
3. Did you modify the sdkconfig via `idf.py menuconfig`?

KEEPHEADU, 2024-01-25:
Hi @zwx1995esp ,Thank you for your reply. 

> Which type of the interface did you use for connecting BR with RCP, UART or SPI?

UART

>Did you check the connection pin between BR and RCP is correct?

I followed the example image in the README to connect BR and RCP, using the default pins.

>Did you modify the sdkconfig via idf.py menuconfig?

Below is my sdkconfig file.
[sdkconfig--br.txt](https://github.com/espressif/esp-idf/files/14046440/sdkconfig--br.txt)
[sdkconfig--rcp.txt](https://github.com/espressif/esp-idf/files/14046441/sdkconfig--rcp.txt)

zwx1995esp, 2024-01-25:
Hi, I reproduce this issue on my local env, will investigate it ASAP. I will sync to you if I find something or fix it. So, please just give me some time.

zwx1995esp, 2024-01-25:
Hi @KEEPHEADU  , could you please try the rx pin `GPIO1` and tx pin `GPIO0` for C3? I tried these pins and worked well. If these pins work well, could you please try them first? And for the pin GPIO4 and pin GPIO5 do not work, I will discuss with the HW team and reply here later.

zwx1995esp, 2024-01-25:
Hi, we find a solution tio use GPIO4 and GPIO5 on C3, could you please try to add these code after [this line](https://github.com/espressif/esp-idf/blob/master/components/openthread/src/port/esp_openthread_uart.c#L66)?
```
        if (config->tx_pin != -1) {
            gpio_set_drive_capability(config->tx_pin, GPIO_DRIVE_CAP_3);
        }
        if (config->rx_pin != -1) {
            gpio_set_drive_capability(config->rx_pin, GPIO_DRIVE_CAP_3);
        }
```
(before calling the function `uart_set_pin`)

KEEPHEADU, 2024-01-26:
> Hi, we find a solution tio use GPIO4 and GPIO5 on C3, could you please try to add these code after [this [...]

Yes, ot-br works fine after adding this code.

closed 2024-01-26 by KEEPHEADU