espressif/esp-idf #13031: USB Host HID Demo throws Exception after attaching Keyboard

danielBreitlauch, 2024-01-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

framework-espidf @ 3.50102.0

### Espressif SoC revision.

ESP32-S3

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3 Dev board N16R8

### Power Supply used.

USB

### What is the expected behavior?

No exception

### What is the actual behavior?

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x42009281  PS      : 0x00060930  A0      : 0x8200937c  A1      : 0x3fc9dfd0  
A2      : 0x00000000  A3      : 0x3fc9e61c  A4      : 0x00000001  A5      : 0x00000001  
A6      : 0x0000003b  A7      : 0x3fc9e637  A8      : 0x8200927f  A9      : 0x3fc9dfa0  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x3fc9dfe0  
A14     : 0x3fc9e778  A15     : 0xb33fffff  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000005  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  

Backtrace: 0x4200927e:0x3fc9dfd0 0x42009379:0x3fc9e020 0x42009e7a:0x3fc9e050 0x42003337:0x3fc9e070 0x42008cd2:0x3fc9e0a0 0x4037ce75:0x3fc9e0c0

  #0  0x4200927e:0x3fc9dfd0 in hid_host_interface_list_create at managed_components/espressif__usb_host_hid/hid_host.c:425
  #1  0x42009379:0x3fc9e020 in hid_host_device_init_attempt at managed_components/espressif__usb_host_hid/hid_host.c:492 (discriminator 2)
  #2  0x42009e7a:0x3fc9e050 in client_event_cb at managed_components/espressif__usb_host_hid/hid_host.c:608
  #3  0x42003337:0x3fc9e070 in usb_host_client_handle_events at /Users/am_lazytude/.platformio/packages/framework-espidf/components/usb/usb_host.c:794
  #4  0x42008cd2:0x3fc9e0a0 in event_handler_task at managed_components/espressif__usb_host_hid/hid_host.c:156
  #5  0x4037ce75:0x3fc9e0c0 in vPortTaskWrapper at /Users/am_lazytude/.platformio/packages/framework-espidf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

### Steps to reproduce.

1. Build [esp-idf usb host example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/hid)
2. Upload
3. Attach Keyboard

### Debug Logs.

```plain
[... 139 lines cut ...]
                bmAttributes 0x3        INT
                wMaxPacketSize 64
                bInterval 4
        *** Interface descriptor ***
        bLength 9
        bDescriptorType 4
        bInterfaceNumber 2
        bAlternateSetting 0
        bNumEndpoints 1
        bInterfaceClass 0x3
        bInterfaceSubClass 0x1
        bInterfaceProtocol 0x1
        iInterface 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x83   EP 3 IN
                bmAttributes 0x3        INT
                wMaxPacketSize 64
                bInterval 4
I (10365) CLASS: Getting Manufacturer string descriptor
Barcode 
I (10365) CLASS: Getting Product string descriptor
AFANDA BARCODE
```

peter-marcisovsky, 2024-01-22:
Hi @danielBreitlauch, thank you for opening the issue. 
Could you please provide us with with more detailed log outptu? 
In menuconfig please go to `Component config → Log output → Default log verbosity` and change `Info` to `Debug`. 
Thank you.

danielBreitlauch, 2024-01-22:
I updated the logs in the issue.

roma-jam, 2024-01-23:
Hi @danielBreitlauch, 
The provided information is enough, thank you so much it.

I will make a fix ASAP.

closed 2024-01-25 by roma-jam