espressif/esp-idf #13033: problem with enabling automatic light sleep with ble " using the  bleprph/nimble/ power_save project

AhmedSalem636, 2024-01-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello ESP community, 
I have used the nimble/ power-save example for esp32 with external 32.768khz crystal to run my ble application on the lowest possible power consumption. I doubt my automatic light-sleep mode is enabled and I am not sure how to check if it does or not. 

my application uses the exact same power-save example with the addition of one more task with a xtaskdelay of 30 seconds 
can this additional task block the cpu frfom entering light-sleep?. even though I am yielding this task for 30 seconds allwoing the Freeartos to be in idle state?

the system seems to detect the XTAL and it has no problem with it, 

here is the prints the following at the very begining using espressif IDE
 
[image]

Also here are my code:

/*
 * SPDX-FileCopyrightText: 2015-2023 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include "esp_log.h"
#include "nvs_flash.h"
/* power management */
#include "esp_pm.h"
/* BLE */
#include "nimble/nimble_port.h"
#include "nimble/nimble_port_freertos.h"
#include "host/ble_hs.h"
#include "host/util/util.h"
#include "console/console.h"
#include "services/gap/ble_svc_gap.h"
#include "bleprph.h"
#include "sysinit/sysinit.h"
#include "host/ble_hs.h"
#include "services/gatt/ble_svc_gatt.h"
#include <time.h>
#include "nvs_flash.h"
#include "nvs.h"
#include "driver/gpio.h"
#include "esp_task_wdt.h"

#define Service_UUID 0xF0FF
#define READ_UUID 0xF01F
#define WRITE_UUID 0xF02F
uint16_t requesterID=5;
nvs_handle_t my_handle;
 char value[100];
 char deviceName[24]={'S','m','a','r','t','P','o','t','\0'};

 char Mode; // stores the mode we are working on ( hourly, daily, weekly)
 char Freq; // stores how frequently to water based on the selected mode
 char duration; // stores how long to water ( scheduled modes only)
 int WaterMeDuration; // stores the duration to manually water the POT when requested by the user
 uint32_t mSec_GlobalCounter=0; //counts the ms's parts( used to calculate for hrs,days, weeks)
 uint32_t Sec_GlobalCounter=0; //counts the Sec's parts( used to calculate for hrs,days, weeks)
 uint64_t GlobalCounter=0; // GlobalCounter= Sec_GlobalCounter*1000+mSec_GlobalCounter
 char Password[9]={'0','0','0','0','0','0','0','0','\0'}; // password default for unpaird device 00000000
 size_t password_KEY_Length=0;
 size_t name_KEY_Length=0;
 size_t mode_KEY_Length=0;
 size_t freq_KEY_Length=0;
 size_t duration_KEY_Length=0;
[... 732 lines cut ...]
          { // if this 34's value isn't blank, copy it in Freq
        	  nvs_get_i8(my_handle,"Freq", (int8_t*)&Freq);
          }
          if(nvs_get_i8(my_handle,"Duration",&duration_KEY_Length)!=ESP_ERR_NVS_NOT_FOUND)
          { // if this 35's value isn't blank, copy it in duration
        	  nvs_get_i8(my_handle,"Duration", (int8_t*)&duration);
          }
          nvs_close(my_handle);

    /* Set the default device name. */

    /* XXX Need to have template for store */
	  rc = ble_svc_gap_device_name_set(deviceName);
	  assert(rc == 0);
    ble_store_config_init();
    nimble_port_freertos_init(bleprph_host_task);

    xTaskCreate(&myTask, "myTask", configMINIMAL_STACK_SIZE * 3, NULL, 1, NULL);
    /* Initialize command line interface to accept input from user */
    rc = scli_init();
    if (rc != ESP_OK) {
        ESP_LOGE(tag, "scli_init() failed");
    }

}

AhmedSalem636, 2024-01-23:
Note: I am following this [link ](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble/power_save)

update, the system does enter into light-sleep mode after enabling this:
*] configUSE_TICKLESS_IDLE
(3)     configEXPECTED_IDLE_TIME_BEFORE_SLEEP

however, the current consumption is in range 40/50mA which is way too much compared to the documentation that says 2mA

 I notice that in the menuconfig, I cannot find the option Component config > PHY > [*] Power down MAC and baseband of Wi-Fi and Bluetooth when PHY is disabled
 
 can this be the cause of the issue?
 
 I would appreciate your support. 
 thanks

xiongweichao, 2024-01-24:
Hi @AhmedSalem636 ,

1. May I ask what the current consumption is when you run this [example](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble/power_save)?
2. Please provide sdkconfig and complete log.

Thanks

AhmedSalem636, 2024-01-24:
Hi thanks for tour response, 
Here is the sdkconfig file and the complete log below:
[sdkconfig.json](https://github.com/espressif/esp-idf/files/14039761/sdkconfig.json)

for power consumption, I cannot measure them accurately but I am sure it is about 30:15mA, at any case, I think it is much more than the 1.9mA that's specified in the example

I (33) boot: ESP-IDF v5.1.2-dirty 2nd stage bootloader
I (33) boot: compile time Jan 24 2024 16:26:54
I (33) boot: Multicore bootloader
I (37) boot: chip revision: v1.1
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (98) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=19c94h (105620) map
I (144) esp_image: segment 1: paddr=00029cbc vaddr=3ff80000 size=00004h (     4) load
I (144) esp_image: segment 2: paddr=00029cc8 vaddr=3ffbdb60 size=04b24h ( 19236) load
I (158) esp_image: segment 3: paddr=0002e7f4 vaddr=40080000 size=01824h (  6180) load
I (161) esp_image: segment 4: paddr=00030020 vaddr=400d0020 size=63c9ch (408732) map
I (314) esp_image: segment 5: paddr=00093cc4 vaddr=40081824 size=1ac50h (109648) load
I (374) boot: Loaded app from partition at offset 0x10000
I (375) boot: Disabling RNG early entropy source...
I (386) cpu_start: Multicore app
I (386) cpu_start: Pro cpu up.
I (387) cpu_start: Starting app cpu, entry point is 0x400817b0
0x400817b0: call_start_cpu1 at C:/Espressif/frameworks/esp-idf-v5.1.2/components/esp_system/port/cpu_start.c:157

I (0) cpu_start: App cpu up.
I (574) cpu_start: Pro cpu start user code
I (574) cpu_start: cpu freq: 160000000 Hz
I (574) cpu_start: Application information:
I (577) cpu_start: Project name:     power_save
I (582) cpu_start: App version:      1
I (587) cpu_start: Compile time:     Jan 24 2024 16:26:46
I (593) cpu_start: ELF file SHA256:  24071a3517c6005c...
I (599) cpu_start: ESP-IDF:          v5.1.2-dirty
I (604) cpu_start: Min chip rev:     v0.0
I (609) cpu_start: Max chip rev:     v3.99 
I (614) cpu_start: Chip rev:         v1.1
I (33) boot: ESP-IDF v5.1.2-dirty 2nd stage bootloader
I (33) boot: compile time Jan 24 2024 16:26:54
I (33) boot: Multicore bootloader
I (37) boot: chip revision: v1.1
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (98) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=19c94h (105620) map
I (144) esp_image: segment 1: paddr=00029cbc vaddr=3ff80000 size=00004h (     4) load
I (144) esp_image: segment 2: paddr=00029cc8 vaddr=3ffbdb60 size=04b24h ( 19236) load
I (158) esp_image: segment 3: paddr=0002e7f4 vaddr=40080000 size=01824h (  6180) load
I (161) esp_image: segment 4: paddr=00030020 vaddr=400d0020 size=63c9ch (408732) map
I (314) esp_image: segment 5: paddr=00093cc4 vaddr=40081824 size=1ac50h (109648) load
I (374) boot: Loaded app from partition at offset 0x10000
I (375) boot: Disabling RNG early entropy source...
I (386) cpu_start: Multicore app
I (386) cpu_start: Pro cpu up.
I (387) cpu_start: Starting app cpu, entry point is 0x400817b0
0x400817b0: call_start_cpu1 at C:/Espressif/frameworks/esp-idf-v5.1.2/components/esp_system/port/cpu_start.c:157

I (0) cpu_start: App cpu up.
I (574) cpu_start: Pro cpu start user code
I (574) cpu_start: cpu freq: 160000000 Hz
I (574) cpu_start: Application information:
I (577) cpu_start: Project name:     power_save
I (582) cpu_start: App version:      1
I (587) cpu_start: Compile time:     Jan 24 2024 16:26:46
I (593) cpu_start: ELF file SHA256:  24071a3517c6005c...
I (599) cpu_start: ESP-IDF:          v5.1.2-dirty
I (604) cpu_start: Min chip rev:     v0.0
I (609) cpu_start: Max chip rev:     v3.99 
I (614) cpu_start: Chip rev:         v1.1

xiongweichao, 2024-01-30:
Hi @AhmedSalem636 ,

1.9mA refers to the light sleep state. May I ask if you are talking about average power consumption?

Thanks

AhmedSalem636, 2024-01-30:
Hi, 
Thanks for your reply. I see now the problem, Yes, I meant the average. is there any way to lower this average as low as possible?

xiongweichao, 2024-01-31:
Hi @AhmedSalem636 ,

1. The average power consumption of `release/v5.1` will be smaller than` v5.1.2`.
2. Increase adv (connection) interval.

I'm not sure how much the average power consumption can be reduced, need you to actually measure it.

Thanks

AhmedSalem636, 2024-02-02:
Hi 
many thanks for the answer, changing with the connection interval seems to work pretty well.

closed 2024-02-02 by AhmedSalem636