espressif/esp-idf #13036: CMakeLists.txt Question

diplfranzhoepfinger, 2024-01-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Why does this Work: 

```
```

but this not: 

```
```

the later one gives an Error: 

```
```

fhrbata, 2024-01-23:
Hello @diplfranzhoepfinger 

I believe this is by design.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#component-requirements

> The values of REQUIRES and PRIV_REQUIRES should not depend on any configuration choices (CONFIG_xxx macros). This is because requirements are expanded before the configuration is loaded. Other component variables (like include paths or source files) can depend on configuration choices.

HTH

igrr, 2024-01-23:
(You can, however, use the `target` variable which you have already initialized — just replace `if(CONFIG_IDF_TARGET_ESP32S3)` with `if(target STREQUAL esp32s3)`)

KonstantinKondrashov, 2024-01-24:
Hi @diplfranzhoepfinger!
I close it, here is a summary:
- Avoid using CONFIG_xxx macros for REQUIRES and PRIV_REQUIRES.
- You can use `if(target STREQUAL esp32s3)` for example.

closed 2024-01-24 by KonstantinKondrashov