espressif/esp-idf #13038: OTA of a new partition table

meakashrao, 2024-01-23:
### Is your feature request related to a problem?

So, a year back some of our products were shipped with min_spiffs.csv and we are very close to hitting the partition space(we are using 90% of the space)
After more analysis, it was figured out that Flash is actually 8MB but we never used the other 4MB more than that defined in the min_spiffs.csv

So, we need a method to update the partition table through OTA, I tried accessing information. There's no official way of doing it,

### Describe the solution you'd like.

If it's possible I would like to update the partitions.bin somehow to take care of the issue. 

Is it possible?

### Describe alternatives you've considered.

Alternate would be serial upload but that is troublesome because there are devices everywhere in the country

igrr, 2024-01-23:
Are you referring to [min_spiffs.csv](https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/min_spiffs.csv) from arduino-esp32 project? If yes, are you using the coredump partition? If no, could you please attach the partition table you are using?

Currently, there is no supported and recommended way of updating the partition table or the 2nd stage bootloader over the air.

There are two possible workarounds for this issue. Both are **not recommended** as they involve some risks.

1. If you only need to increase the SPIFFS partition size: it is possible to modify the values of esp_partition_t structure describing SPIFFS partition at run time. Changing the size of SPIFFS partition will result in it being reformatted, so make sure you are okay with that. Also, if there is another partition after SPIFFS (such as Coredump), this won't work.
  
   ```c++
   // hack: overwrite the size in the partition structure before initializing SPIFFS
   esp_partition_t* spiffs_part = (esp_partition_t*) esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_SPIFFS, NULL);
   spiffs_part->size = (desired size);
   // now, initialize SPIFFS as usual
   ```

2. If you also need to increase the size of OTA (application) partitions: rewriting the partition table is the only way. However, this is risky and not recommended. There is only one copy of the partition table in Flash, which means that the device will be bricked if it is powered off during such an OTA update. To overwrite the partition table, you need to disable `CONFIG_SPI_FLASH_DANGEROUS_WRITE_ABORTS` (enable `CONFIG_SPI_FLASH_DANGEROUS_WRITE_ALLOWED`), then use esp_flash_erase_range and esp_flash_write functions to overwrite the partition table.

meakashrao, 2024-01-23:
Are you referring to [min_spiffs.csv](https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/min_spiffs.csv) from arduino-esp32 project? If yes, are you using the coredump partition? If no, could you please attach the partition table you are using?

I am just not using the coredump partition and the extra 10000 size is added to spiffs and made 30000. 

The problem that happened is only 4MB is used in the min_spiffs.csv with the coredump modification but due to a need we need to use the other 4mb as well, so the only way is to update the partitions.bin

ammaree, 2024-01-23:
We have 100% successfully implemented option 2 to achieve a) the creation of an LFS partition and b) adjusting the size of the core dump partition.

Both operations we achieved in 1 step by checking the partition table very early in the startup procedure and making the changes as required. Once the changes were completed successfully the device was restarted to ensure a clean startup with the new partition table. Bricking the devices in our case was a very small risk since all devices operating of a large 24V 2-stage battery backup solution.

Since these devices are running with a 6 year old boot loader we are now in the process of implementing a field upgrade of the boot-loader using the same mechanism, downloading it to the LFS partition as a file, rebooting the device, and then doing the boot loader upgrade at a very early stage (with no Wifi or other tasks running) before rebooting with the new boot-loader in a clean startup.

As a safety precaution we are also looking at saving the old boat-loader to LFS just in case, not sure yet how we would be able to use it should something go wrong....

igrr, 2024-01-23:
@meakashrao  Do you need the other 4MB to be part of the same SPIFFS partition, or are you okay creating the 2nd SPIFFS partition in the remaining 4MB? The latter option would not require modifying the partition table, and could also be done if you are using the Arduino IDE.

meakashrao, 2024-01-23:
> @meakashrao Do you need the other 4MB to be part of the same SPIFFS partition, or are you okay creating the 2nd SPIFFS partition in the remaining 4MB? The latter option would not require modifying the partition table, and could also be done if you are using the Arduino IDE.

Hello, I want to increase the size of ota0 and ota1 not just spiffs, the whole 8MB space is to be utilized.

meakashrao, 2024-01-23:
> Both operations we achieved in 1 step by checking the partition table very early in the startup procedure and making the changes as required. Once the changes were completed successfully the device was restarted to ensure a clean startup with the new partition table

Hello, 

Please could you shed more info as to how did you use the functions to update the partition size, then how did you manage starting of ESP to run the firmware from the correct location?

ammaree, 2024-01-23:
In our case the OTA's stayed the same just changed the 2 last entries. You might require the change the flash size as well which is a different process.

To change the size of your OTA partitions to utilise the full 8MB will be a far more complex process, possibly multi step process requiring you to ensure that the device is booting into the lowest OTA slot, or possibly the factory firmware slot, if you still have that.

Once you are booted into firmware running right at the start of the flash, then modifying the partition table for the OTA starts and sizes might become possible, but suggest you follow @igrr lead on that part, I cannot say at all.

Jason2866, 2024-01-23:
We, do all the stuff mentioned. Project Tasmota had in older versions two OTA slots with identic sizes. Getting in trouble for newer versions with firmware size we decided for a different way of doing updates. There is now a factory partition (smaller in size) which gets activated for OTA overwriting the one OTA slot with a new firmware. If this process fails the factory firmware kicks in again. The conversion from the two OTA scheme to the factory one OTA slot is done completely on the device.
Using safty belt checks (verifying) during the needed copy & changes we have zero bricked devices doing this converting.

meakashrao, 2024-01-23:
> We, do all the stuff mentioned. Project Tasmota had in older versions two OTA slots with identic sizes. Getting in trouble for newer versions with firmware size we decided for a different way of [...]

So, 

factory_app
app0
spiffs

is what you are referring to?

Also, 
Am I on older version of littlefs?
[image]
I am using LITTLEFS @2.0.0

Could you please help and share more light on the safety checks and how to go about it?

Jason2866, 2024-01-23:
> My fear is that the partition won't allow the bootloader to run at the firmware from the correct location

No problem. Entry needs to be set in partition subtype `ota`

Jason2866, 2024-01-23:
> Could you please help and share more light on the safety checks and how to go about it?

Already answered in Tasmota Discord ;-)

meakashrao, 2024-01-23:
> > Could you please help and share more light on the safety checks and how to go about it?
> 
> Already answered in Tasmota Discord ;-)

oh, sorry. checking now

meakashrao, 2024-01-24:
> @meakashrao Do you need the other 4MB to be part of the same SPIFFS partition, or are you okay creating the 2nd SPIFFS partition in the remaining 4MB? The latter option would not require modifying the partition table, and could also be done if you are using the Arduino IDE.

How?

meakashrao, 2024-01-25:
So, there's a new trouble that's happening 

I am trying to erase the flash sector (offset: 0x8000 and size 0x1000 which is for my partition table space), I am using ESP32-Wroom-32E. It is said that although the length of the partition table is 0xC00 bytes but due to checksum it ends up occupying the whole flash sector, I have verified the offset to be 0x8000 from the sdkconfig.h
#define CONFIG_PARTITION_TABLE_OFFSET 0x8000

I am using a simple check to find my current size of ota_0 and ota_1 partition, if the size is 0x1E0000 as is for min_spiffs.csv(I removed the coredump and added the extra space to spiffs), then I go and check for the current running partition, if the partition is ota_1, I try to update the partition table by first erasing it and then writing the new data. 

I have defined a const char array which contains my partition table 
[image]

and I use, 
bool write_partition_table(const char *new_table)
{
    if (CONFIG_PARTITION_TABLE_OFFSET < 0x10000)    //Add proper conditions, need to be redone, make sure not writing over bootloader or application code
    {

        ESP_LOGI(PT_TAG, "SPI Erase Starts");
        esp_err_t ret = spi_flash_erase_range(0x8000, 0x1000);
        if(ret != ESP_OK)
        {
            ESP_LOGI(PT_TAG,"Flash Erase Failed Error: %d\n", ret);
            //TODO: Send Error Log, return something
            return false;
        }
        else
        {
            ESP_LOGI(PT_TAG, "Some Other Erase Error");
            //TODO: Send Error Log, return something
            return false;
        }

        ESP_LOGI(PT_TAG, "SPI Erase Ends, Write Starts");
        ret = spi_flash_write(CONFIG_PARTITION_TABLE_OFFSET, new_table, 0x1000);
        if (ret != ESP_OK)
        {
            ESP_LOGI(PT_TAG,"Flash Write Failed Error: %d\n", ret);
            //TODO: Send Error Log, return something
            return false;
        }
        else
        {
            ESP_LOGI(PT_TAG, "Some Other Write Error");
            //TODO: Send Error Log, return something
            return false;
        }

        ESP_LOGI(PT_TAG, "SPI Write Ends, NVS Starts");       
        nvs_flash_erase();
        nvs_flash_init();
        ESP_LOGI(PT_TAG, "NVS Ends, Rebooting");
        esp_restart();
    }
    else
    {
        ESP_LOGI(PT_TAG,"Either the Partition Table Address Offset is more than 10000");
        //TODO: Send Error Log, return something
        return false;
    }
}

But while erasing the flash sector, I get an error E (16696) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
[image]

As, I previously mentioned that I am not using coredump, nonethless, I erased the whole flash, did serial coding with min_spiffs.csv with coredump as the partition and then tried to update the partition but I still get the same error. 

[image]

Is there something that I am doing fundamentally wrong? Please help with what the error is and how to tackle it.

Jason2866, 2024-01-25:
@meakashrao You cant write to whole flash with standard Arduino framework. `CONFIG_SPI_FLASH_DANGEROUS_WRITE_ABORTS` is active!

You can try with our Tasmota platform where `CONFIG_SPI_FLASH_DANGEROUS_WRITE_ALLOWED` is set.
Use with
```
platform = https://github.com/tasmota/platform-espressif32/releases/download/2024.01.01/platform-espressif32.zip
```

meakashrao, 2024-01-25:
I am using Platform IO and Arduino as the framework, it's a bit weird as to if that option is configurable for ESP-IDF then why not for Platform IO with Arduino Framework or the Arduino IDE all together. 
@igrr any comments on this?

@Jason2866 Thank you for letting me know that Tasmota platfrom has the option. I will try to put this up. 

Also, @ammaree do you use ESP-IDF?

ammaree, 2024-01-25:
Yes, ESP-IDF with VSCode + plugin

closed 2024-02-16 by meakashrao