espressif/esp-idf #13041: tiT Task Priority doesn't scale when configMAX_PRIORITIES is adjusted downward.

SolidStateLEDLighting, 2024-01-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I like the idea of having a maximum Task priory range of less than 25 (which is the default in freeRTOS).   I want my list of tasks to be close to each other in numerical scale because I believe this would yield a more efficient context change.

When I reduce configMAX_PRIORITIES, all the RTOS priorities scale down with it (ipc, wif, esp_timer, sys_evt, etc) Except tiT will not scale.   That interrupt priority has been fixed at 18.  The better solution would be for it to know how many number below configMAX_PRIORITIES it should assign itself to as the other tasks do.

KonstantinKondrashov, 2024-01-24:
Hi @SolidStateLEDLighting!
The priority of the tiT task is defined by https://github.com/espressif/esp-idf/blob/master/components/esp_system/include/esp_task.h#L55. You can manually adjust it, it would be a simple way to do it. 

> I want my list of tasks to be close to each other in numerical scale because I believe this would yield a more efficient context change.

Could you give some points why you think it is more efficient? From your perspective, it does not have any benefits or noticeable benefits.

Thanks.

SolidStateLEDLighting, 2024-01-24:
I just took another look at for tiT -- I found that priority setting inside menuconfig.   I didn't see that before.   So, I can get control of that.    So, I guess I'm wrong about there being a mistake in the actual IDF build.

KonstantinKondrashov, 2024-01-24:
Yes, there is a Kconfig option - https://github.com/espressif/esp-idf/blob/master/components/lwip/Kconfig#L29-L35, which can be changed through `idf.py menuconfig`.
Usually, we have a Kconfig option for a component to set the task priority (if it is not linked to configMAX_PRIORITIES).

If the issue is solved I close it. Thanks.

closed 2024-01-24 by KonstantinKondrashov

sudeep-mohanty, 2024-01-24:
Hi @SolidStateLEDLighting,
On the dual-core system such as the S3, having a lower `configMAX_PRIORITIES` can make a small difference to the context switching performance of the kernel but it shouldn't be a noticeable difference. The only scenario where this would come into picture is if there are a few high priority ready tasks and a few low priority ready tasks, with nothing in between. Even then the scheduler is pretty optimized to quickly skip the priority levels which do not have any ready tasks. See [here](https://github.com/espressif/esp-idf/blob/b3f7e2c8a4d354df8ef8558ea7caddc07283a57b/components/freertos/FreeRTOS-Kernel/tasks.c#L3541) for some insights. Thanks!

SolidStateLEDLighting, 2024-01-24:
Yes, I would imagine that you would never notice anything until you have a very loaded system full of busy tasks.

My style of development is to do very small amounts of work and then allow the scheduler to change tasks if it needs to.  This gives me the best possible opportunity to share CPU time between tasks, but I'm also subject to a larger number of context changes.

This is why I'm looking for everything that I can to tune performance.

Thanks for the info.