espressif/esp-idf #13045: Wifi does not reconnect. recv deauth, reason=0xfc

mzakharocsc, 2024-01-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.2

### Espressif SoC revision.

ESP32-S3 v0.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-DevkitC-1

### Power Supply used.

USB

### What is the expected behavior?

Wifi Reconnects

### What is the actual behavior?

Wifi does not reconnect

### Steps to reproduce.

1. Connect to Wifi
2. Wait for deauth request

### Debug Logs.

```plain
V (1808355) wifi:mgmt dir=2 not NODS
V (1808375) wifi:tx process: alloc static tx buf, pb=0x0 l=42 hdr=24 data=18
V (1808395) wifi:tx process: alloc static tx buf, pb=0x3fccc044 l=43 hdr=0 data=43
D (1808415) wifi:recv deauth, reason=0xfc
I (1808415) wifi:state: run -> init (fcc0)
D (1808415) wifi:recv deauth/disassoc, stop beacon/connect timer
D (1808415) wifi:connect status 5 -> 4
D (1808425) wifi:sta leave
I (1808425) wifi:pm stop, total sleep time: 0 us / 1800780585 us

D (1808435) wifi:stop CSA timer
I (1808435) wifi:<ba-del>idx:0, tid:0
I (1808435) wifi:<ba-del>idx:1, tid:3
V (1808445) wifi:bss connection retry cnt=0
D (1808445) wifi:remove f4:0f:1b:dd:dc:52 from rc list
I (1808445) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
D (1808455) wifi:filter: set rx policy=8
D (1808455) wifi:Send disconnect event, reason=252, AP number=0
V (1808465) wifi:tx process: alloc static tx buf, pb=0x0 l=42 hdr=24 data=18
D (1808475) event: running post WIFI_EVENT:5 with handler 0x42027d74 and context 0x3fceee10 on loop 0x3fce9ba0[0m
[... 229 lines cut ...]
V (1811825) wifi:tx process: alloc static tx buf, pb=0x0 l=42 hdr=24 data=18
V (1811875) wifi:tx process: alloc static tx buf, pb=0x0 l=42 hdr=24 data=18
[0;31mE (1811905) ping_sock: send error=0[0m
V (1811925) wifi:tx process: alloc static tx buf, pb=0x0 l=42 hdr=24 data=18
V (1811975) wifi:tx process: alloc static tx buf, pb=0x0 l=42 hdr=24 data=18
[0;31mE (1812005) ping_sock: send error=0[0m
V (1812025) wifi:tx process: alloc static tx buf, pb=0x0 l=42 hdr=24 data=18
V (1812075) wifi:tx process: alloc static tx buf, pb=0x0 l=42 hdr=24 data=18
[0;31mE (1812105) ping_sock: send error=0[0m
V (1812125) wifi:tx process: alloc static tx buf, pb=0x0 l=42 hdr=24 data=18
V (1812175) wifi:tx process: alloc static tx buf, pb=0x0 l=42 hdr=24 data=18
[0;31mE (1812205) ping_sock: send error=0[0m
V (1812225) wifi:tx process: alloc static tx buf, pb=0x0 l=42 hdr=24 data=18
V (1812275) wifi:tx process: alloc static tx buf, pb=0x0 l=42 hdr=24 data=18
[0;31mE (1812305) ping_sock: send error=0[0m
V (1812305) esp_adapter: thread sem get: sem=0x3fcf7aec[0m
D (1812305) wifi:Start wifi disconnect
D (1812305) wifi:connect status 0 -> 0
D (1812305) wifi:filter: set rx policy=8
V
```

### More Information.

Reboot of ESP32 solves the problem.

nishanth-radja, 2024-01-29:
@mzakharocsc Can we have the following details
1.AP model being used and the security in which it is configured in ?
2.Sniffer capture when the issue is seen ?
3.can we have the code snippet of the API's used for connecting the STA to AP ?
4.What is the commit ID of the IDF build being used ?

mzakharocsc, 2024-02-06:
it turns out reconnect logic was incorrect on the application side.  disconnect reason=0xfc  does allow STA to reconnect, however it is still unclear what reason=0xfc means.

closed 2024-02-06 by mzakharocsc

MaxwellAlan, 2024-02-07:
@mzakharocsc 
Yeah, the reason code of 0xfc is so weird, could you get the sniffer logs?
Or could share some more detail about this Router(Model, firmware versions etc).
Thanks.