espressif/esp-idf #13046: Tools: idf.py monitor breaks non-ascii output

nebkat, 2024-01-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

See also report in https://github.com/espressif/esp-idf-monitor/issues/12

--------

Running on Windows 11, PowerShell 7.4.1, esp-idf-monitor 1.3.4:

When printing various non-ascii characters in serial output over `esp-idf-monitor`, the Unicode [replacement character](https://en.wikipedia.org/wiki/Specials_(Unicode_block)#Replacement_character) `0xEF 0xBF 0xBD` is prepended before every such character.

# Example

```cpp
ESP_LOGI(TAG, "╔════════════════════════════════════════════════╗");
ESP_LOGI(TAG, "║ Version     │ %-32s "                         "║", app_desc->version);
ESP_LOGI(TAG, "║ ESP-IDF     │ %-32s "                         "║", app_desc->idf_ver);
ESP_LOGI(TAG, "║ ELF SHA256  │ %-32s "                         "║", app_desc.elf_sha256_hex(8).c_str());
ESP_LOGI(TAG, "╟─────────────┼──────────────────────────────────╢");
ESP_LOGI(TAG, "║ Chip        │ %-32s "                         "║", chip_info.model_name());
ESP_LOGI(TAG, "║ Revision    │ %-32d "                         "║", chip_info->revision);
ESP_LOGI(TAG, "║ Cores       │ %-32d "                         "║", chip_info->cores);
ESP_LOGI(TAG, "╟─────────────┼──────────────────────────────────╢");
ESP_LOGI(TAG, "║ Partition   │ %-32s "                         "║", running_partition->label);
ESP_LOGI(TAG, "║ Boot reason │ %-32s "                         "║", sys::reset_reason_string());
ESP_LOGI(TAG, "╚═════════════╧══════════════════════════════════╝");
```

```
I (848) APP       : ���╔���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���╗
I (888) APP       : ���║ Version     ���│ 0d46b8c-dirty                    ���║
I (898) APP       : ���║ ESP-IDF     ���│ v5.2-beta2-dirty                 ���║
I (908) APP       : ���║ ELF SHA256  ���│ 4ebdab159d1674d4                 ���║
I (918) APP       : ���╟���─���─���─���─���─���─���─���─���─���─���─���─���─���┼���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���╢
I (928) APP       : ���║ Chip        ���│ ESP32                            ���║
I (938) APP       : ���║ Revision    ���│ 300                              ���║
I (948) APP       : ���║ Cores       ���│ 2                                ���║
I (958) APP       : ���╟���─���─���─���─���─���─���─���─���─���─���─���─���─���┼���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���─���╢
I (968) APP       : ���║ Partition   ���│ ota_0                            ���║
I (978) APP       : ���║ Boot reason ���│ POWERON                          ���║
I (988) APP       : ���╚���═���═���═���═���═���═���═���═���═���═���═���═���═���╧���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���═���╝
````

------

This is "fixed" by disabling the ANSI conversion, either with `--no-hints` or by modifying this line (force `convert_output` to `False`)

https://github.com/espressif/esp-idf/blob/b3f7e2c8a4d354df8ef8558ea7caddc07283a57b/tools/idf_py_actions/tools.py#L398

Problem does _not_ occur when running `python -m esp_idf_monitor` directly.

fhrbata, 2024-01-25:
Hello @nebkat,

thank you for the thorough report. It appears that there's a bug in the converter related to how it processes multibyte characters. The problem lies in the 'decode_buffer,' where input bytes are being added multiple times, as we are redundantly including bytes that are already present in the buffer when the `decode` fails.

The difference between `python -m esp_idf_monitor` and `idf.py monitor` seems to be that monitor is writing bytes directly to `sys.stdout.buffer`, which does not undergo decode in converter, while idf.py is using `sys.stdout`, where the decode it done.
 
The following patch seems to fix the problem for me.

```diff
diff --git a/esp_idf_monitor/base/ansi_color_converter.py b/esp_idf_monitor/base/ansi_color_converter.py
index c982be1..2dbf8ec 100644
--- a/esp_idf_monitor/base/ansi_color_converter.py
+++ b/esp_idf_monitor/base/ansi_color_converter.py
@@ -68,8 +68,8 @@ class ANSIColorConverter(object):
     def _output_write(self, data):  # type: (Union[str, bytes]) -> None
         try:
             if self.decode_output:
-                data = self.decode_buffer + data  # type: ignore
-                self.output.write(data.decode())  # type: ignore
+                self.decode_buffer += data  # type: ignore
+                self.output.write(self.decode_buffer.decode())  # type: ignore
                 self.decode_buffer = b''
             else:
                 self.output.write(data)  # type: ignore
@@ -82,7 +82,6 @@ class ANSIColorConverter(object):
             # (garbage bytes, etc)
             pass
         except UnicodeDecodeError:
-            self.decode_buffer += data  # type: ignore
             if len(self.decode_buffer) > 4:
                 # Multi-byte character contain up to 4 bytes and if buffer have more then 4 bytes
                 # and still can not decode it we can just ignore some bytes
```

If you would like to test this, you can apply the above patch to [esp-idf-monitor](https://github.com/espressif/esp-idf-monitor) and install it in the editable mode with `pip install -e <path to the esp-idf-monitor git clone with the patch applied>`. It has to be installed within the `esp-idf` environment. Meaning after running e.g. `export.bat`.

Thank you

nebkat, 2024-01-25:
Hi @fhrbata, thank you for the quick response - that does indeed also fix the problem for me!

fhrbata, 2024-01-25:
> Hi @fhrbata, thank you for the quick response - that does indeed also fix the problem for me!

Hello @nebkat ,

thank you very much for such a quick confirmation. It's very much appreciated!

closed 2024-01-30 by espressif-bot