espressif/esp-idf #13047: support version macros in Component Registry components

dannybackx, 2024-01-24:
### Is your feature request related to a problem?

I don't see a way to specify a component's version just once and have it propagate to other files.

In a component (not mine) I'm using, I find 6 distinct places (not counting the changelog) where the number has been individually maintained : 

victus: {106} find managed_components/bblanchon__arduinojson -type f -exec grep 7.0.2 {} \; -print
version=7.0.2
managed_components/bblanchon__arduinojson/library.properties
version: 7.0.2
managed_components/bblanchon__arduinojson/idf_component.yml
#define ARDUINOJSON_VERSION "7.0.2"
managed_components/bblanchon__arduinojson/src/ArduinoJson/version.hpp
version: 7.0.2.{build}
managed_components/bblanchon__arduinojson/appveyor.yml
v7.0.2 (2024-01-19)
managed_components/bblanchon__arduinojson/CHANGELOG.md
  "version": "7.0.2",
managed_components/bblanchon__arduinojson/library.json
project(ArduinoJson VERSION 7.0.2)
managed_components/bblanchon__arduinojson/CMakeLists.txt
victus: {107} 

### Describe the solution you'd like.

I'm sure we can do better than this.

igrr, 2024-01-24:
For the managed components, there is already
```cmake
idf_component_get_property(ver namespace__componentname COMPONENT_VERSION)
```
and you can then propagate the version into C source code using `target_compile_definitions`. So I think this is already supported from IDF side.

That said, with tools like https://commitizen-tools.github.io/commitizen/ it's not really a problem to automatically update the version number in multiple files when making a release. So it's up to the maintainer of the particular component to decide which approach to use.

dannybackx, 2024-01-25:
Thanks, but I can't find anything about that in the docs. How do I call or use this ?
Looking for COMPONENT_VERSION on https://docs.espressif.com returns no results.
Same thing for 
  find $IDF_PATH -type f -exec grep COMPONENT_VERSION {} \; -print

igrr, 2024-01-25:
Sorry about that, we forgot to mention this in the docs. It's only mentioned in the changelog of the component manager: https://github.com/espressif/idf-component-manager/blob/287b9e4351afd363a534aee9a7f2e8ea5fe1162e/CHANGELOG.md?plain=1#L178.

You can add this line to the CMakeLists.txt file where you need to access the version. The first argument is the variable which will receive the version value. The second argument is the component name. If the component is obtained from the registry, use the namespace and the component name, separated by double underscore. Here's an example:

https://github.com/espressif/esp-bsp/blob/f0c6bc2befa1daec8bcd86c931e36bb98e7caa9b/components/esp_lvgl_port/CMakeLists.txt#L3

We'll add this to the docs.

dannybackx, 2024-01-27:
Took a bit of experimenting to get this to work. I assume this is what you mean ? (Lines 4 and 36)
`
#
# Grab the Acmeclient version number
#
idf_component_get_property(acmeclient_version dannybackx__acmeclient COMPONENT_VERSION)
#
# message(STATUS "acmeclient version: ${acmeclient_version}")
#

set(SOURCES
        App.cpp Ota.cpp WebServer.cpp Lcd.cpp Config.cpp
        Sensors.cpp Alarm.cpp Siren.cpp Peers.cpp
        Logger.cpp
        App.h Ota.h WebServer.h Lcd.h Config.h
        Sensors.h Alarm.h Siren.h Peers.h
        esp32s3-zx3d95ce01s.h screen.c
        Logger.h
        Network.cpp Network.h
)

idf_component_register(
    SRC_DIRS ${DIRS} "."
    INCLUDE_DIRS "." ${DIRS} ${BUILD_DIR}
    EMBED_TXTFILES icon.png
    REQUIRES esp_lcd driver lvgl esp_timer bus touch_panel
      esp_event nvs_flash esp_wifi esp_timer esp_netif
      app_update
      esp_http_server esp_https_server
      arduinojson acmeclient
      WhatsApp rcswitch ftpserver
)

target_compile_options(${COMPONENT_LIB} PRIVATE "-Wno-format" "-Wno-deprecated-enum-enum-conversion")
#
# Insert Acmeclient version number into build
#
target_compile_options(${COMPONENT_LIB} PRIVATE "-DACMECLIENT_VERSION=\"${acmeclient_version}\"")

add_custom_command(OUTPUT ${BUILD_DIR}/build.h
        COMMAND ${COMPONENT_DIR}/generate_build_date ${BUILD_DIR}/build.h
        DEPENDS ${SOURCES} CMakeLists.txt ${BUILD_DIR}/config/sdkconfig.h
        VERBATIM)

add_custom_target(buildh DEPENDS ${BUILD_DIR}/build.h)
add_dependencies(${COMPONENT_LIB} buildh)
`

dannybackx, 2024-01-27:
Looks like I can, see https://sourceforge.net/p/esp32-acme-client/code/HEAD/tree/trunk/CMakeLists.txt .

closed 2024-01-27 by dannybackx