espressif/esp-idf #13048: HALT occurs when esp_psram_init() is used on a SoC that does not have PSRAM

nopnop2002, 2024-01-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2-602-gdb1e54a0c5

### Espressif SoC revision.

ESP32D0WDQ5 (revision 3)

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 Development Board

### Power Supply used.

USB

### What is the expected behavior?

I want to know if my development board has PSRAM.
I tried to use esp_psram_init() to find out if it had PSRAM.
When my development board doesn't have PARAM, I expected esp_psram_init() to result in ESP_FAIL.
HALT occurs when esp_psram_init() is used on a SoC that does not have PSRAM.

### What is the actual behavior?

When my development board does not have PARAM, esp_psram_init() becomes ESP_FAIL.

### Steps to reproduce.

1. idf.py menuconfig
Component config  ---> ESP PSRAM ---> Enable Support for external, SPI-connected RAM

2. Build this Code:
```
#include <stdio.h>
#include "esp_log.h"
#include "esp_psram.h"

static const char *TAG = "MAIN";

void app_main()
{
        esp_err_t ret = esp_psram_init();
        ESP_LOGI(TAG, "esp_psram_init=%d", ret);
}
```

0xjakob, 2024-01-25:
@nopnop2002 Have you considered [CONFIG_SPIRAM_IGNORE_NOTFOUND](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-spiram-ignore-notfound)? It is also mentioned in the [PSRAM documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/external-ram.html#failure-to-initialize).

nopnop2002, 2024-01-25:
@0xjakob 

Thank you for your reply.

I enabled CONFIG_SPIRAM_IGNORE_NOTFOUND.

[image]

No longer HALT.

closed 2024-01-25 by nopnop2002