espressif/esp-idf #13049: esp_psram_init return value is incorrect

nopnop2002, 2024-01-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2-602-gdb1e54a0c5

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 Development board

### Power Supply used.

USB

### What is the expected behavior?

The return value of esp_psram_init is esp_err_t.
```
components/esp_psram/esp_psram.c:119:esp_err_t esp_psram_init(void)
components/esp_psram/include/esp_psram.h:27:esp_err_t esp_psram_init(void);
```

### What is the actual behavior?

The return value of esp_psram_init is not esp_err_t.

### Steps to reproduce.

1. idf.py menuconfig
Component config  ---> ESP PSRAM ---> Enable Support for external, SPI-connected RAM

2. Build this Code:
```
#include <stdio.h>
#include "esp_err.h"
#include "esp_log.h"
#include "esp_psram.h"

static const char *TAG = "MAIN";

void app_main()
{
    esp_err_t ret = esp_psram_init();
    ESP_LOGI(TAG, "esp_psram_init=%d", ret);
    if (ret == ESP_OK) {
        ESP_LOGE(TAG, "esp_psram_init success");
        size_t psram_size = esp_psram_get_size();
        ESP_LOGI(TAG, "PSRAM size: %d bytes", psram_size);
    } else {
        ESP_LOGE(TAG, "esp_psram_init fail");
    }
}
```

### Debug Logs.

```plain
I (2014) main_task: Calling app_main()
I (2014) MAIN: esp_psram_init=259
E (2024) MAIN: esp_psram_init fail
I (2024) main_task: Returned from app_main()
```
```

nopnop2002, 2024-01-25:
@0xjakob 

I ran it on an ESP32S3 that has PSRAM.

But the return value of esp_psram_init is 259 instead of ESP_OK.

```
I (2014) MAIN: esp_psram_init=259
```

0xjakob, 2024-01-25:
The [documentation of `esp_psram_init()`](https://github.com/espressif/esp-idf/blob/master/components/esp_psram/include/esp_psram.h#L24) says:
> - ESP_ERR_INVALID_STATE: PSRAM is initialized already

The PSRAM initialization is usually done automatically at startup. This is why `ESP_ERR_INVALID_STATE` is returned; because it is initialized already. You should see some lines in the log of ESP-IDF which indicate that PSRAM has been found:
> I (220) esp_psram: Found 8MB PSRAM device
I (220) esp_psram: Speed: 40MHz

nopnop2002, 2024-01-25:
I foud this on my logging.
```
I (172) esp_psram: Found 8MB PSRAM device
```

---

For others:
How to know if your ESP32 has PSRAM.
Component config ---> ESP PSRAM ---> Enable Support for external, SPI-connected RAM
Component config ---> ESP PSRAM ---> SPI RAM config  ---> Enable Ignore PSRAM when not found

If your ESP32 has PSRAM, esp_psram_get_size() returns the actual PSRAM size.

If your ESP32 does not have PSRAM, esp_psram_get_size() returns 0.

closed 2024-01-25 by nopnop2002