espressif/esp-idf #13050: ESP32S3在主机模式下，调用spi_device_release_bus()后，CS不失效

393877719, 2024-01-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

我使用ESP32S3的SPI主机(SPI2)，访问一个SPI外设，根据外设需求，我需要在ESP32S3发送完一帧数据后，CS保持有效，并持续发送数据轮询，直到外设响应数据之后，将CS失效。
我按照api指南，
1.调用spi_device_acquire_bus()
2.配置spi_transaction_t::flags==SPI_TRANS_CS_KEEP_ACTIVE
3.调用spi_device_transmit(),发送SPI数据，并循环，直到收到SPI从机外设响应，CS持续有效
4.调用spi_device_release_bus()，CS依然保持有效状态，我的期望是此时CS应该失效。
需要再次配置spi_transaction_t::flags==0，重新发送一帧SPI数据之后CS才能失效。
查了ESP32S3的参考手册，找到一个寄存器位SPI_SOFT_RESET：置位此位，可复位 SPI 时钟线、CS 线和数据线。但是我没有找到对应的API可操作。
尝试直接写寄存器的方式(*((volatile uint32_t *)0x60024000+0xE0)) |= (uint32_t)(1<<27);但是没有效果。
怀疑是寄存器地址错误，尝试读reset值，但是ESP32S3的响应值与手册reset值不相符。
测试代码：
`#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

void app_main(void)
{
    uint32_t reg_data;
    while (1)
    {
        reg_data = (*((volatile uint32_t *)0x60024000 + 0x08));
        printf("SPI_CTRL_REG=0x%lX\r\n", reg_data);
        reg_data = (*((volatile uint32_t *)0x60024000 + 0xE0));
        printf("SPI_SLAVE_REG=0x%lX\r\n", reg_data);
        vTaskDelay(pdMS_TO_TICKS(5000));
    }
    vTaskDelete(NULL);
}`
响应值：
`SPI_CTRL_REG=0x0
SPI_SLAVE_REG=0x0`
这个CS问题怎么解决？
[image]

wanckl, 2024-01-25:
@393877719 你好：      

`spi_device_acquire_bus`和`spi_device_release_bus`可能不是你理解的意思，这两个是驱动内部操作软件锁的，来保证有多个device的时候你KEEP期间不会被别的device打断。他们不会修改硬件行为。      

对于你的情况，你可以在调用`spi_device_release_bus`前最后发送一次空的传输(data, length, flag ...全是0)，你的做法是对的。    
第二个方法，你可以配置 `devcfg.spics_io_num = -1` 然后手动控制CS引脚。      

If all those can't reserve your issue, feel free to update here again, we can find new way together, best wishes !

393877719, 2024-01-25:
非常感谢你的答复。
按照第一个方法，我在调用`spi_device_release_bus`前最后发送一次空的传输(data, length, flag ...全是0)，但是实际监测到的信号是第二次SPI发送真实发生了，并且即便我length设置为0，发送的字节数依然是前一次CS拉低时设置的字节数。测试代码如下：
`#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/spi_master.h"
#include "driver/gpio.h"

#define AFE_TOTAL 3
#define ADI_MOSI 19
#define ADI_MISO 20
#define ADI_SCLK 21
#define ADI_CS_N 48
#define HSD_IO1 3
#define LSD_IO1 4

spi_device_handle_t ADI_SPI_hdl;
spi_transaction_t ADI_trans_desc;

void ESP_GPIO_init(void)
{
    gpio_config_t ioconfig = {
        .mode = GPIO_MODE_OUTPUT,
        .intr_type = GPIO_INTR_DISABLE,
        .pull_up_en = GPIO_PULLUP_DISABLE,
        .pull_down_en = GPIO_PULLDOWN_ENABLE,
        .pin_bit_mask = (1ULL << HSD_IO1) | (1ULL << LSD_IO1),
    };
    gpio_config(&ioconfig);
}
void ESP_SPI_init(void)
{
    spi_bus_config_t ADI_busconfig = {
        .mosi_io_num = ADI_MOSI,
        .miso_io_num = ADI_MISO,
        .sclk_io_num = ADI_SCLK,
        .quadhd_io_num = -1,
        .quadwp_io_num = -1,
    };
    spi_bus_initialize(SPI2_HOST, &ADI_busconfig, SPI_DMA_CH_AUTO);
    spi_device_interface_config_t ADI_devconfig = {
        .spics_io_num = ADI_CS_N,
        .clock_speed_hz = 1000000, // 频率大容易导致传输大量数据时CS乱跳
        .mode = 3,
        .queue_size = 3,
        .cs_ena_pretrans = 3,
        .cs_ena_posttrans = 1,
    };
    spi_bus_add_device(SPI2_HOST, &ADI_devconfig, &ADI_SPI_hdl);
}
void ESP_init(void)
{
    ESP_GPIO_init();
    ESP_SPI_init();
}

void app_main(void)
{
    uint8_t tmp[12]={0};
    ESP_init();
    while (1)
    {
        spi_device_acquire_bus(ADI_SPI_hdl,portMAX_DELAY);
        ADI_trans_desc.length = 4 * 8;
        ADI_trans_desc.rxlength = 0;
        ADI_trans_desc.tx_buffer = tmp;
        ADI_trans_desc.rx_buffer = NULL;
        ADI_trans_desc.flags |= SPI_TRANS_CS_KEEP_ACTIVE;
        spi_device_transmit(ADI_SPI_hdl, &ADI_trans_desc);
        
        ADI_trans_desc.length = 0;
        ADI_trans_desc.rxlength = 0;
        ADI_trans_desc.tx_buffer = NULL;
        ADI_trans_desc.rx_buffer = NULL;
        ADI_trans_desc.flags &= ~SPI_TRANS_CS_KEEP_ACTIVE;
        spi_device_transmit(ADI_SPI_hdl, &ADI_trans_desc);
        spi_device_release_bus(ADI_SPI_hdl);
        vTaskDelay(pdMS_TO_TICKS(2000));
    }
    vTaskDelete(NULL);
}
`
[image]
按照第二种方式的话，手动控制CS时，在MCU上电后的第一次SPI传输时，数据是错误的。

wanckl, 2024-02-04:
`空的传输(data, length, flag ...全是0)` 需要length=1，我不知道那天是怎么测出来的了，，   

另外，手动控制CS时为什么第一次会出错，好奇 @393877719

closed 2025-04-07 by Alvin1Zhang