espressif/esp-idf #13051: The old_snd field in WIFI_EVENT_HOME_CHANNEL_CHANGE seems wrong

AxelLin, 2024-01-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-1353-gb3f7e2c8a4

### Espressif SoC revision.

ESP32S3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Freenove ESP32-S3-WROOM Board

### Power Supply used.

USB

### What is the expected behavior?

The old_snd should be correct. (At least at an invalid value like 6)

### What is the actual behavior?

I found a new WIFI_EVENT_HOME_CHANNEL_CHANGE event when test 5.3-dev.

It looks like the old_snd is wrong.

device is running as STA:

I (1796) net_utils: WIFI_EVENT_HOME_CHANNEL_CHANGE: old_chan=0/old_snd=0, new_chan=1/new_snd=0
I (1801) net_utils: STA started
I (4212) wifi:new:<5,1>, old:<1,0>, ap:<255,255>, sta:<5,1>, prof:1
I (4573) wifi:state: init -> auth (b0)
I (4574) test: WIFI_EVENT_HOME_CHANNEL_CHANGE: old_chan=1/old_snd=1, new_chan=5/new_snd=1

In above log, the old_snd should be 0?

device is running as AP:
I (2131) phy_init: phy_version 620,ec7ec30,Sep  5 2023,13:49:13
I (2167) wifi:mode : softAP (34:85:18:6e:fa:7d)
I (2171) wifi:Total power save buffer number: 8
I (2172) wifi:Init max length of beacon: 752/752
I (2172) wifi:Init max length of beacon: 752/752
I (2174) test: WIFI_EVENT_HOME_CHANNEL_CHANGE: old_chan=0/old_snd=0, new_chan=6/new_snd=1
I (2183) test: WIFI_EVENT_HOME_CHANNEL_CHANGE: old_chan=6/old_snd=6, new_chan=6/new_snd=0

In above log, the old_snd is 6? (But the valid wifi_second_chan_t is 0/1/2)

### Steps to reproduce.

Just print WIFI_EVENT when received WIFI_EVENT_HOME_CHANNEL_CHANGE.

zhangyanjiaoesp, 2024-02-05:
@AxelLin Thanks for reporting, will fix it ASAP.

closed 2024-02-20 by espressif-bot

AxelLin, 2024-02-22:
Just remind the WIFI_EVENT_HOME_CHANNEL_CHANGE is also available in v5.2 branch now, which also needs fix.

zhangyanjiaoesp, 2024-02-22:
> Just remind the WIFI_EVENT_HOME_CHANNEL_CHANGE is also available in v5.2 branch now, which also needs fix.

Yes, we will backport it to v5.2, thanks for your reminder