espressif/esp-idf #13052: Any ways to encrypt a sensitive free format file by the NVS encryption Key?

emesamwong, 2024-01-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, 
I want to flash a file into the NVS regions externally. 
Since the NVS_ENCRYPTION is enabled, it requires an NVS Encryption key and using the Key to encrypt the file.
Any Method or tool can use the **NVS Encryption key** to **encrypt** the **external file** and decrypt the file using API??

0xjakob, 2024-01-26:
@emesamwong Are you talking about an actual "file" or is it just some data that you want to store? How large (order of magnitude) would your data be, and how often would it be changed?

emesamwong, 2024-01-26:
hi, @0xjakob thank you for your reply
Actually, it is just a <100 bytes of binary data. It would be written to the NVS one time only and would not be edited anymore.

rrtandler, 2024-01-29:
Hi, @emesamwong  - Have you considered NVS encryption as described in [this section](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_encryption.html) of documentation ?

AdityaHPatwardhan, 2024-02-01:
Hi @rrtandler Can you please use [NVS Partition generator utility](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_partition_gen.html#nvs-partition-generator-utility)
for generating the encrypted partition.
You can see the [flash_encryption](https://github.com/espressif/esp-idf/tree/master/examples/security/flash_encryption) example which also showcases the NVS encryption feature.

igrr, 2024-02-01:
Specifically, there is a `file` type which you can use in CSV to reference data from another file.

AdityaHPatwardhan, 2024-02-20:
Hi @rrtandler the issue should be resolved with above information.
I am closing the issue.
Feel free to re-open it if further information is required.
Thanks,
Aditya

closed 2024-02-20 by AdityaHPatwardhan