espressif/esp-idf #13054: Undefined CCC behavior on writeless HID devices

HelloOO7, 2024-01-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.2

### Espressif SoC revision.

ESP32-S3-WROOM-1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

When using the IDF's ble_hid component, creating an HID device that only sends notifications/indications and does not receive explicit CCC writes (or can receive writes, but none are issued by the host device) should be able to send the notifications regardless of whether the host device issues a write that would confirm its CCC value.

### What is the actual behavior?

Currently, the value of `hidd_report_item_t.ccc` is _uninitialized_ before the host device issues a CCC write, which sets the field to a consistent value ([here](https://github.com/espressif/esp-idf/blob/b3f7e2c8a4d354df8ef8558ea7caddc07283a57b/components/esp_hid/src/ble_hidd.c#L532)). I have observed 3 possible behaviors that this can cause:

1. On the `esp_hid_device` with no modifications made, and every other scenario where `esp_hid_parse_report_map` returns a result with `reports_len` of 1, everything works OK. It appears that this is because in my current environment, the `malloc` call [here](https://github.com/espressif/esp-idf/blob/b3f7e2c8a4d354df8ef8558ea7caddc07283a57b/components/esp_hid/src/ble_hidd.c#L701) returns a pointer to uninitialized data in which the `ccc` field has the bits required for notification/indications randomly set.
2. On the same example, if I change the previously mentioned `malloc` to a `calloc`, thus initializing the `ccc` field with a zero, every subsequent attempt to notify/indicate fails with `BLE_HIDD: Indicate Not Enabled: 0`.
3. On other environments, if `esp_hid_parse_report_map` returns a result with `reports_len` greater than 1, chances are that the malloc will also return a pointer to a memory area that is zeroed out, resulting in a similar behavior to case 2.

In turn, unless the host device issues a CCC write, the value of the `hidd_report_item_t.ccc` field is undefined and often wrongly disables notifications/indications.

### Steps to reproduce.

1. Use the `esp_hid_device` example.
2. Change [this](https://github.com/espressif/esp-idf/blob/b3f7e2c8a4d354df8ef8558ea7caddc07283a57b/components/esp_hid/src/ble_hidd.c#L701) to a `calloc` to ensure consistent memory behavior.
3. Connect to the example device from an Android host, then disconnect (this is needed because on first connection, the Android host appears to be issuing a CCC write, which corrects the behavior).
4. Reset the board.
5. Connect to the example device from the same host again (the device must be paired from step 3).
6. The ESP should now report `BLE_HIDD: Indicate Not Enabled` every time it tries to indicate to the host device.

### Debug Logs.

```plain
(410) HID_DEV_DEMO: setting hid gap, mode:1
(420) BLE_INIT: BT controller compile version [59725b5]
(420) BLE_INIT: Bluetooth MAC: 34:85:18:XX:XX:XX
(420) phy_init: phy_version 620,ec7ec30,Sep  5 2023,13:49:13
(490) BT_BTM: BTM_BleWriteAdvData, Partial data write into ADV
(490) HID_DEV_DEMO: setting ble device
(500) main_task: Returned from app_main()
(500) HID_DEV_BLE: START
(1810) HID_DEV_BLE: CONNECT
(2080) ESP_HID_GAP: BLE GAP AUTH SUCCESS
(2080) HID_DEV_DEMO: Send the volume
(2080) BLE_HIDD: Indicate Not Enabled: 0
(2180) BLE_HIDD: Indicate Not Enabled: 0
```

zhp0406, 2024-01-29:
@HelloOO7 
thanks for reporting the issue,I will check it.

mitchellcairns, 2025-02-03:
This is still an issue. I'm running into this and it's hindering my use of BLE HID

mitchellcairns, 2025-02-03:
@HelloOO7 

I found a solution for this particular problem by adding this code block just below: https://github.com/espressif/esp-idf/blob/c5865270b50529cd32353f588d8a917d89f3dba4/components/esp_hid/src/ble_hidd.c#L472-L477

```C
// ADD THIS BLOCK: Initialize CCC for all reports
            for (int i = 0; i < dev->devices[device_index].reports_len; i++) {
                hidd_le_report_item_t *map = &dev->devices[device_index].reports[i];
                if (map->ccc_handle && !map->ccc.notify_enable && !map->ccc.indicate_enable) {
                    ESP_LOGW(TAG, "Initializing CCC for report %d", map->report_id);
                    map->ccc.notify_enable = true;
                    map->ccc.indicate_enable = true;
                    map->ccc.value = 0x03; // Enable notifications and indications
                }
            }
```

HelloOO7, 2025-02-03:
Oh, I have already fixed it myself as well, albeit in another way, because I was not allowed to modify the IDF sources:
```C++
void esp_hid_bugfix_apply(esp_hidd_dev_t* device) {
    esp_hidd_dev_s_hack* device_ = (esp_hidd_dev_s_hack*)device;
    if (device_->transport == ESP_HID_TRANSPORT_BLE) {
        esp_ble_hidd_dev_s_hack* bleDev = (esp_ble_hidd_dev_s_hack*)device_->dev;
        
        for (size_t d = 0; d < bleDev->devices_len; d++) {
            for (size_t r = 0; r < bleDev->devices[d].reports_len; r++) {
                if (bleDev->devices[d].reports[r].report_type == ESP_HID_REPORT_TYPE_INPUT) {
                    bleDev->devices[d].reports[r].ccc.notify_enable = true;
                }
            }
        }
    }
}
```
However, I'd like to see this resolved as well as it's not very practical to have to update this hack with each new IDF revision.

mitchellcairns, 2025-02-03:
I completely agree with you. There should be a stack-level fix provided by Espressif, or a patch supplied so it can be resolved definitively. Just providing another solution here also helps others who may come across this issue later.

favoritewky, 2025-07-07:
How can i fix this in the example?Where should I call this?

> Oh, I have already fixed it myself as well, albeit in another way, because I was not allowed to modify the IDF sources:
> 
> void esp_hid_bugfix_apply(esp_hidd_dev_t* device) {
>     esp_hidd_dev_s_hack* device_ = (esp_hidd_dev_s_hack*)device;
>     if (device_->transport == ESP_HID_TRANSPORT_BLE) {
>         esp_ble_hidd_dev_s_hack* bleDev = (esp_ble_hidd_dev_s_hack*)device_->dev;
>         
>         for (size_t d = 0; d < bleDev->devices_len; d++) {
>             for (size_t r = 0; r < bleDev->devices[d].reports_len; r++) {
>                 if (bleDev->devices[d].reports[r].report_type == ESP_HID_REPORT_TYPE_INPUT) {
>                     bleDev->devices[d].reports[r].ccc.notify_enable = true;
>                 }
>             }
>         }
>     }
> }
> 
> However, I'd like to see this resolved as well as it's not very practical to have to update this hack with each new IDF revision.

favoritewky, 2025-07-07:
with single report id like:
```
  0x05, 0x01,        // Usage Page (Generic Desktop)
        0x09, 0x04,        // Usage (Joystick)
        0xA1, 0x01,        // Collection (Application)
        0x85, 0x01,        //   Report ID (1)
        // Buttons (80 buttons, 1 bit each, 10 bytes)
        0x05, 0x09,        //   Usage Page (Button)
        0x19, 0x01,        //   Usage Minimum (Button 1)
        0x29, 0x50,        //   Usage Maximum (Button 80)
        0x15, 0x00,        //   Logical Minimum (0)
        0x25, 0x01,        //   Logical Maximum (1)
        0x75, 0x01,        //   Report Size (1 bit)
        0x95, 0x50,        //   Report Count (80)
        0x81, 0x02,        //   Input (Data, Variable, Absolute)
        // Padding to align to 16 bytes (80 bits = 10 bytes, add 6 bytes padding)
        0x75, 0x08,        //   Report Size (8 bits)
        0x95, 0x06,        //   Report Count (6)
        0x81, 0x03,        //   Input (Constant)
        // Analog Axes (X, Y, Z, Rx, 8 bits each, 4 bytes)
        0x05, 0x01,        //   Usage Page (Generic Desktop)
        0x09, 0x30,        //   Usage (X)
        0x09, 0x31,        //   Usage (Y)
        0x09, 0x32,        //   Usage (Z)
        0x09, 0x33,        //   Usage (Rx)
        0x15, 0x00,        //   Logical Minimum (0)
        0x25, 0xFF,        //   Logical Maximum (255)
        0x75, 0x08,        //   Report Size (8 bits)
        0x95, 0x04,        //   Report Count (4)
        0x81, 0x02,        //   Input (Data, Variable, Absolute)
        0xC0,              // End Collection
```  it works well , but with addition reportid like 
```
   0x06, 0xFF, 0xFF, // Usage Page (Vendor Defined)
        0x09, 0x01,        // Usage (Vendor Defined Usage)
        0xA1, 0x01,        // Collection (Application)
        0x85, 0x02,        //   Report ID (2)
        0x75, 0x08,        //   Report Size (8 bits)
        0x95, 0x40,        //   Report Count (64)
        0x91, 0x02,        //   Output (Data, Variable, Absolute)
        0xC0               // End Collection
```
`BLE_HIDD: Indicate Not Enabled: 0` will happend.

HelloOO7, 2025-07-07:
> Where should I call this

Once after device initialization, in my case:
```C++
ESP_ERROR_CHECK(esp_hidd_dev_init(&m_Config, m_Transport, detail::HIDDevEventCallback, &m_Device));
esp_hid_bugfix_apply(m_Device);
```

Given that the HID structures are mostly private, it's needed to create independent public definitions, which is the purpose od the _hack types in my workaround.

favoritewky, 2025-07-07:
Thank you!
Does hid device can only use one report id? It will not happend with only one report id.

---- Replied Message ----
| From | ***@***.***> |
| Date | 07/07/2025 16:54 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [espressif/esp-idf] Undefined CCC behavior on writeless HID devices (IDFGH-11980) (Issue #13054) |
HelloOO7 left a comment (espressif/esp-idf#13054)

Where should I call this

Once after device initialization, in my case:

ESP_ERROR_CHECK(esp_hidd_dev_init(&m_Config, m_Transport, detail::HIDDevEventCallback, &m_Device));
esp_hid_bugfix_apply(m_Device);

Given that the HID structures are mostly private, it's needed to create independent public definitions, which is the purpose od the _hack types in my workaround.

HelloOO7, 2025-07-07:
It depends - the bug is undefined behavior. For me, it just so happened to work OK with one report ID. But my use case had multiple (different host OSs/drivers handle these differently), which this hack should fix.

esp-zhp, 2025-07-10:
Hi @HelloOO7 and all,

Thank you again for your detailed feedback and your contributions to identifying and working around this issue.

We've prepared a patch (`ISSUE13054.patch`) to address the undefined CCC behavior on writeless HID devices. This patch is based on **ESP-IDF v5.1.2** and can be applied to your local project using either of the following commands:

```bash
git am ISSUE13054.patch
```

or

```bash
git apply ISSUE13054.patch
```

---
We have addressed the issue you reported regarding the loss of the HID module’s CCC (Client Characteristic Configuration) subscription state. The improvements we implemented are as follows:

Persistent Storage of CCC State in NVS:
When the client writes a CCC value, it is now saved persistently in the non-volatile storage (NVS) using the client’s MAC address as the key, ensuring that each device’s subscription state is stored separately.

Automatic Restoration of CCC State on Reconnect or Device Restart:
Upon device reboot or client reconnection, the stored CCC subscription state is automatically restored, preventing loss of Notify/Indicate functionality without the need for manual re-subscription.

Added nvs_flash Component Dependency:
The build configuration now includes a dependency on the nvs_flash component to support this feature properly.

We have completed internal verification of this fix. Could you please help test the following scenarios on your side?

After device reboot, verify whether the client’s Notify/Indicate subscriptions are automatically restored without manual intervention.

Test with multiple clients connected and confirm that each client’s subscription states are restored correctly.

Observe if there are any other related issues or abnormal behaviors during your typical usage.

Please share any relevant logs or issues you encounter during testing, and we will continue to optimize the solution accordingly.

Thank you very much for your ongoing support and cooperation. Looking forward to your feedback.
[ISSUE13054.patch](https://github.com/user-attachments/files/21162060/ISSUE13054.patch)

HelloOO7, 2025-07-10:
Thank you for the patch. I have tested the changes on devices running Android 13, Windows 8.1 and Windows 11 and verified that all of them write the "notify" bit (0x1), which is correctly restored from NVS on reset/reconnect.

The only issue I noticed was that when pairing the ESP from the Windows 11 machine, then unpairing and attempting to pair again, it would get stuck on "BTM_SetEncryption busy, enqueue request". Resetting the board fixed this. I do not think this is related to the CCC issue though and may as well be a problem on my end.