espressif/esp-idf #13055: [BLE] bluetooth randomly ending connections; GAP disconnect; reason: 0x0216 hci, CONN_TERM_LOCAL

chipweinberger, 2024-01-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Espressif SoC revision.

ESP32-s3

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3

### Power Supply used.

USB

### What is the expected behavior?

The BLE connection should stay forever

### What is the actual behavior?

after some amount of time (few minutes), the ESP32-S3 ends the connection

### Steps to reproduce.

I am use Nimble

### Debug Logs.

```plain
I (1234713) pd ble: GAP disconnect; reason: 0x0216 hci, CONN_TERM_LOCAL     gapConnHandle: 1
I (1234713) pd ble: Connection Description:
I (1234713) pd ble:   role: Slave
I (1234713) pd ble:   gap connection handle: 1
I (1234713) pd ble:   our id:  Public 10:64:77:65:62:71
I (1234713) pd ble:   our ota: Public 10:64:77:65:62:71
I (1234713) pd ble:   peer id:  Public f8:4d:89:7c:1c:7c
I (1234713) pd ble:   peer ota: Public f8:4d:89:7c:1c:7c
I (1234723) pd ble:   connection interval:     12 (15.000ms)
I (1234723) pd ble:   connection latency:      0 times
I (1234723) pd ble:   supervision timeout:     72 (720ms)
I (1234723) pd ble:   Secure - encrypted:      0
I (1234723) pd ble:   Secure - authenticated:  0
I (1234723) pd ble:   Secure - bonded:         0
I (1234733) pd ble:   Secure - keySize:        0
```

chipweinberger, 2024-01-26:
What can I do to debug this? thanks.

chipweinberger, 2024-01-26:
@rahult-github @darshandobariya7

chipweinberger, 2024-01-30:
note: i have wifi enabled in hotspot mode, but no clients connected.

darshan7patel, 2024-01-30:
Hi @chipweinberger ,
I'm having trouble to reach this point, in the meantime could you kindly share the nimble debug enabled logs to figure out who terminated the connection?

darshan7patel, 2024-02-09:
Hii @chipweinberger ,
Any update ?

chipweinberger, 2024-02-09:
The problem has not happened for a awhile. I'll close this issue for now. 

If the problem happens again, I'll get more logs.

Thanks for asking.

closed 2024-02-09 by chipweinberger