espressif/esp-idf #13057: Thread Commissioner stuck in petitioning mode

luisyague, 2024-01-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

**Issue Summary:**
Hi, we are initiating the development of a production-ready Thread Border Router device using an ESP32-C6 (ESP32-C6-DevKitC-1-N8) with the esp-idf version release/v5.2. In our testing phase, we encountered several issues, with the primary one being the inability to start the border router from our application layer.

**Details:**
We began our development by utilizing the example of the border router provided by esp-idf in its release/5.2 version. (Source code attached below.) While the CLI interface works well during tests, we face challenges when attempting to start the commissioner from our application layer.

We observed that the commissioner functions as expected when using the CLI mode (reaching **OT_COMMISSIONER_STATE_ACTIVE**). However, when starting the commissioner (**otCommissionerStart**) directly from our code without using the _"commissioner start"_ command through the CLI, it remains stuck in the **OT_COMMISSIONER_STATE_PETITION** state.

During debugging, we noticed that the CLI passes a pointer to the caller class's context (this) to the "otCommissionerStart" command. Although we attempted to run the command directly from a custom C++ class to pass the "this" pointer, the commissioner still doesn't reach the **OT_COMMISSIONER_STATE_ACTIVE** state. Additionally, we investigated the event firing mechanism, leading us to the "HandleLeaderPetitionResponse" function, which is not called when starting the commissioner from our code.

**Questions:**
1. Is the esp-idf production-ready for Thread devices? While the provided examples work well with the CLI, which is suitable for development and debugging, we are concerned about the production readiness.
   
2. Is it possible to initiate the commissioner from user-application layers? We attempted different approaches, but none seem to achieve the desired state.

Your insights and assistance in resolving this matter would be greatly appreciated. Thank you.

**Source code**

```c
#include <stdio.h>
#include <string.h>
#include <stdint.h>
#include <stdlib.h>

#include "esp_check.h"
#include "esp_coexist.h"
#include "esp_err.h"
#include "esp_event.h"
#include "esp_log.h"
#include "esp_mac.h"
#include "esp_netif.h"
#include "esp_netif_ip_addr.h"
#include "esp_netif_net_stack.h"
#include "esp_netif_types.h"
#include "esp_openthread.h"
#include "esp_openthread_border_router.h"
#include "esp_openthread_cli.h"
#include "esp_openthread_lock.h"
#include "esp_openthread_netif_glue.h"
#include "esp_openthread_types.h"
#include "esp_ot_cli_extension.h"
#include "esp_ot_config.h"
#include "esp_ot_wifi_cmd.h"
#include "esp_vfs_dev.h"
#include "esp_vfs_eventfd.h"
#include "esp_wifi.h"
#include "mdns.h"
#include "nvs_flash.h"
#include "protocol_examples_common.h"
#include "sdkconfig.h"
#include "driver/uart.h"
#include "hal/uart_types.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "freertos/portmacro.h"
#include "openthread/dataset.h"
#include "openthread/dataset_ftd.h"
#include "openthread/dataset_updater.h"
#include "openthread/border_router.h"
#include "openthread/cli.h"
[... 243 lines cut ...]

void mdnsInit(void) {
  ESP_ERROR_CHECK(mdns_init());
  ESP_ERROR_CHECK(mdns_hostname_set("esp-ot-br"));
}

void app_main(void)
{
  eventfdInit();
  flashInit();

  ESP_ERROR_CHECK(esp_netif_init());
  ESP_ERROR_CHECK(esp_event_loop_create_default());

  ESP_ERROR_CHECK(example_connect());
  ESP_ERROR_CHECK(esp_wifi_set_ps(WIFI_PS_MIN_MODEM));
  ESP_ERROR_CHECK(esp_coex_wifi_i154_enable());
  esp_openthread_set_backbone_netif(get_example_netif());
  // esp_ot_wifi_netif_init();
  // esp_openthread_set_backbone_netif(esp_netif_get_handle_from_ifkey("WIFI_STA_DEF"));

  mdnsInit();
  xTaskCreate(otTaskWorker, "ot_br_main", 20480, xTaskGetCurrentTaskHandle(), 5, NULL);
}
```

zwx1995esp, 2024-01-29:
Hi, @luisyague Thansk for your detailed description. I see in your application, you create another task to operate some OPENTHREAD APIs. You need to acquire the ot lock before you call any openthread APIs and after operating that, you need to release the ot lock. Please refer to this API reference: [esp_openthread_lock_acquire](https://github.com/espressif/esp-idf/blob/master/components/openthread/include/esp_openthread_lock.h#L35)(From line 35 to line 54).

Actually the instance which returned from the API `esp_openthread_get_instance` is a global variable, when access this variable in different tasks, we need to protect it using the locks. So could you please check the logic in your application, add the ot lock acquired and released, then try again.

If you have any other questions, please feel free to let me know.

zwx1995esp, 2024-01-29:
For example, the function you defined:
```
static void otLoopTask(void *aContext) {
  otInstance* instance = esp_openthread_get_instance();

  while(otThreadGetDeviceRole(instance) == OT_DEVICE_ROLE_DISABLED || otThreadGetDeviceRole(instance) == OT_DEVICE_ROLE_DETACHED) {
    vTaskDelay(2000 / portTICK_PERIOD_MS);
  }

  otCommissionerStart(esp_openthread_get_instance(), &handleStateChanged, &handleJoinerEvent, NULL);
  while(otCommissionerGetState(instance) != OT_COMMISSIONER_STATE_ACTIVE) {
    vTaskDelay(2000 / portTICK_PERIOD_MS);
  }
  ESP_LOGI(TAG, "Commissioner started");
  otCommissionerAddJoiner(esp_openthread_get_instance(), NULL, "JOINME" ,60);

  ESP_LOGI(TAG, "Joiner add started");
  for (;;) {
    vTaskDelay(1000 / portTICK_PERIOD_MS);
  }
}
```

This should be like:
```
static void otLoopTask(void *aContext) {
  // Before acquire the openthread instance, you need to take the lock first.
  esp_openthread_lock_acquire(portMAX_DELAY);
  otInstance* instance = esp_openthread_get_instance();

  while(otThreadGetDeviceRole(instance) == OT_DEVICE_ROLE_DISABLED || otThreadGetDeviceRole(instance) == OT_DEVICE_ROLE_DETACHED) {
    vTaskDelay(2000 / portTICK_PERIOD_MS);
  }

  otCommissionerStart(esp_openthread_get_instance(), &handleStateChanged, &handleJoinerEvent, NULL);
  while(otCommissionerGetState(instance) != OT_COMMISSIONER_STATE_ACTIVE) {
    vTaskDelay(2000 / portTICK_PERIOD_MS);
  }
  ESP_LOGI(TAG, "Commissioner started");
  otCommissionerAddJoiner(esp_openthread_get_instance(), NULL, "JOINME" ,60);
  // then give the lock here when the process finished.
  esp_openthread_lock_release();

  ESP_LOGI(TAG, "Joiner add started");
  for (;;) {
    vTaskDelay(1000 / portTICK_PERIOD_MS);
  }
}
```

zwx1995esp, 2024-01-29:
For your commissioning logic, I think this should be more reasonable:

Add a state changed callback for detecting the roll change, if the role changed to `OT_DEVICE_ROLE_DISABLED` or `OT_DEVICE_ROLE_DETACHED`, start the commissioner. 

```
static void handle_ot_role_change(otInstance* instance)
{
    otDeviceRole role = otThreadGetDeviceRole(instance);
    if (role == OT_DEVICE_ROLE_DISABLED || role == OT_DEVICE_ROLE_DETACHED) {
        otCommissionerStart(esp_openthread_get_instance(), &handleStateChanged, &handleJoinerEvent, NULL);
    }
}
// then call this API to register this callback, in the otTaskWorker, anywhere after calling the function `esp_openthread_init(&config))`
otSetStateChangedCallback(instance, handle_ot_role_change, NULL);
```
And for the callback `handleStateChanged` of the `otCommissionerStart`, you can add joiner when the commissioner state changed to `active`:
```
static void handleStateChanged(otCommissionerState aState, void *aContext) {
  ESP_LOGI(TAG, "Commissioner state changed");
  switch (aState) {
    case OT_COMMISSIONER_STATE_DISABLED:
      ESP_LOGI(TAG, "New state: Disabled");
      break;
    case OT_COMMISSIONER_STATE_PETITION:
      ESP_LOGI(TAG, "New state: Starting");
      break;
    case OT_COMMISSIONER_STATE_ACTIVE:
      ESP_LOGI(TAG, "New state: Active");
      otCommissionerAddJoiner(esp_openthread_get_instance(), NULL, "JOINME" ,60);
      break;
    default:
      ESP_LOGI(TAG, "New state: Unknown");
      break;
  }
}
```
For this logic, no need for a new task, and all APIs should be called in openthread task, also no need to consider if the lock should be acquired or released.

But if you really want a new task for calling the openthread APIs, the ot lock is needed for protecting the global variable `ot instance`.

zwx1995esp, 2024-01-29:
And for your two question:
    Q1.Is the esp-idf production-ready for Thread devices? While the provided examples work well with the CLI, which is suitable for development and debugging, we are concerned about the production readiness.

    A1. Yes, the esp-idf is production-ready for Thread devices, the reason why your application does not work but the the CLI works well is that you did not using the lock mechanism when calling the functions in other task. No more concerned about the production readiness.
    
    2.Is it possible to initiate the commissioner from user-application layers? We attempted different approaches, but none seem to achieve the desired state.
    A2. Please see the comment above.

luisyague, 2024-01-29:
Thanks for the report! Added the lock_acquire and lock release around the openthread api calls and works fine. I close the issue.

closed 2024-01-29 by luisyague

zwx1995esp, 2024-01-29:
> Thanks for the report! Added the lock_acquire and lock release around the openthread api calls and works fine. I close the issue.

Sure, but I recommend that you use this logic:

> For your commissioning logic, I think this should be more reasonable:
>
> Add a [state changed callback](https://github.com/openthread/openthread/blob/main/include/openthread/instance.h#L262) for [...]