espressif/esp-idf #13058: Coexistence BLE/WiFi: slow BLE Throughput and GATT Error 133 in ESP-IDF 5.1.2

MDB80, 2024-01-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

Release 5.1.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

**Environment:**
Hardware: ESP32-WROVER-E(M213EH2864PH3Q0)
Firmware/SDK: ESP-IDF 5.1.2

**Description:**
I've encountered a significant issue in the latest release of ESP-IDF (5.1.2) affecting the coexistence of BLE and WiFi functionalities. This issue was not present in the previous version (5.1.1).

**Issue Details:**
- In ESP-IDF version 5.1.1, BLE and WiFi coexisted without any noticeable performance issues.
- After updating to ESP-IDF 5.1.2, without any changes to the code, the BLE (Bluetooth Low Energy) performance drastically degrades when WiFi is active.
- The BLE throughput becomes extremely slow, and occasionally the BLE client disconnects with a GATT error 133.
- This problem does not occur when WiFi is inactive.

**Expected Behavior:**
The BLE should function with similar performance and stability as in ESP-IDF 5.1.1, even when WiFi is active.

ESP-IDF 5.1.1 (see timelines)
[image]

BLE-SPIFFS command received = 6 packets * 512Bytes Read/Write

### What is the actual behavior?

- Significantly slower BLE throughput when WiFi is active.
- Occasional disconnection of BLE client with a GATT error 133.

ESP-IDF 5.1.2 (see timelines)
[image]

BLE-SPIFFS command received = 6 packets * 512Bytes Read/Write

### Steps to reproduce.

1. Use ESP32 with ESP-IDF 5.1.2.
2. Implement a BLE GATT Bluedroid service and enable WiFi.
3. Observe the performance and stability of BLE when transferring large amounts of data (e.g., 1KB).

Do the same thing again with 5.1.1!

### More Information.

- No changes were made to the BLE implementation code between ESP-IDF versions 5.1.1 and 5.1.2.
- The issue seems to be specifically related to the coexistence of BLE and WiFi in the 5.1.2 release.
- Clients: tried on Android / Ios and Windows.

MDB80, 2024-01-26:
Dear Support Team,

Thank you for your response, but I'm afraid there has been a misunderstanding. My issue specifically pertains to **Bluedroid** in ESP-IDF version 5.1.2, not **NimBLE !!!!!**. Unfortunately, the solution you provided does not apply to my situation.

I have invested time in thoroughly documenting the problem and would greatly appreciate it if you could revisit my report with this information in mind. A response more targeted to the Bluedroid issue would be highly appreciated.

Thank you in advance for your attention to this matter and I look forward to a more specific follow-up.

[image]

The problem remains the same, also with NimBLE in 'master'.

Have more respect for people who are trying to cooperate!

zhp0406, 2024-01-29:
@MDB80 
Hi,thanks for report the issue.
According to your description, version 5.1.1 is normal, and version 5.1.2 has issues, correct? Can you use the git bisect command to identify the first problematic commit? This will be beneficial for our analysis.

MDB80, 2024-01-30:
Hi @zhp0406,
after extensive investigation, I attempted to use git bisect to identify the source of the issues I was experiencing with version 5.1.2, but it was challenging to pinpoint the exact commit responsible. However, through careful analysis, I realized that the problem was not solely with WiFi, but rather a combination of three functions: WiFi, BLE GATT Server, and BLE GAP Client scanning.

In version 5.1.1, these functionalities worked seamlessly together, but changes introduced in 5.1.2 seemed to disrupt my code. Initially, I suspected it might be a priority issue causing slow data transfer. To investigate further, I revisited the BLE client example in 5.1.2 and adjusted the intervals to match those in the example. Surprisingly, this resolved the issue.

[image]

I'm not entirely sure why this adjustment fixed the problem, but I wanted to share this finding with your team. It might be beneficial to look into how these changes between 5.1.1 and 5.1.2 could impact other projects using a similar combination of functionalities.

For now, the problem seems to be solved. However, I thank you for your attention.

closed 2024-01-30 by MDB80