espressif/esp-idf #13059: ESP32-C6 lp-core interrupts

txf-, 2024-01-26:
### Is your feature request related to a problem?

Currently the C6 lp core can only poll for GPIO reads. Ideally, I'd like to use the LP core to count pulses while the main core is in various modes of sleep, both light and deep. 

### Describe the solution you'd like.

It would be nice if "ulp_lp_core_gpio.h" had functions that supported gpio interrupts on the LP GPIOs.

### Describe alternatives you've considered.

I'm currently using the Pulse counter for this function. However I don't know if the pulse counter will run while the main core is in light sleep. 

If the pulse counter does also work in light 

### Additional context.

If the pulse counter does also work in light sleep, I'd be interested in comparing which solution consumes less power. With the LP-core I wouldn't have to even have to wake up the main CPU to reset the count of pulses for second by second readings, I could fill a buffer of readings and only wake the main core when the buffer was full.

0xjakob, 2024-01-30:
@txf- We'll take a look at this issue soon and come back with an answer.

sudeep-mohanty, 2024-01-31:
Hi @txf- Thank you for approaching us with this feature request. We plan to support LP-Core interrupts in the near-term and hopefully have GPIO interrupts covered as well.

As for your question regarding the current consumption of a pulse counter, it really depends on the frequency of the pulses that you intend to count and how often would the CPU (main CPU or LP core) need to wakeup to perform this activity. Having the solution on either of the cores is possible and you would need to make a quantitative analysis of the current consumption in either cases. We don't have a pulse counter example yet for the RISC-V based ULP/LP-core but we do have one for the ULP-FSM (ESP32/S2/S3) for reference [here](https://github.com/espressif/esp-idf/tree/master/examples/system/ulp/ulp_fsm/ulp). We plan to extend this example to the other types of ULPs as well.

txf-, 2024-01-31:
Thank you for the detailed reply

>  how often would the CPU (main CPU or LP core) need to wakeup to perform this activity. Having the solution on either of the cores is possible and you would need to make a quantitative analysis of the current consumption in either cases. 

Yes. The device will see bursts of activity, but will spend most of its time sleeping. Ultimately, during active periods, the LP core would not be sleeping, but counting interrupts over a period of 1 -> 10 seconds. And after each period it should wake the main core from light sleep to send that data, whereupon the main core would return to light sleep.

The paradigm should be the same with the pulse counter. But I'm still unclear if the pulse counter can actively accumulate pulses whilst the main core is in light sleep (I believe the PCNT still does not use the new power management framework).

Ultimately what I'm looking for here is a which of these solutions would be of the lowest power consumption.

igrr, 2024-01-31:
> But I'm still unclear if the pulse counter can actively accumulate pulses whilst the main core is in light sleep (I believe the PCNT still does not use the new power management framework)

Regarding this part: in light sleep mode, the peripherals, memory and the CPU are clock-gated, so it's only possible to use the LP peripherals and the LP core.

txf-, 2024-01-31:
So really the only viable solution to accumulate pulses during sleep is with the LP core. Thank you.

closed 2024-05-16 by espressif-bot