espressif/esp-idf #13061: Support custom printf format via `register_printf_specifier(...)`

hyansuper, 2024-01-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

In normal C program, I can include <printf.h> and use `register_printf_specifier(...)` to add costum print format.

But in ESP-IDF, it says <printf.h> "no such file" on compile

0xjakob, 2024-01-29:
@hyansuper I'm afraid the libc used in ESP-IDF, *Newlib*, does not provide this API. This API is specific to glibc.

However, what you can do is [register a custom printf function](https://github.com/espressif/esp-idf/blob/master/components/log/include/esp_log.h#L113) which is called by the [ESP_LOG](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/log.html) macros.

Why would you need custom formatters in the first place, btw?

hyansuper, 2024-01-29:
Thank you.
I am trying to print some data types(date, time, hex color, etc) to json string, being able to customize would make use of sprintf more comfortable.
Of course I can do it in another way(define format macro)

closed 2024-01-29 by hyansuper