espressif/esp-idf #13062: new i2c master driver hangs with no pullup on SCL/SDA pins

cpehonk, 2024-01-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2 developer branch (i2c master read function patch)

### Espressif SoC revision.

ESP32 Rev1.0

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

WEMOS LOLIN32 (ESP32 WROVER)

### Power Supply used.

USB

### What is the expected behavior?

Calling i2c_master_new_bus() with GPIO pins not connected to any pullups (flags for internal pullups set to false, too) should return with a valid handle.

### What is the actual behavior?

i2c_master_new_bus() hangs and does not return. Thus, an application cannot continue and hangs as well.

### Steps to reproduce.

Simply call i2c_master_new_bus() with GPIO pins specified which have no pullups.

mythbuster5, 2024-01-29:
I knew this behavior and I don't know anything wrong. That is, you cannot use I2C without any pull-ups (either internal or external). If GPIO pins are not connected, software has no way to know and do some check. What is your use case that this makes you annoy..?

cpehonk, 2024-01-29:
It isn't an use case. I encountered this during development when I specified the pins in wrong order. The old i2c legacy driver did not show this behaviour. So I thought it might be a bug in the implementation. But if pullups are a prerequisite for the new i2c driver model I have no problems with it. In this case I suggest to extent the very good documentation by a short paragraph about the necessary pullups. Otherwise I expect that you will get more messages like mine.

closed 2024-01-29 by cpehonk