espressif/esp-idf #13066: Strange time behaviour after restart

damian-kurek-wizzdev, 2024-01-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have a fleet of around 250k devices (esp32/esp32s3) connected to the internet and synchronized via NTP. Last week I made an update that updated around 40k of them. In this update, I changed esp-idf from 4.4.3 and added a check to assume that the device is synchronized after restart if it stores the time on the device is larger than 01.01.2024. After some time I noticed that for around 20% of the devices(around 8k), timestamps are messed up by up to +/- 8 hours.
When I tested it with a few dev boards + and a few devices I never observed this issue. 
What is the expected behaviour of the system clock when you restart the esp. When restarting esp it is powered on all the time.

I get time using this code:

```
int64_t getCurrentSystemClockTimestampMs()
{
    struct timeval currentTime = {};
    gettimeofday(&currentTime, nullptr);
    int64_t millis = static_cast<int64_t>(currentTime.tv_sec) * 1000 + (currentTime.tv_usec / 1000);
    return millis;
}
```

AxelLin, 2024-03-05:
@damian-kurek-wizzdev
Do you mean it works with esp-idf-4.3.3 but has issue with newer esp-idf version? Which version (git describe --tags output)  is used when the issue happened? 
@KonstantinKondrashov Any update?

damian-kurek-wizzdev, 2024-03-05:
@AxelLin 
I forget to add this 4.4.3 to 5.1.1. I am not sure if it happened on the earlier idf version.

AxelLin, 2024-03-08:
@damian-kurek-wizzdev 

Just want to clarify if the +/- 8 hours is related to timezone.
What is your current timezone setting?

damian-kurek-wizzdev, 2024-03-08:
@AxelLin sorry I did not have time to look into it again, but It wasn't 8 hours flat it was something like 8:23. I asked my devops to get it from the database.  Additionally, sometimes it was 8:xx and sometimes 4:yz

damian-kurek-wizzdev, 2024-03-08:
-54 years was lack of synchronization for some reason (I did a bit later). But there you can see the time difference.
"0f026bbc-958c-4eb7-afb3-3901ad527483"	"-01:11:47.875"
"16c199b2-f9e7-40e4-99d5-f96ea9983eac"	"02:23:47.6"
"17b5e75d-6c56-43ce-bb93-808746163e81"	"-54 years -15 days -05:43:08.2"
"22fe0009-22d3-4338-a387-60424421eab8"	"-01:02:57.9375"
"281bf913-a910-4a5f-81a2-0ec3ab6b3a7c"	"-02:10:41.75"
"39fe62d2-836b-4de7-a5d4-a80cd8db771c"	"-01:07:19.483333"
"42a5cc04-abfe-448c-b7f7-cbcbe25ae3b7"	"-54 years -15 days -04:37:37.863014"
"6f81463e-0f9d-41c4-bce4-6c5e0afeff02"	"-01:08:02.05814"
"a36c32d5-13a6-45e1-a2fb-caf1b4cd97d6"	"-01:44:36.837209"
"b4198da8-8eb1-418d-b153-e48da429eca1"	"-01:04:21.074074"
"ba514624-bf60-4a24-a9ee-63059e93a9b9"	"-54 years -15 days -07:24:28"
"ba55394d-66c6-4cf1-933a-6397caba9a2f"	"-54 years -15 days -04:16:59"
"ccfb39ae-e9bb-42cb-a5aa-94194156fe79"	"03:16:32.678571"
"ce06f6ce-40fc-4eb7-bf64-6ba81af6d980"	"-54 years -15 days -05:49:07.892857"

AxelLin, 2024-06-06:
@KonstantinKondrashov Any feedback about the symptom reported by https://github.com/espressif/esp-idf/issues/13066#issuecomment-1985258949 ?