espressif/esp-idf #13068: Failed to set-target esp32 using esp-idf

S-Code-dot, 2024-01-29:
I am trying to set the target for esp32 but it is giving error by using esp-wrover-kit. but getting some following error.
This log before deleting existing build folder -> 
smollick@MGC1WT7CG7XH light % idf.py set-target esp32
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Directory '/Users/smollick/ESP-Matter/light/build' doesn't seem to be a CMake build directory. Refusing to automatically delete files in this directory. Delete the directory manually to 'clean' it.

Then if try after deleting the build folder and then run " idf.py set-target esp32 " i am seeing the following error 
smollick@MGC1WT7CG7XH light % idf.py set-target esp32
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/Users/smollick/ESP-Matter/light/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32, new sdkconfig will be created.
Running cmake in directory /Users/smollick/ESP-Matter/light/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/Users/smollick/.espressif/python_env/idf5.1_py3.9_env/bin/python -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=1 /Users/smollick/ESP-Matter/light"...
-- Found Git: /usr/bin/git (found version "2.39.3 (Apple Git-145)") 
-- Component directory /Users/smollick/ESP-Matter/esp-matter/examples/common/blemesh_platform does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/smollick/ESP-Matter/esp-matter/examples/common/external_platform does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/smollick/ESP-Matter/esp-matter/examples/common/relinker does not contain a CMakeLists.txt file. No component will be added
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/smollick/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/smollick/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/smollick/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
Dependencies lock doesn't exist, solving dependencies.
NOTICE: Skipping optional dependency: espressif/cmake_utilities
CMake Error at /Users/smollick/ESP-Matter/esp-idf/tools/cmake/build.cmake:540 (message):
  WARNING: HTTPSConnectionPool(host='components-file.espressif.com',
  port=443): Max retries exceeded with url:
  /components/espressif/esp_encrypted_img.json (Caused by
  SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol
  (_ssl.c:1129)')))

  WARNING: HTTPSConnectionPool(host='components-file.espressif.com',
  port=443): Max retries exceeded with url:
  /components/espressif/esp_insights.json (Caused by SSLError(SSLEOFError(8,
  'EOF occurred in violation of protocol (_ssl.c:1129)')))

  WARNING: HTTPSConnectionPool(host='components-file.espressif.com',
  port=443): Max retries exceeded with url:
  /components/espressif/esp_secure_cert_mgr.json (Caused by
  SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol
  (_ssl.c:1129)')))

  WARNING: HTTPSConnectionPool(host='components-file.espressif.com',
  port=443): Max retries exceeded with url: /components/espressif/mdns.json
  (Caused by SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol
  (_ssl.c:1129)')))

  WARNING: HTTPSConnectionPool(host='components-file.espressif.com',
  port=443): Max retries exceeded with url:
  /components/espressif/json_generator.json (Caused by
  SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol
  (_ssl.c:1129)')))

  WARNING: HTTPSConnectionPool(host='components-file.espressif.com',
  port=443): Max retries exceeded with url:
  /components/espressif/json_parser.json (Caused by SSLError(SSLEOFError(8,
  'EOF occurred in violation of protocol (_ssl.c:1129)')))

  WARNING: HTTPSConnectionPool(host='components-file.espressif.com',
  port=443): Max retries exceeded with url:
  /components/espressif/esp-serial-flasher.json (Caused by
  SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol
  (_ssl.c:1129)')))

  WARNING: HTTPSConnectionPool(host='components-file.espressif.com',
  port=443): Max retries exceeded with url:
  /components/espressif/esp_rcp_update.json (Caused by
  SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol
  (_ssl.c:1129)')))

  WARNING: HTTPSConnectionPool(host='components-file.espressif.com',
  port=443): Max retries exceeded with url:
  /components/espressif/led_strip.json (Caused by SSLError(SSLEOFError(8,
  'EOF occurred in violation of protocol (_ssl.c:1129)')))

  WARNING: HTTPSConnectionPool(host='components-file.espressif.com',
  port=443): Max retries exceeded with url: /components/espressif/button.json
  (Caused by SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol
  (_ssl.c:1129)')))

  HINT: Please check manifest file of the following component(s): iot_button

  ERROR: Because project depends on espressif/button (=2.5.0) which doesn't
  match any versions, version solving failed.

Call Stack (most recent call first):
  /Users/smollick/ESP-Matter/esp-idf/tools/cmake/project.cmake:547 (idf_build_process)
  CMakeLists.txt:44 (project)

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1, output of the command is in the /Users/smollick/ESP-Matter/light/build/log/idf_py_stderr_output_9483 and /Users/smollick/ESP-Matter/light/build/log/idf_py_stdout_output_9483

can i get any help to over come the issue..

Thanks

kumekay, 2024-01-30:
Hello @S-Code-dot,
The first idea is the firewall or the network provider may restrict some requests,

Could you please try to run this command to make sure that the certificates are correct:

```openssl s_client -showcerts -servername components-file.espressif.com -connect components-file.espressif.com:443 </dev/null```

The output should be like:

```
Connecting to 65.9.95.86
CONNECTED(00000007)
depth=2 C=US, O=DigiCert Inc, OU=www.digicert.com, CN=DigiCert Global Root G2
verify return:1
depth=1 C=US, O=DigiCert Inc, CN=DigiCert Global G2 TLS RSA SHA256 2020 CA1
verify return:1
depth=0 C=CN, ST=上海市, O=乐鑫信息科技（上海）股份有限公司, CN=*.espressif.com
verify return:1
---
Certificate chain
 0 s:C=CN, ST=上海市, O=乐鑫信息科技（上海）股份有限公司, CN=*.espressif.com
   i:C=US, O=DigiCert Inc, CN=DigiCert Global G2 TLS RSA SHA256 2020 CA1
   a:PKEY: rsaEncryption, 2048 (bit); sigalg: RSA-SHA256
   v:NotBefore: May 13 00:00:00 2023 GMT; NotAfter: Jun 12 23:59:59 2024 GMT
-----BEGIN CERTIFICATE-----
MIIG8DCCBdigAwIBAgIQBxPX3RYiJPDTCuBVSekC0DANBgkqhkiG9w0BAQsFADBZ
MQswCQYDVQQGEwJVUzEVMBMGA1UEChMMRGlnaUNlcnQgSW5jMTMwMQYDVQQDEypE
aWdpQ2VydCBHbG9iYWwgRzIgVExTIFJTQSBTSEEyNTYgMjAyMCBDQTEwHhcNMjMw
NTEzMDAwMDAwWhcNMjQwNjEyMjM1OTU5WjB2MQswCQYDVQQGEwJDTjESMBAGA1UE
CAwJ5LiK5rW35biCMTkwNwYDVQQKDDDkuZDpkavkv6Hmga/np5HmioDvvIjkuIrm
tbfvvInogqHku73mnInpmZDlhazlj7gxGDAWBgNVBAMMDyouZXNwcmVzc2lmLmNv
bTCCASIwDQYJKoZIhvcNAQEBBQADggEPADCCAQoCggEBAOM8dvxz1qOiKGO9b3t+
APqUzQDMYCknwZAEA4MgiVZjIcUN74Z0BSfhEupsvPPAZ8Mc2rGGsd8YuY8Fi+yz
93pISvMNveAHNXejb1HH2bnwRDezdoUG1TVrqVVR6ue5cdOptyXiG3mrpmMcV7Qq
5QFbAyCIqOcin0IuMHAqycZGL8w9UOAa94yO/QCBgTOyeTKnhwOzJr5rB7OUVVvO
ejlK60hw6eLZ8YSwrk6lM73JIOkodT7UMqpzjUpYiohhyTiJldJrP3l3Mre2DIVE
4dgX5Sm3oRoEN0H/E7pey0KxadmjfJaIDpJyKVsNtB0FjN5WMSDcSaIhiwEgcvuI
+y0CAwEAAaOCA5UwggORMB8GA1UdIwQYMBaAFHSFgMBmx9833s+9KTeqAx2+7c0X
MB0GA1UdDgQWBBT+kb9kSRcyvCt6ioiwjxWMdAsp5TApBgNVHREEIjAggg8qLmVz
cHJlc3NpZi5jb22CDWVzcHJlc3NpZi5jb20wDgYDVR0PAQH/BAQDAgWgMB0GA1Ud
JQQWMBQGCCsGAQUFBwMBBggrBgEFBQcDAjCBnwYDVR0fBIGXMIGUMEigRqBEhkJo
dHRwOi8vY3JsMy5kaWdpY2VydC5jb20vRGlnaUNlcnRHbG9iYWxHMlRMU1JTQVNI
QTI1NjIwMjBDQTEtMS5jcmwwSKBGoESGQmh0dHA6Ly9jcmw0LmRpZ2ljZXJ0LmNv
bS9EaWdpQ2VydEdsb2JhbEcyVExTUlNBU0hBMjU2MjAyMENBMS0xLmNybDA+BgNV
HSAENzA1MDMGBmeBDAECAjApMCcGCCsGAQUFBwIBFhtodHRwOi8vd3d3LmRpZ2lj
ZXJ0LmNvbS9DUFMwgYcGCCsGAQUFBwEBBHsweTAkBggrBgEFBQcwAYYYaHR0cDov
L29jc3AuZGlnaWNlcnQuY29tMFEGCCsGAQUFBzAChkVodHRwOi8vY2FjZXJ0cy5k
aWdpY2VydC5jb20vRGlnaUNlcnRHbG9iYWxHMlRMU1JTQVNIQTI1NjIwMjBDQTEt
MS5jcnQwCQYDVR0TBAIwADCCAXwGCisGAQQB1nkCBAIEggFsBIIBaAFmAHUA7s3Q
ZNXbGs7FXLedtM0TojKHRny87N7DUUhZRnEftZsAAAGIFGxK1wAABAMARjBEAiBN
6+4gWafddVfp3rEZ0YgGA5/ehOvqBMG+ZP+0dNkjfQIgELQ1DptL17wZack+mt95
9XxMxFegLdAIYb4MgbXuamMAdgBIsONr2qZHNA/lagL6nTDrHFIBy1bdLIHZu7+r
OdiEcwAAAYgUbEruAAAEAwBHMEUCIE9/x99Cw9JaznglCKFuI4T1qZEWtPC//bYF
Cehfwmo/AiEAu2Wi8cgc9MR8hs4vlnAFA4/sxa/rXRNaPxbzhiNyBKMAdQDatr9r
P7W2Ip+bwrtca+hwkXFsu1GEhTS9pD0wSNf7qwAAAYgUbEq0AAAEAwBGMEQCIEX4
o36umIXUFsZ9eGyuBrot1j00+n98LBQgqemQbAG1AiBTXukDwZOQCLgkiYuAWjgP
9lrGSjJ5+rsQdfQV0A+6ijANBgkqhkiG9w0BAQsFAAOCAQEAUM0maP4T6O+D5En+
cfwnPkwdCugThVT9a0m/ssKsr19vak2f2utwlSt4J25+eKVIcymL9qb4RtgShXCJ
1mZl5GonESeBL4WWqYyF0g0I9yaXgU+g5AbGGpCw+vZccM4sa+CfSfzVWNbISn3c
l5F5V7kRS9Rb22u1YemqReH9fwYhMffew1QKZtrnZxH7aazsfJEOxxtSrAjw5DAA
/3QakoOV1kC1j2VwkQg/Ls6QwDnipi84dQBl/A/GyPu0oQHYPjpNwgVcqxAKOPiP
8Pakkdzg7PeZY1mPIQbk1gFAI1DMFgUQwKo7CQOSvl1D9bVTyxM4cYC6jRqZD/Q4
ALpPgQ==
-----END CERTIFICATE-----
 1 s:C=US, O=DigiCert Inc, CN=DigiCert Global G2 TLS RSA SHA256 2020 CA1
   i:C=US, O=DigiCert Inc, OU=www.digicert.com, CN=DigiCert Global Root G2
   a:PKEY: rsaEncryption, 2048 (bit); sigalg: RSA-SHA256
   v:NotBefore: Mar 30 00:00:00 2021 GMT; NotAfter: Mar 29 23:59:59 2031 GMT
-----BEGIN CERTIFICATE-----
MIIEyDCCA7CgAwIBAgIQDPW9BitWAvR6uFAsI8zwZjANBgkqhkiG9w0BAQsFADBh
[... 25 lines cut ...]
-----END CERTIFICATE-----
---
Server certificate
subject=C=CN, ST=上海市, O=乐鑫信息科技（上海）股份有限公司, CN=*.espressif.com
issuer=C=US, O=DigiCert Inc, CN=DigiCert Global G2 TLS RSA SHA256 2020 CA1
---
No client certificate CA names sent
Peer signing digest: SHA256
Peer signature type: RSA-PSS
Server Temp Key: X25519, 253 bits
---
SSL handshake has read 3557 bytes and written 401 bytes
Verification: OK
---
New, TLSv1.3, Cipher is TLS_AES_128_GCM_SHA256
Server public key is 2048 bit
This TLS version forbids renegotiation.
Compression: NONE
Expansion: NONE
No ALPN negotiated
Early data was not sent
Verify return code: 0 (ok)
---
DONE
```

S-Code-dot, 2024-01-31:
Hi Silnov,
     Thank you so much for the reply.
You are correct for the network privacy only i was facing the issue. Now i
successfully builded the project and working fine.

Thank you

closed 2024-01-31 by kumekay

Alanzalander, 2025-02-09:
I have the same problem but sorry, I don't understand the solution above.
So I tried to use the prompt
openssl s_client -showcerts -servername components-file.espressif.com -connect components-file.espressif.com:443 </dev/null
 in a debug terminal in VScode and in powershell on folder c:\ and the response is

> In Zeile:1 Zeichen:114
> + ... le.espressif.com -connect components-file.espressif.com:443 </dev/nul ...
> +                                                                 ~
> Der Operator "<" ist für zukünftige Versionen reserviert.
>     + CategoryInfo          : ParserError: (:) [], ParentContainsErrorRecordException
>     + FullyQualifiedErrorId : RedirectionNotSupported_

and without the trailing </dev/null it says

> openssl : Die Benennung "openssl" wurde nicht als Name eines Cmdlet, einer Funktion, einer Skriptdatei oder eines ausführbaren Programms erkannt. Überprüfen Sie die
> Schreibweise des Namens, oder ob der Pfad korrekt ist (sofern enthalten), und wiederholen Sie den Vorgang.
> In Zeile:1 Zeichen:1
> + openssl s_client -showcerts -servername components-file.espressif.com ...
> + ~~~~~~~
>     + CategoryInfo          : ObjectNotFound: (openssl:String) [], CommandNotFoundException
>     + FullyQualifiedErrorId : CommandNotFoundException

So where does this openssl command reside?
And if a network setting is to blame, which is it? Is it a setting on my computer or something with my internet provider I cannot change?

Alanzalander, 2025-02-09:
I found the couse that prohibited to set set the target. As long as the project folder is on my NAS drive it doesn't work and shows the error message about a build folder that can't be cleared. Calling "SDK configuration Editor (menuconfig)" takes very long time and does not finish, throwing some cryptic error messages on the way.

All is good as soon as I create the same "blink" project on drive C:, I can set the target and calling "..menuconfig" to set up the variables does also work!!

kumekay, 2025-02-10:
@Alanzalander The command you tried to execute is only for linux or macOS, 

Because this issue is closed long time ago, it's likely that your problem is different. 
If you still have a problem could you please open a new ticket.

To trouble shoot the original problem described in this issue on Windows, you can try these instructions: https://github.com/espressif/idf-component-manager/issues/69#issuecomment-2262721940