espressif/esp-idf #13072: Lacking double-checked locking optimization leads to significantly slower code for static local variables on RISC-V

andylinpersonal, 2024-01-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v5.1 and master

### Espressif SoC revision.

esp32c6: v0.1, esp32c3: v0.4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32c3: custom | esp32c6: esp32-c6-devkitc-1-n8 | esp32s3: custom

### Power Supply used.

USB

### What is the expected behavior?

* C++11 standard requires the thread-safe, on-demand initialization of [local static variables](https://en.cppreference.com/w/cpp/language/storage_duration#Static_local_variables) [N2660](https://wg21.link/N2660), so GCC introduce a guard variable and guard functions to protect the underlying static variable.
  * When first time control passes through their declaration, the guard functions ```__cxa_guard_*```  are called and ensure the successful initialization must be performed exactly once.
  * The guard variable records the current state of the local static variable.
  * Guard functions require some synchronization mechanisms to work, so they're somewhat heavy.
  * GCC introduce another inlined check (double-checked locking optimization) to bypass the guard functions after the local static variable successfully initialized.
* Problem: Some targets may lack the double-checked locking optimization, and the guard functions are always called no matter whether the local static variable is initialized or not.

### What is the actual behavior?

* Accessing to the local static variable on some targets/gcc combination are significantly slower than others.
* After a bit of disassembling, I found the double-checked locking optimization is missing from all RISC-V targets on GCC 12.2, and still missing from RV32IMC targets on GCC 13.2.

* Affected targets:

    | IDF ver.     | GCC ver.                            | C2  | C3  | C6  | H2  | P4  | S3 (XT) |
    | ------------ | ----------------------------------- | --- | --- | --- | --- | --- | ------- |
    | release/v5.1 | riscv32-esp-elf/esp-12.2.0_20230208 | +   | +   | +   | +   | x   | -       |
    | master       | riscv32-esp-elf/esp-13.2.0_20230928 | +   | +   | -   | -   | -   | -       |

  * '+' for affected, '-' for not affected, 'x' for not supported.
  * S3 is an Xtensa target, not affected by this problem. Just for comparison.

* Summary of the implemented ISA extensions for each RISC-V target:

    | Target   | Implemented ISA Ext.     |
    | -------- | ------------------------ |
    | ESP32-C2 | rv32imc_zicsr_zifencei   |
    | ESP32-C3 | rv32imc_zicsr_zifencei   |
    | ESP32-C6 | rv32imac_zicsr_zifencei  |
    | ESP32-H2 | rv32imac_zicsr_zifencei  |
    | ESP32-P4 | rv32imafc_zicsr_zifencei |

  * \-march parameters were taken from the particular toolchain files.

[... 336 lines cut ...]
            auto dur_oh = test_runner<rand_opt_handcraft_local_static>();
            auto dur_nh = test_runner<rand_naive_handcraft_local_static>();
            auto dur_g = test_runner<rand_global_static>();
            ESP_LOGI(TAG, "native local static duration:                 %*u", width, dur_l);
            ESP_LOGI(TAG, "optimized handcrafted local static duration:  %*u", width, dur_oh);
            ESP_LOGI(TAG, "naive handcrafted local static duration:      %*u", width, dur_nh);
            ESP_LOGI(TAG, "global static duration:                       %*u", width, dur_g);
            ESP_LOGI(TAG, "penalty of local static: %.3f%%", (1.0f - (float)dur_g / dur_l) * 100.0f);
            vTaskDelay(pdMS_TO_TICKS(1000));
        }
    }

    extern "C" void app_main()
    {
    #if portNUM_PROCESSORS > 1
        const BaseType_t core = 1;
    #else
        const BaseType_t core = 0;
    #endif
        TaskHandle_t p_tsk;
        assert(xTaskCreatePinnedToCore(&test_task, TAG, 4096, nullptr, CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT, &p_tsk,
                                    core) == pdPASS);
    }

    ```

igrr, 2024-01-29:
> Problem: Some targets may lack the double-checked locking optimization, and the guard functions are always called no matter whether the local static variable is initialized or not.

Unfortunately, I think this is probably expected, since the double-checked lock initialization using an atomic guard variable is only implemented in GCC on targets with support for atomic instructions (i.e. `a` extension on RISC-V):

- Static initialization expansion relies on `get_guard_cond` ([code](https://github.com/gcc-mirror/gcc/blob/f3fdaa3eecd155dbdc78c1ec9a259dfa4e379ea4/gcc/cp/decl2.cc#L4267))
- `get_guard_cond` generates a constant zero expression if `is_atomic_expensive_p` is true ([code](https://github.com/gcc-mirror/gcc/blob/f3fdaa3eecd155dbdc78c1ec9a259dfa4e379ea4/gcc/cp/decl2.cc#L3703-L3706))
- `is_atomic_expensive_p` calls `can_compare_and_swap_p` with the 2nd argument (allow_libcall) set to `false` ([code](https://github.com/gcc-mirror/gcc/blob/f3fdaa3eecd155dbdc78c1ec9a259dfa4e379ea4/gcc/cp/decl2.cc#L3639-L3640))

So if the target doesn't support atomics via instructions (only via library calls) then `is_atomic_expensive_p` will return `true`, and the atomic guard related code won't be generated.

I think this probably needs to be reported in upstream GCC as a "allow atomic libcalls for double-check guard implementation" type of a feature request.

Newer Espressif RISC-V chips should all have the "A" extension, so this probably won't be an issue going forward.

andylinpersonal, 2024-01-30:
Hi igrr:
  But even on the RV32IMAC targets (C6 and onward), it seems like the generated code doesn't utilize any instructions from the A extension at all?

igrr, 2024-01-30:
That's right, the load operation generated by build_atomic_load_type is `lbu`, which isn't an atomic instruction. Since the "initialized" flag is represented by 1 byte, and we don't need to atomically modify the flag, it's not necessary to use `lr`/`sc` instructions there, so seems like the compiler is doing the right thing there.

Unfortunately, I don't know enough about other architectures which GCC targets to tell if there is a specific reason for using `build_atomic_load_type` there. I do see the same behavior for other architectures, though — https://godbolt.org/z/cz5cvabv8 illustrates the same issue on Xtensa. On ESP32-S2 (no atomic instructions) no double-checked locking is used, but on ESP32 (has atomic instructions) it is used.

closed 2024-01-30 by andylinpersonal