espressif/esp-idf #13073: Nimble power_save esp32s3 current at ~15mA

berkutta, 2024-01-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello

I'm a bit confused about power_save in nimble BLE for esp32s3. This example: https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble/power_save

The table states the esp32s3 should take ~230uA in light sleep. I understand, that the esp32s3 needs to transmit BLE packets from time to time, so it won't spend 100% in light sleep. But I guess (I don't know for sure) it should stay a good amount of time in light sleep in this example.

I have built a small testing jig with an ESP32-S3-WROOM-1-N16R8 module. The current is measured directly with a DMM without any LDO or similar. A 32.768kHz oscillator is connected. USB is not connected. No GPIO's are connected.

The example was flashed with `idf.py -D SDKCONFIG_DEFAULTS="sdkconfig.defaults;sdkconfig.defaults.esp32s3" set-target esp32s3 build flash monitor`. When measuring the current while it is running, I measure ~15mA.

Is ~15mA the expected average current in this example? Or should it be way lower? Are there other actions I can take to minimize the current in BLE mode?

Boot Log:
```
I (26) boot: ESP-IDF v5.1.2-dirty 2nd stage bootloader
I (27) boot: compile time Jan 29 2024 23:57:25
W (27) boot: Unicore bootloader
I (30) boot: chip revision: v0.1
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c060020 size=1a4d4h (107732) map
I (119) esp_image: segment 1: paddr=0002a4fc vaddr=3fc99100 size=03628h ( 13864) load
I (122) esp_image: segment 2: paddr=0002db2c vaddr=40374000 size=024ech (  9452) load
I (127) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=59e10h (368144) map
I (199) esp_image: segment 4: paddr=00089e38 vaddr=403764ec size=12bb4h ( 76724) load
I (216) esp_image: segment 5: paddr=0009c9f4 vaddr=600fe000 size=0005ch (    92) load
I (225) boot: Loaded app from partition at offset 0x10000
I (226) boot: Disabling RNG early entropy source...
I (237) cpu_start: Unicore app
I (237) cpu_start: Pro cpu up.
I (237) cpu_start: Single core mode
I (422) cpu_start: Pro cpu start user code
I (422) cpu_start: cpu freq: 160000000 Hz
I (422) cpu_start: Application information:
I (425) cpu_start: Project name:     power_save
I (430) cpu_start: App version:      v5.1.2-dirty
I (436) cpu_start: Compile time:     Jan 29 2024 23:57:17
I (442) cpu_start: ELF file SHA256:  de7af2ce4b34cd1b...
I (448) cpu_start: ESP-IDF:          v5.1.2-dirty
I (453) cpu_start: Min chip rev:     v0.0
I (458) cpu_start: Max chip rev:     v0.99 
I (463) cpu_start: Chip rev:         v0.1
I (468) heap_init: Initializing. RAM available for dynamic allocation:
I (475) heap_init: At 3FC9EA38 len 0004ACD8 (299 KiB): DRAM
I (481) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (488) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (494) heap_init: At 600FE05C len 00001F8C (7 KiB): RTCRAM
I (501) spi_flash: detected chip: gd
I (504) spi_flash: flash io: dio
W (508) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (522) sleep: Configure to isolate all GPIO pins in sleep state
I (528) sleep: Enable automatic switching of GPIO sleep configuration
I (547) coexist: coex firmware version: b6d5e8c
I (547) coexist: coexist rom version e7ae62f
I (548) app_start: Starting scheduler on CPU0
I (551) main_task: Started on CPU0
I (555) main_task: Calling app_main()
I (565) pm: Frequency switching config: CPU_MAX: 160, APB_MAX: 80, APB_MIN: 40, Light sleep: ENABLED
I (569) sleep: Code start at 0x42000020, total 368143, data start at 0x3c060020, total 107732 Bytes
0x42000020: _stext at ??:?

I (579) BLE_INIT: BT controller compile version [59725b5]
I (586) BLE_INIT: Bluetooth MAC: f4:12:fa:f6:89:1a

I (591) phy_init: phy_version 620,ec7ec30,Sep  5 2023,13:49:13
I (631) NimBLE_BLE_PRPH: BLE Host Task Started
I (632) uart: queue free spaces: 8
I (634) main_task: Returned from app_main()
I (647) NimBLE: GAP procedure initiated: stop advertising.

I (649) NimBLE: Device Address: 
I (649) NimBLE: f4:12:fa:f6:89:1a
I (651) NimBLE: 

I (656) NimBLE: GAP procedure initiated: advertise; 
I (659) NimBLE: disc_mode=2
I (663) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
I (673) NimBLE:
```

xiongweichao, 2024-01-30:
Hi @berkutta ,

In light sleep state ~230uA. May I ask if what you measured is the average value?

Thanks

berkutta, 2024-01-30:
Hi @xiongweichao 

I have measured the average current with a DMM. Is ~15mA within specification for this power_save BLE example?

xiongweichao, 2024-01-30:
Hi @berkutta ,

The value I measured was ~9.1mA(release/v5.1). The value I measured using DMM is almost the same as this.
[image]

By the way, the average power consumption of v5.1.2 will be greater than release/v5.1.

Thanks

berkutta, 2024-01-30:
Hi @xiongweichao 

Thanks for your measurement! I guess the difference between 9mA and 15mA might be due to different DMM and different IDF version.

How does a ESP32-H2 compare to this? Is it also in the range of >1mA?

xiongweichao, 2024-10-25:
Hi @berkutta 

I'm very sorry for replying to you so late.

Running the power save example on ESP32H2 chips, the average current is around 6.5 mA with the external 32.768 kHz XTAL and 8.2 mA  without using v5.1.2 IDF.

In default setting, the advertising interval is set to 30 ms, which means the chip will transmit BLE packets every 30 ms. I would recommend changing the advertising interval and Tx power according to your application, it will reduce the average current a lot. For example, if I set the advertising interval to 300 ms, the average current will reduce to 0.76 mA with with the external 32.768 kHz XTAL.

Alvin1Zhang, 2025-01-10:
Thanks for reporting, feel free to reopen.

closed 2025-01-10 by Alvin1Zhang

berkutta, 2025-03-17:
@xiongweichao @Alvin1Zhang Hello together, I have now tried the power_save example on ESP32-H2. The advertisement interval is modified like that to 300ms
```
diff --git a/examples/bluetooth/nimble/power_save/main/main.c b/examples/bluetooth/nimble/power_save/main/main.c
index 3e1a5acc28..a001123d77 100644
--- a/examples/bluetooth/nimble/power_save/main/main.c
+++ b/examples/bluetooth/nimble/power_save/main/main.c
@@ -120,8 +120,8 @@ ext_bleprph_advertise(void)
     //params.tx_power = 127;
     params.sid = 1;
 
-    params.itvl_min = BLE_GAP_ADV_FAST_INTERVAL1_MIN;
-    params.itvl_max = BLE_GAP_ADV_FAST_INTERVAL1_MIN;
+    params.itvl_min = BLE_GAP_ADV_ITVL_MS(300);
+    params.itvl_max = BLE_GAP_ADV_ITVL_MS(300);
```

This is the boot log
```
--- esp-idf-monitor 1.5.0 on /dev/ttyUSB0 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32h2-20221101
Build:Nov  1 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x408460e0,len:0x17ac
load:0x4083cad0,len:0xefc
load:0x4083efd0,len:0x2ce4
entry 0x4083cada
I (23) boot: ESP-IDF v5.3.1-dirty 2nd stage bootloader
I (24) boot: compile time Mar 17 2025 21:50:53
I (25) boot: chip revision: v0.1
I (27) boot.esp32h2: SPI Speed      : 64MHz
I (32) boot.esp32h2: SPI Mode       : DIO
I (37) boot.esp32h2: SPI Flash Size : 2MB
I (41) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=00010020 vaddr=42080020 size=1a7f4h (108532) map
I (126) esp_image: segment 1: paddr=0002a81c vaddr=40800000 size=057fch ( 22524) load
I (135) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=7f2c8h (520904) map
I (297) esp_image: segment 3: paddr=000af2f0 vaddr=408057fc size=0e1c8h ( 57800) load
I (317) esp_image: segment 4: paddr=000bd4c0 vaddr=408139d0 size=026ach (  9900) load
I (322) esp_image: segment 5: paddr=000bfb74 vaddr=50000000 size=00050h (    80) load
I (324) esp_image: segment 6: paddr=000bfbcc vaddr=50000050 size=00008h (     8) load
I (338) boot: Loaded app from partition at offset 0x10000
I (338) boot: Disabling RNG early entropy source...
I (355) cpu_start: Unicore app
I (540) cpu_start: Pro cpu start user code
I (540) cpu_start: cpu freq: 96000000 Hz
I (540) app_init: Application information:
I (543) app_init: Project name:     power_save
I (548) app_init: App version:      v5.3.1-dirty
I (553) app_init: Compile time:     Mar 17 2025 21:50:45
I (559) app_init: ELF file SHA256:  35d016950...
I (564) app_init: ESP-IDF:          v5.3.1-dirty
I (570) efuse_init: Min chip rev:     v0.0
I (574) efuse_init: Max chip rev:     v0.99 
I (579) efuse_init: Chip rev:         v0.1
I (584) heap_init: Initializing. RAM available for dynamic allocation:
I (591) heap_init: At 40818E40 len 00034540 (209 KiB): RAM
I (597) heap_init: At 4084D380 len 00002B60 (10 KiB): RAM
I (605) spi_flash: detected chip: generic
I (608) spi_flash: flash io: dio
W (612) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (626) sleep: Configure to isolate all GPIO pins in sleep state
I (632) sleep: Enable automatic switching of GPIO sleep configuration
I (650) pm: Frequency switching config: CPU_MAX: 96, APB_MAX: 32, APB_MIN: 32, Light sleep: DISABLED
I (654) coexist: coex firmware version: 4482466
I (656) main_task: Started on CPU0
I (659) main_task: Calling app_main()
I (673) pm: Frequency switching config: CPU_MAX: 96, APB_MAX: 32, APB_MIN: 32, Light sleep: ENABLED
I (674) BLE_INIT: Using main XTAL as clock source
I (684) BLE_INIT: ble controller commit:[a95ec45]
W (685) BLE_INIT: BLE modem sleep is enabled
I (689) BLE_INIT: Modem BLE MAC retention initialization
W (695) BLE_INIT: Enable light sleep, the wake up source is BLE timer
I (713) phy: phy_version: 310,2, 8fa9cb4, Jun  5 2024, 14:19:10
I (731) phy: libbtbb version: a024da6, Jun  5 2024, 14:19:22
I (732) sleep_clock: Modem Power, Clock and Reset sleep retention initialization
I (737) NimBLE_BLE_PRPH: BLE Host Task Started
I (745) NimBLE: Device Address: 
I (745) NimBLE: 48:31:b7:c0:6a:c7
I (748) NimBLE: 

I (752) NimBLE: GAP procedure initiated: extended advertise; instance=0

I (762) uart: queue free spaces: 8
I (763) main_task: Returned from app_main()
```

Current consumption (measured with nRF PPK2) is 2.4mA AVG. Any idea why I can't read the 0.76mA you've mentioned?

[image]

ChrysalisChenJ, 2025-03-18:
@berkutta Hello,

From your boot log, this line `I (674) BLE_INIT: Using main XTAL as clock source` means that you are using the main XTAL instead of the external 32.768 kHz crystal as the clock source. This will cause much higher light sleep current.
If you have the external 32kHz crystal implemented, you may need to check if these configs are set correctly:

> CONFIG_BT_LE_LP_CLK_SRC_DEFAULT=y
> CONFIG_RTC_CLK_SRC_EXT_CRYS=y

Otherwise, you can use the Internal 136 kHz RC oscillator for lower power consumption by setting:
> CONFIG_BT_LE_LP_CLK_SRC_DEFAULT=y
> CONFIG_RTC_CLK_SRC_INT_RC=y
However, this clock source does not have the clock accuracy required by the Bluetooth LE to maintain connection. If you only need advertising or scanning, then this clock source is also good for you. But overall, the external 32 kHz crystal is highly recommended.