espressif/esp-idf #13075: Problem with esp32 c3 rebooting when closing the serial port

AtechLim, 2024-01-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

my board is ESP32 C3 Supermini

After modifying Board.txt, the board reset disappeared when opening and closing the Arduino built-in serial monitor.
However, a reset occurs when close the port in the serial port monitor program (Eltima).
Both codes below appear to use HWCDC, but HWCDC does not have any code to disable DTR and RTS.

AppData\Local\Arduino15\packages\esp32\hardware\......\board.txt
```
esp32c3.serial.disableDTR=true
esp32c3.serial.disableRTS=true
``` 

```c++
//USB-CDC on boot : DISABLED
void setup() {
  pinMode(10, OUTPUT);
  USBSerial.begin();
}

void loop() {
  digitalWrite(10, LOW); 
  USBSerial.println("USB");
  delay(1000);
}

//USB-CDC on boot : ENABLED
void setup() {
  pinMode(10, OUTPUT);
  Serial.begin();
}

void loop() {
  digitalWrite(10, LOW); 
  Serial.println("SERIAL");
  delay(1000);
}
``` 

My Windows programs cannot be modified. Is there a way to disable DTR and RTS in HWCDC?
.
.
.
And I read trm v1.1

```
Table 30-2. CDC-ACM Settings with RTS and DTR
....
All these functions can also be disabled by programming various eFuses, please refer to Chapter 4 eFuse Controller (EFUSE) for more details.
```

I looked into the efuse controller, but I didn't know exactly what part to modify, so I tried it.

```
DIS_USB_SERIAL_JTAG_DOWNLOAD_MODE
DIS_DOWNLOAD_MODE
```
and also disabled Joint Download Mode by connecting GPIO 8 to GND, but still my Esp32 c3 Supermini resets when I close the serial port (Eltima). And I can't upload any more new programs...

nopnop2002, 2024-01-31:
This is the place for ESP-IDF issues.

Issues of Arduino for ESP32 is here:

https://github.com/espressif/arduino-esp32/issues

AtechLim, 2024-02-02:
Thank you, I will move it to the Arduino issue.

peter-marcisovsky, 2025-01-21:
I will close this issue, as it has been moved to Arduino related issues

closed 2025-01-21 by peter-marcisovsky