espressif/esp-idf #13077: SNTP example crashes because of possible multiple calls to 'obtain time' which sets operation mode and initializes sntp

Gaai, 2024-01-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.4-latest

### Espressif SoC revision.

any

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

any

### Power Supply used.

USB

### What is the expected behavior?

initialize_sntp() should only be called once. Therefore obtain_time() should only be called once... because it calls initialize_sntp(). If obtain_time fails for some reason. It should be possible to call it again. 

### What is the actual behavior?

The example isn't very clear on the fact that the obtain_time() function is expected to run only once. There are some issues. Because anytime obtain_time() IS called again. It calls initialize_sntp. The check if (timeinfo.tm_year < (2016 - 1900)) I think is not foolproof. Suppose unsuspecting users use it in a main loop, it can fail to set the time if the retry count is exceeded (which happens if for whatever reason the server is not reachable or wifi disconnected). Meaning the year is not set and obtain time + initialize_sntp() will get called again in next loop run. Which leads the the assert fail and crash+reboot: 
`assert failed: sntp_setoperatingmode IDF/components/lwip/lwip/src/apps/sntp/sntp.c:724 (Operating mode must not be set while SNTP client is running)`

In main:
```
    if (timeinfo.tm_year < (2016 - 1900)) {
        ESP_LOGI(TAG, "Time is not set yet. Connecting to WiFi and getting time over NTP.");
        obtain_time();
        // update 'now' variable with current time
        time(&now);
    }
```

In obtain_time() you first call initialize_sntmp() which executes:
```
    (...)
    sntp_setoperatingmode(SNTP_OPMODE_POLL);
    sntp_init();
    (...)
```
and then:
```
    const int retry_count = 10; //or 15 in latest idf 5+ example
    while (sntp_get_sync_status() == SNTP_SYNC_STATUS_RESET && ++retry < retry_count) {
        ESP_LOGI(TAG, "Waiting for system time to be set... (%d/%d)", retry, retry_count);
        vTaskDelay(2000 / portTICK_PERIOD_MS);
    }
    time(&now);
    localtime_r(&now, &timeinfo);
```
I think the initialize_sntp should be called seperately from obtain_time() or there should be some sort of check so it is run only once. Or at least the example could use a comment explaining this possible gotcha. 

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain
assert failed: sntp_setoperatingmode IDF/components/lwip/lwip/src/apps/sntp/sntp.c:724 (Operating mode must not be set while SNTP client is running)

Backtrace: 0x400822b5:0x3ffb96a0 0x4008fa9d:0x3ffb96c0 0x40094259:0x3ffb96e0 0x40114ac1:0x3ffb9810 0x400de7bc:0x3ffb9830 0x400d9d8f:0x3ffb98b0 0x401c967a:0x3ffb99c0
0x400822b5: panic_abort at C:/Espressif/frameworks/esp-idf-v4.4.4/components/esp_system/panic.c:408

0x4008fa9d: esp_system_abort at C:/Espressif/frameworks/esp-idf-v4.4.4/components/esp_system/esp_system.c:137

0x40094259: __assert_func at C:/Espressif/frameworks/esp-idf-v4.4.4/components/newlib/assert.c:85

0x40114ac1: sntp_setoperatingmode at C:/Espressif/frameworks/esp-idf-v4.4.4/components/lwip/lwip/src/apps/sntp/sntp.c:724 (discriminator 1)
```
```

Gaai, 2024-01-31:
There is already a flag available btw, so adding `if (sntp_enabled()) return;` to initialize_sntp() would solve any issues with multiple calls to obtain_time(). And the function could stay as is.

david-cermak, 2024-03-15:
Hi @Gaai 

Thanks for bringing this issue to our attention and proposing a simple fix. Adding the condition you mentioned would help, but there are other issues with `obtain_time()` function, such as it calls certain one-time operations, like `esp_event_loop_create_default()`. So I think, adding denationalization of these subsystem would be a cleaner solution. As for the SNTP, I would add `esp_sntp_stop()` just after the disconnection here:

https://github.com/espressif/esp-idf/blob/d3a0f26c84328b1dc716c261748a264bcc6b177e/examples/protocols/sntp/main/sntp_example_main.c#L152-L154

Please note that this problem is fixed already in IDF >= v5.1 (de-initializing the SNTP service):

https://github.com/espressif/esp-idf/blob/3e526ef774b6c00036cf47dfe6e4a044acc6b896/examples/protocols/sntp/main/sntp_example_main.c#L223-L226

david-cermak, 2024-03-19:
As said, the problem is fixed in `v5.1` and later, and since earlier releases are in the maintenance period and this issue is in the example only, we'd close it.
Thanks again for reporting and sorry for the slow turnaround, here's a patch against `v4.4` for the reference:
```patch
diff --git a/examples/protocols/sntp/main/sntp_example_main.c b/examples/protocols/sntp/main/sntp_example_main.c
index bd7975dafb..6a4a09cd5a 100644
--- a/examples/protocols/sntp/main/sntp_example_main.c
+++ b/examples/protocols/sntp/main/sntp_example_main.c
@@ -151,6 +151,7 @@ static void obtain_time(void)
     localtime_r(&now, &timeinfo);
 
     ESP_ERROR_CHECK( example_disconnect() );
+    esp_sntp_stop();
 }
 
 static void initialize_sntp(void)
```

closed 2024-03-19 by david-cermak

Gaai, 2024-03-19:
@david-cermak thanks for the heads up and patch advise.