espressif/esp-idf #13078: esp_mqtt_client_enqueue sometimes blocks for 10+ milliseconds

readmodifywrite, 2024-01-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-564-g8d2dbd461f

### Espressif SoC revision.

ESP32-C3

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

According to the documentation, esp_mqtt_client_enqueue should return immediately.

### What is the actual behavior?

esp_mqtt_client_enqueue is sometimes blocking for many milliseconds, sometimes more than 30 ms.  This is not a non-blocking API, although the documentation claims it is.

### Steps to reproduce.

Time esp_mqtt_client_enqueue with an active connection.  Notice how your user thread hangs.

euripedesrocha, 2024-01-31:
Hi @readmodifywrite thanks for reporting, we identified the issue already, but unfortunately it isn't an easy fix at the moment. We plan to attack the problem soon.

readmodifywrite, 2024-02-07:
Great to here that!  Let me know when you have a possible fix, I'd be happy to test it!

readmodifywrite, 2024-02-12:
Are you referring to the workaround here?
[https://github.com/espressif/esp-mqtt/issues/231](https://github.com/espressif/esp-mqtt/issues/231)

FWIW, I'm not interested in patching ESP-MQTT myself, that's a non-starter.

What if I just replace the entire queuing functionality myself?  Can I just shovel messages into a FreeRTOS queue, receive them in another thread, and then just call esp_mqtt_client_publish?

Also, if it is that simple, why would the fix be so complicated?  Am I missing something?

readmodifywrite, 2024-02-12:
I guess the difference might be that I only need to queue for a single client, but ESP-MQTT needs to handle multiple clients/queues?

euripedesrocha, 2024-02-13:
@readmodifywrite the workaround I mentioned was to use `esp_mqtt_dispatch_custom_event` to send the data to the internal event queue and later process the publish/enqueue in the registered event handler.

The challenge of this relies on the reorganization of the internal data structures to avoid the locks keeping the thread safety and without breaking compatibility. It is feasible but as I mentioned needs extra testing.

readmodifywrite, 2024-02-13:
I think for my use case, I can get away with a blocking thread using publish and a queue for the non-blocking thread.

euripedesrocha, 2024-02-14:
@readmodifywrite this will work. The solution I proposed uses the resources that are already in place and may save you the extra thread. You might need the queue for the id of messages published in case of QoS > 0.

KaeLL, 2025-03-15:
> We plan to attack the problem soon.

@euripedesrocha Define soon.

euripedesrocha, 2025-03-17:
@KaeLL sorry for the delay addressing the issue. I was unfortunately caught by other issues, and my initial attempt to solve  this issue proved to be not a good long term solution. 

The issue is still on our pipeline to fix, is this behavior causing you some trouble?

dwmw2, 2025-03-17:
Yes, this makes it fairly much unusable. AFAICT it's not just "10+ milliseconds" as the title of this issue says, it's "10+ seconds", since the thread running `esp_mqtt_task()` can hold the lock (that `esp_mqtt_client_enqueue()` is waiting for) for that long waiting for a network timeout.

And even when it doesn't block for that long, it still doesn't *send* the queued messages promptly. It doesn't *wake* the `esp_mqtt_task()` thread when there's a message to send... that thread just sends *one* message every time it wakes up, which is when its call to `esp_transport_poll_read()` times out after `MQTT_POLL_READ_TIMEOUT_MS` (default 1 second).

For ESPHome I've just given up and I invoke the synchronous methods directly from a thread of ESPHome's own (in *addition* to the thread that esp-mqtt runs, which is a bit of a waste of resources).

Brandon-Ballard25, 2026-09-07:
This is affecting us as well, we need it where the application thread can submit a message (publish or subscribe) without being blocked by network/modem activity.

We are also finding the blocking behaviour of `esp_mqtt_client_stop` problematic since that can also block for the 10 seconds.