espressif/esp-idf #13081: TG1WDT_SYS_RESET when cache disabled if I2S is enabled - CONFIG_I2S_ISR_IRAM_SAFE has no effect

tpbedford-oomdata, 2024-01-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Using ESP-IDF 5.1.2 and third-party BlueKitchen BTStack for Bluetooth audio, and ESP-ADF 2.6. Hardware is custom ESP32-based board with external I2S codec.

Calling i2s_stream_init() during init to setup the audio pipeline element - which calls i2s_driver_install() which calls i2s_start() - will later cause a crash when disabling cache for SPI flash operations such as flash partition erase.

I can perform an partition erase with crash if I call it before calling the _i2s_stream_init()_ function.
If I immediately _i2s_stop(portnum)_ after calling i2s_stream_init() then I can prevent the crash.
If I comment out _i2s_init_dma_intr()_ to prevent allocation the DMA interrupt, then it will still crash.

CONFIG_I2S_ISR_IRAM_SAFE is the only **_ISR_IN_IRAM option or **_IRAM_SAFE option enabled in my SDKCONFIG. Whilst BTStack  officially doesn't support CONFIG_I2S_ISR_IRAM_SAFE=Y it will still crash if I comment out the interrupt allocation inside _i2s_init_dma_intr()_ in ESP-IDF to prevent the BTStack callback being called.

L-KAYA, 2024-01-31:
Unfortunately, ESP-ADF 2.6 uses the legacy I2S driver, which does not support IRAM_SAFE option. And currently ADF doesn't support to select I2S driver. You may wait for the new ADF version, or you can try the new I2S driver in the IDF first (new I2S driver is supported since IDF v5.0).

tpbedford-oomdata, 2024-01-31:
Is the fix in https://github.com/espressif/esp-idf/pull/10997 not for the legacy driver? That seems to indicate that IRAM_SAFE is supported, and working effectively. The correct interrupt flags from that issue seem present in 5.1.2 so at least that specific fix doesn't seem to have regressed.

tpbedford-oomdata, 2024-01-31:
Even if the IRAM_SAFE flags aren't supported, doesn't disabling the flash cache prevent ISRs from being called if ESP_INTR_FLAG_IRAM is not set? Whilst i2s_dma_tx_callback() and rx_callback()  are both in IRAM, they are allocated without this flag in i2s_init_dma_intr() when IRAM_SAFE=n, but I still see a crash when the cache is disabled and IRAM_SAFE=n

tpbedford-oomdata, 2024-01-31:
OOOH I've been mixing legacy and new driver. Some of the function names are common between the two drivers which has VSCode confused.

For now, adding _i2s_cfg.i2s_config.intr_alloc_flags &= ~ESP_INTR_FLAG_IRAM_ to my application code before _i2s_stream_init(..)_ is sufficient to prevent the crash when cache is disabled. Likely I'll suffer some audio dropouts/stuttering but I'll manage that by constraining when flash might be accessed and thus cache disabled.

So, I shall continue using legacy driver with ESP_INTR_FLAG_IRAM cleared. I have not tested new I2S driver as this is not yet supported with ADF 2.6

closed 2025-01-13 by Alvin1Zhang