espressif/esp-idf #13082: Build fails if CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC_SIZE is set to 0x4 in IDF v5.1

vlol-re, 2024-01-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

I expect the application to be successfully built if the `CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC_SIZE` is set to 4 (or any multiple of 4 bytes). This is how it works seamlessly in IDF v4.4.

### What is the actual behavior?

The build fails during linking if the `CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC_SIZE` is set to 4.

### Steps to reproduce.

1. Create new project using IDF template `sample_project`
2. Start > idf.py menuconfig
3. Enable `Reserve RTC FAST memory for custom purposes` in Bootloader config and set `Size in bytes for custom purposes` to 4
4. Build the application using > idf.py app
5. Build fails...

### Build or installation Logs.

```plain
-- Build files have been written to: C:/_/Data/ESP/ESP32-S3/custom_reserve_rtc/build
[908/910] Linking CXX executable custom_reserve_rtc.elfFAILED: custom_reserve_rtc.elf
cmd.exe /C "cd . && C:\_\esp\v5.1\tools\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-g++.exe -mlongcalls -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32S3=0 -Wl,--Map=C:/_/Data/ESP/ESP32-S3/custom_reserve_rtc/build/custom_reserve_rtc.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32s3.peripherals.ld -T esp32s3.rom.ld -T esp32s3.rom.api.ld -T esp32s3.rom.libgcc.ld -T esp32s3.rom.newlib.ld -T esp32s3.rom.version.ld -T memory.ld -T sections.ld @CMakeFiles\custom_reserve_rtc.elf.rsp -o custom_reserve_rtc.elf  && cd ."
c:/_/esp/v5.1/tools/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: custom_reserve_rtc.elf section `.rtc_reserved' will not fit in region `rtc_reserved_seg'
c:/_/esp/v5.1/tools/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: region `rtc_reserved_seg' overflowed by 4 bytes
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

### More Information.

It seems that build works with `CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC_SIZE` set to multiple of 8 (successfully built with 0, 8, 16, 24), while it doesn't work with some multiples of 4 bytes (failed with 4, 12, 20).

o-marshmallow, 2024-02-01:
@vlol-re Thank you for reporting this issue!

I was able to reproduce it, I opened an internal merge request to fix it. In the meanwhile, here is the patch that you can apply locally:
[linker.diff.txt](https://github.com/espressif/esp-idf/files/14120258/linker.diff.txt)

To apply it, here are the commands:
```
cd esp-idf/
patch -p1 < [path_to_the_diff]
```

vlol-re, 2024-02-01:
Hi @o-marshmallow, thanks for prompt reply. The patch helps partially - the example is successfully built now. But there still seems to be a problem with the location of custom data. For testing I used an example project `bootloader_hooks` modified so that the data was passed from bootloader to the application:

in hooks.c:
```c
void bootloader_before_init(void)
{
    ESP_LOGW("HOOK", "This hook is called BEFORE bootloader initialization");

    rtc_retain_mem_t* mem = bootloader_common_get_rtc_retain_mem();
    *(uint32_t*)mem->custom = 0x12345678;
    bootloader_common_update_rtc_retain_mem(NULL, false);
}
```

in main.c:
```c
void app_main(void)
{
    rtc_retain_mem_t* mem = bootloader_common_get_rtc_retain_mem();
    ESP_LOGW("APP", "%08lX", *(uint32_t*)mem->custom);
}
```

It works as expected if `CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC_SIZE` is set to 8, but an incorrect value is displayed in the application if it is set to 4.

o-marshmallow, 2024-02-02:
@vlol-re  Right, the address must be aligned too in the bootloader. Here is a patch for all the changes, please have a try:
[linker.diff.txt](https://github.com/espressif/esp-idf/files/14134648/linker.diff.txt)

vlol-re, 2024-02-02:
Thanks @o-marshmallow , it works now.

closed 2025-05-21 by espressif-bot