espressif/esp-idf #13083: Panic on xEventGroupWaitBits

txf-, 2024-01-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2-684-g7cbee80fb9

### Espressif SoC revision.

ESP32 

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 -devkitC_V4

### Power Supply used.

USB

### What is the expected behavior?

Wait forever on xEventGroupWaitBits

```
    void ThreadMain(){
      ESP_LOGI( "testTask", "thread start");
      
      while(!initDone){
        vTaskDelay(pdMS_TO_TICKS( 2000));
        ESP_LOGI( "testTask", "initDone %d",initDone);
      }

      for(;;){
        ESP_LOGI( "testTask", "wait eventgroup");
        xEventGroupWaitBits(  testEventGroup,
                        TEST_EVENT_BIT,
                        pdFALSE,
                        pdTRUE,
                        portMAX_DELAY );
        ESP_LOGI( "testTask", "passed eventGroupWait");
      }
```

### What is the actual behavior?

It crashes with and interrupt wdt timeout.

### Steps to reproduce.

Run the attached project. It is a simple project that dynamically allocates a CRTP class that runs a thread.

Make sure the class has no task affinity (default)

[blink_debug_eventsgroup.zip](https://github.com/espressif/esp-idf/files/14111917/blink_debug_eventsgroup.zip)

### Debug Logs.

```plain
ELF file SHA256: 9274d899

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7128
load:0x40078000,len:14536
load:0x40080400,len:4
0x40080400: _init at ??:?

load:0x40080404,len:3812
entry 0x4008064c
I (29) boot: ESP-IDF v5.1.2-684-g7cbee80fb9 2nd stage bootloader
I (29) boot: compile time Jan 31 2024 12:51:10
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v1.0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=09b84h ( 39812) map
I (118) esp_image: segment 1: paddr=00019bac vaddr=3ffb0000 size=02140h (  8512) load
I (122) esp_image: segment 2: paddr=0001bcf4 vaddr=40080000 size=04324h ( 17188) load
I (131) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=142ech ( 82668) map
I (162) esp_image: segment 4: paddr=00034314 vaddr=40084324 size=080cch ( 32972) load
I (182) boot: Loaded app from partition at offset 0x10000
I (182) boot: Disabling RNG early entropy source...
I (194) cpu_start: Multicore app
I (194) cpu_start: Pro cpu up.
I (194) cpu_start: Starting app cpu, entry point is 0x40081204
0x40081204: call_start_cpu1 at /home/tiago/esp/v5.1/esp-idf/components/esp_system/port/cpu_start.c:159

I (183) cpu_start: App cpu up.
I (212) cpu_start: Pro cpu start user code
I (212) cpu_start: cpu freq: 160000000 Hz
I (212) cpu_start: Application information:
I (217) cpu_start: Project name:     blink
I (222) cpu_start: App version:      1
I (226) cpu_start: Compile time:     Jan 31 2024 12:51:03
I (232) cpu_start: ELF file SHA256:  9274d899...
I (238) cpu_start: ESP-IDF:          v5.1.2-684-g7cbee80fb9
I (244) cpu_start: Min chip rev:     v0.0
I (249) cpu_start: Max chip rev:     v3.99 
I (254) cpu_start: Chip rev:         v1.0
I (258) heap_init: Initializing. RAM available for dynamic allocation:
I (266) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (272) heap_init: At 3FFB29E0 len 0002D620 (181 KiB): DRAM
I (278) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (284) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (291) heap_init: At 4008C3F0 len 00013C10 (79 KiB): IRAM
I (298) spi_flash: detected chip: generic
I (301) spi_flash: flash io: dio
W (305) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (319) app_start: Starting scheduler on CPU0
I (324) app_start: Starting scheduler on CPU1
I (324) main_task: Started on CPU0
I (334) main_task: Calling app_main()
I (334) testTask: thread start
I (334) testTask: wait eventgroup
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x4008636c  PS      : 0x00060934  A0      : 0x80086b11  A1      : 0x3ffb69d0  
0x4008636c: spinlock_acquire at /home/tiago/esp/v5.1/esp-idf/components/esp_hw_support/include/spinlock.h:112
 (inlined by) xPortEnterCriticalTimeout at /home/tiago/esp/v5.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:501

A2      : 0x3ffae710  A3      : 0xffffffff  A4      : 0x0000cdcd  A5      : 0x00060923  
A6      : 0x034abea9  A7      : 0xb33fffff  A8      : 0x80086378  A9      : 0x3ffb69b0  
A10     : 0x00000000  A11     : 0x734518e7  A12     : 0x00000000  A13     : 0x0000000c  
A14     : 0x3ffb6afc  A15     : 0x3ffaffd4  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd  
0x400014fd: strlen in ROM
0x4000150d: strlen in ROM

Backtrace: 0x40086369:0x3ffb69d0 0x40086b0e:0x3ffb6a00 0x400d57a7:0x3ffb6a30 0x400d57c5:0x3ffb6a50 0x400861dd:0x3ffb6a70
0x40086369: spinlock_acquire at /home/tiago/esp/v5.1/esp-idf/components/esp_hw_support/include/spinlock.h:117
 (inlined by) xPortEnterCriticalTimeout at /home/tiago/esp/v5.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:501
0x40086b0e: vPortEnterCritical at /home/tiago/esp/v5.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:575
 (inlined by) xEventGroupWaitBits at /home/tiago/esp/v5.1/esp-idf/components/freertos/FreeRTOS-Kernel/event_groups.c:353
0x400d57a7: TestTask::ThreadMain() at /home/tiago/Documents/debugAWSC6/blink_debug_eventsgroup/main/testTask.hpp:66 (discriminator 13)
0x400d57c5: ThreadTaskDynamic<TestTask>::ThreadMain() at /home/tiago/Documents/debugAWSC6/blink_debug_eventsgroup/main/taskClass.hpp:77
 (inlined by) ThreadTaskDynamic<TestTask>::task(void*) at /home/tiago/Documents/debugAWSC6/blink_debug_eventsgroup/main/taskClass.hpp:65
0x400861dd: vPortTaskWrapper at /home/tiago/esp/v5.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

Core  1 register dump:
PC      : 0x40084796  PS      : 0x00060734  A0      : 0x800d20d6  A1      : 0x3ffb58b0  
0x40084796: esp_cpu_wait_for_intr at /home/tiago/esp/v5.1/esp-idf/components/esp_hw_support/cpu.c:121

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffaf8c0  A5      : 0x3ffaf8a0  
A6      : 0x40084548  A7      : 0x00000001  A8      : 0x800d7546  A9      : 0x3ffb5870  
0x40084548: ipc_task at /home/tiago/esp/v5.1/esp-idf/components/esp_system/esp_ipc.c:48

A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ffaf8a0  A13     : 0x3ffaf870  
A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x40084793:0x3ffb58b0 0x400d20d3:0x3ffb58d0 0x40087302:0x3ffb58f0 0x400861dd:0x3ffb5910
0x40084793: xt_utils_wait_for_intr at /home/tiago/esp/v5.1/esp-idf/components/xtensa/include/xt_utils.h:81
 (inlined by) esp_cpu_wait_for_intr at /home/tiago/esp/v5.1/esp-idf/components/esp_hw_support/cpu.c:111
0x400d20d3: esp_vApplicationIdleHook at /home/tiago/esp/v5.1/esp-idf/components/esp_system/freertos_hooks.c:59
0x40087302: prvIdleTask at /home/tiago/esp/v5.1/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4439 (discriminator 1)
0x400861dd: vPortTaskWrapper at /home/tiago/esp/v5.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: 9274d899

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
```

### More Information.

If forced to run on core 1 then it does not crash. However, even if running on core 0 this task should be blocked and then yield to the core 0 idle task.  

This problem presented itself on IDF 5.1 and 5.2, I tested on 5.0 and it worked as expected

This code also was tested on an ESP32C6 (on 5.2) and it also presented itself.

o-marshmallow, 2024-02-01:
Hello @txf- ,

The issue is not with FreeRTOS but with your class implementation. When the CPU enters the `ThreadMain()` function, your class members have not been initialized yet, so `initDone` has not been set to `false` and `testEventGroup` has not been reset to `NULL`.

This is because when you initialize the parent class in your constructor, with `ThreadTaskDynamic{TASK_STACK_SIZE, _priority,_name,_coreNum}`, the latter will create and invoke the new Task directly. However, the constructor has not fnished executing! It still need to initialize the other member of the class, but it's already too late, the `ThreadMain()` has been executed. You can then guess that the behavior is undefined since `testEventGroup` is not NULL and not initialzied and `initDone` is considered `true`.

To sum up, the compiler will compile your constructor as:
```
1. Call parent constructor
2. Set initDone to 0
3. Set testEventGroup to NULL
```

But what actually happens is:
```
1. Call parent constructor
    |
    +--- 1.1 Invoke xTaskCreatePinnedToCore
          |
          +--- 1.2 Execute ThreadMain, initDone/testEventGroup not initialized
2. Set initDone to 0
3. Set testEventGroup to NULL
```

You have several ways of fixing this:
* Initialize explicitly your members in your constructor **before** calling the parent constructor, you'll need -no-reorder option
* Add a tiny delay in the `ThreadMain()` to make sure your variables are initialized, with a `vTaskDelay` of course (main task has to be rescheduled to finished executing the constructor)
* Change you way you verify if your task is ready by adding the ready member directly in your parent class (**cleanest** solution in my opinion)

txf-, 2024-02-01:
Thank you for the explanation. Makes perfect sense.

A curious artifact is that I have been using this pattern from since the 3.x days, and this issue has never occurred before.  The thread would always spin in the delay loop waiting for initdone.

Typically I would always pin the Tasks to core 1, presumably there was an inherent scheduling delay such that the members would have time to be initialized. Or maybe initdone was by happenstance already 0, before being initialized.

I have since decided not to have the init done flag, but instead have the thread suspend itself on launch and then resume the thread with in begin()

closed 2024-02-01 by txf-

o-marshmallow, 2024-02-01:
@txf- By pinning the task on the core 1, the core 0 (main task) can return from the constructor and initialize the members while the core 1 is exiting the Idle task and scheduling the new task.

The solution you took is cleaner :+1:

txf-, 2024-02-01:
> @txf- By pinning the task on the core 1, the core 0 (main task) can return from the constructor and initialize the members while the core 1 is exiting the Idle task and scheduling the new task.
> 
> The solution you took is cleaner 👍

Right, of course. Thank you again for the explanation.