espressif/esp-idf #13093: cc1.exe: error: too many filenames given; type 'cc1.exe --help' for usage cc1.exe: fatal error: @C:\WINDOWS\TEMP\ccDFenPn: Invalid argument

Yan19960220, 2024-02-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-1724-ga5b261f699

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

build with these three bin files 

### What is the actual behavior?

build error: 
......
[241/944] Building C object esp-idf/esp_common/CMakeFiles/__idf_esp_common.dir/src/esp_err_to_name.c.obj
FAILED: esp-idf/esp_common/CMakeFiles/__idf_esp_common.dir/src/esp_err_to_name.c.obj

.......

cc1.exe: error: too many filenames given; type 'cc1.exe --help' for usage
cc1.exe: fatal error: @C:\WINDOWS\TEMP\ccDFenPn: Invalid argument

### Steps to reproduce.

1. Step
2. Step
3. Step
...
Just follow your steps and when build my project, it shows this error

### Build or installation Logs.

```plain
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[241/944] Building C object esp-idf/esp_common/CMakeFiles/__idf_esp_common.dir/src/esp_err_to_name.c.obj
FAILED: esp-idf/esp_common/CMakeFiles/__idf_esp_common.dir/src/esp_err_to_name.c.obj
ccache C:\Espressif\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin\xtensa-esp32-elf-gcc.exe -DESP_PLATFORM -DIDF_VER=\"v5.3-dev-1724-ga5b261f699-dirty\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -Iconfig -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_common/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/newlib/platform_include -IC:/Users/Yan/Desktop/esp/esp-idf/components/freertos/config/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/freertos/config/include/freertos -IC:/Users/Yan/Desktop/esp/esp-idf/components/freertos/config/xtensa/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos -IC:/Users/Yan/Desktop/esp/esp-idf/components/freertos/esp_additions/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_hw_support/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_hw_support/include/soc -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_hw_support/include/soc/esp32 -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_hw_support/dma/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_hw_support/port/esp32/. -IC:/Users/Yan/Desktop/esp/esp-idf/components/heap/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/log/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/soc/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/soc/esp32 -IC:/Users/Yan/Desktop/esp/esp-idf/components/soc/esp32/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/hal/platform_port/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/hal/esp32/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/hal/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_rom/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_rom/include/esp32 -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_rom/esp32 -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_system/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_system/port/soc -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_system/port/include/private -IC:/Users/Yan/Desktop/esp/esp-idf/components/xtensa/esp32/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/xtensa/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/xtensa/deprecated_include -IC:/Users/Yan/Desktop/esp/esp-idf/components/lwip/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/lwip/include/apps -IC:/Users/Yan/Desktop/esp/esp-idf/components/lwip/include/apps/sntp -IC:/Users/Yan/Desktop/esp/esp-idf/components/lwip/lwip/src/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/lwip/port/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/lwip/port/freertos/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/lwip/port/esp32xx/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/lwip/port/esp32xx/include/arch -IC:/Users/Yan/Desktop/esp/esp-idf/components/lwip/port/esp32xx/include/sys -IC:/Users/Yan/Desktop/esp/esp-idf/components/driver/deprecated -IC:/Users/Yan/Desktop/esp/esp-idf/components/driver/i2c/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/driver/touch_sensor/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/driver/twai/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/driver/touch_sensor/esp32/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_pm/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_ringbuf/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_driver_gpio/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_driver_pcnt/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_driver_gptimer/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_driver_spi/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_driver_mcpwm/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_driver_ana_cmpr/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_driver_i2s/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_driver_sdmmc/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/sdmmc/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_driver_sdspi/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_driver_sdio/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_driver_dac/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_driver_rmt/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_driver_tsens/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_driver_sdm/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_driver_i2c/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_driver_uart/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/vfs/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_driver_ledc/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_driver_parlio/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_driver_usb_serial_jtag/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_adc/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_adc/interface -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_adc/esp32/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_adc/deprecated/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/efuse/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/efuse/esp32/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_http_client/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_event/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_http_server/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/http_parser -IC:/Users/Yan/Desktop/esp/esp-idf/components/bootloader_support/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/bootloader_support/bootloader_flash/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/nvs_flash/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/spi_flash/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_partition/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_wifi/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_wifi/wifi_apps/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_phy/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_phy/esp32/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_netif/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/app_update/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_app_format/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/esp_bootloader_format/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/wpa_supplicant/include -IC:/Users/Yan/Desktop/esp/esp-idf/components/wpa_supplicant/port/ [... cut ...]
-- Adding linker script C:/Users/Yan/Desktop/esp/esp-idf/components/bootloader/subproject/main/ld/esp32/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp_app_format esp_bootloader_format esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table soc spi_flash xtensa
-- Component paths: C:/Users/Yan/Desktop/esp/esp-idf/components/bootloader C:/Users/Yan/Desktop/esp/esp-idf/components/bootloader_support C:/Users/Yan/Desktop/esp/esp-idf/components/e
```

dobairoland, 2024-02-02:
Please try if running `ccache -C` helps or not.

If the issue persists then please try `idf.py --no-ccache build` instead.

iXiaoxideYan, 2024-02-13:
> Please try if running `ccache -C` helps or not.
> 
> If the issue persists then please try `idf.py --no-ccache build` instead.

It works! Thanks!

Alvin1Zhang, 2024-02-27:
Thanks for sharing the updates, and glad to hear that it works, feel free to reopen. Thanks.

closed 2024-02-27 by Alvin1Zhang

summerfind, 2024-05-13:
yes, it works for cc1.exe --help;
how about the failed: hello_world.elf---ninja: build stopped: subcommand failed?
thanks