espressif/esp-idf #13098: USB CDC access aka Bootloader) not accessible in Raspian

Ajb2k3, 2024-02-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

On an OSX based system I can access both the Upper level JTAG firmware level and the lower level CDC bootloader levels but on windows and Raspian I can only access the upperlevel JTAG firmware level resulting in some firmware blocking the JTAG port rending the device un reprogrammable. 

Is this a bug in the driver/ESP-IDF as I can't for devices into th bootloader load on windows and Raspian using the onboard BOOT button.

Dazza0, 2024-02-03:
In raspbian, does the device show up if you call `lsusb`?

Ajb2k3, 2024-02-03:
> In raspbian, does the device show up if you call `lsusb`?

Only on the upper level.
WCHModem which is the CDC/Bootloader port for these devices does not appear.

Ajb2k3, 2024-02-03:
Sorry forgot to say its the Latest Raspbian Bookworm but not sure what ESP-IDF version