espressif/esp-idf #13100: allocating file with size

djdisodo, 2024-02-03:
### Is your feature request related to a problem?

sometimes i have to create a file of certain size (for fseek)

it's not a major problem on conventional OSs since posix's ftruncate supports extending, iirc windows has similar feature too
`If fildes refers to a regular file, the ftruncate() function shall cause the size of the file to be truncated to length. If the size of the file previously exceeded length, the extra data shall no longer be available to reads on the file. If the file previously was smaller than this size, ftruncate() shall increase the size of the file. If the file size is increased, the extended area shall appear as if it were zero-filled.`

linux even supports fallocate() which creates a file without initialization

even if filesystem doesn't have a way to do this effectively it should just fill zero to ensure consistency

i found fatfs will reject if i try to extend the file using ftruncate

i don't know if it's also the case for other filesystems

### Describe the solution you'd like.

support extending file with ftruncate for fatfs and other filesystems

better if we can get fallocate support

### Describe alternatives you've considered.

extending file using such apis will be faster than writing zeros to the file
also preventing fragmentations

### Additional context.

[https://pubs.opengroup.org/onlinepubs/9699919799/functions/ftruncate.html](https://pubs.opengroup.org/onlinepubs/9699919799/functions/ftruncate.html)

[https://man7.org/linux/man-pages/man2/fallocate.2.html](https://man7.org/linux/man-pages/man2/fallocate.2.html)

igrr, 2024-02-03:
> i found fatfs will reject if i try to extend the file using ftruncate

Could you please provide more information about the issue you faced: version of ESP-IDF, the code to reproduce the issue, log output (if any), sdkconfig. Thanks in advance.

djdisodo, 2024-02-03:
> > i found fatfs will reject if i try to extend the file using ftruncate
> 
> Could you please provide more information about the issue you faced: version of ESP-IDF, the code to reproduce the issue, log output (if any), sdkconfig. Thanks in advance.

i was using esp-rs with esp-idf version of 5.1.2
what i tried to do was use rust `File::set_len` to extend the file which uses truncate internally

i got error
[`PermissionDenied`](https://doc.rust-lang.org/std/io/enum.ErrorKind.html#variant.PermissionDenied)
with message "Not owner"
and didn't saw anything other than the `set_len` function returning an error

maybe this is just because of my logging settings

the message doesn't make any sense but i think it's just that there's something wrong handling esp-idf error

i tried to track down where did this message came from but couldn't find it

however it seems straightforward that fatfs will not truncate to extend the file length, instead return Permission error
https://github.com/espressif/esp-idf/blob/7cbee80fb9fe8310cefccb320e2c1189c13619c2/components/fatfs/vfs/vfs_fat.c#L1036-L1042

adokitkat, 2024-02-06:
Hello. You should be able to use `lseek` to expand a file, although it will not behave exactly like a POSIX lseek - it uses `f_lseek` from FATFS, read more here: http://elm-chan.org/fsw/ff/doc/lseek.html. 

```
If an offset beyond the file size is specified in write mode, the file size is expanded to the specified offset in this function.
Be careful of the difference in this behavior between FatFs and POSIX.
The file data in the expanded part is undefined, because no data is written to the file in this process.
This is suitable to pre-allocate a data area to the file quickly for fast write operation.
```

djdisodo, 2024-02-08:
@adokitkat it seems to extend the file length and a lot faster than writing zeros

closed 2024-03-06 by espressif-bot