espressif/esp-idf #13101: Allow the ESP32-C6 to access lp_timer values

txf-, 2024-02-03:
### Is your feature request related to a problem?

I need to measure time deltas on the lp_core.

### Describe the solution you'd like.

I believe the lp_timer is accessible, and there are functions already written for the lp_core to access this value, but they're not currently exposed in the headers.

### Describe alternatives you've considered.

something like this presumably works:

uint64_t ulp_lp_core_get_cycles_to_us()
{
    uint64_t cycles = RV_READ_CSR(mcycleh)<<32 +  RV_READ_CSR(mcycle);
    return cycles * (LP_CORE_CPU_FREQUENCY_HZ / 1000000);
}

But surely there is no reason why one couldn't use a proper timer, I don't know how inaccurate this method is. 

### Additional context.

I noticed that in ulp_lp_core_lp_timer_shared.c

there is this:

```
uint64_t IRAM_ATTR lp_timer_hal_get_cycle_count(void)
{
    /* Shifts current count to buffer 0, and the value in buffer 0 to buffer 1 */
    lp_timer_ll_counter_snapshot(lp_timer_context.dev);

    uint32_t lo = lp_timer_ll_get_counter_value_low(lp_timer_context.dev, 0);
    uint32_t hi = lp_timer_ll_get_counter_value_high(lp_timer_context.dev, 0);

    lp_timer_counter_value_t result = {
        .lo = lo,
        .hi = hi
    };
    return result.val;
}
```
but it is currently static and is not found in the header file. Is there a reason this function is not exposed?

o-marshmallow, 2024-02-05:
Hello @txf- ,

Which target and version of IDF are you using?

txf-, 2024-02-05:
> Hello @txf- ,
> 
> Which target and version of IDF are you using?

Target?

I'm currently using v5.2-beta2-190-g4ecc978bd6, but it is also like that in master.

o-marshmallow, 2024-02-05:
@txf- I mean which ESP32 SoC are you using? ESP32-S3, ESP32-C6, ...?

In fact, not all targets have the LP timer that you saw defined in the `ulp_lp_core_lp_timer_shared.c` file

txf-, 2024-02-05:
@o-marshmallow I'm using the C6, sorry. I thought you were asking for something else given the title.

o-marshmallow, 2024-02-06:
@txf- My bad, indeed the title contains it, how could I miss it :smile: 

I opened an internal MR to make this LP timer function public.

A bit more info about the low power cores:
* The FSM core can be found on the ESP32, ESP32-S2 and ESP32-S3
* The RISC-V ULP can be found on the ESP32-S2 and the ESP32-S3
* The LP Core is found on the ESP32-C6 and ESP32-P4
* The ESP32-C2, C3 and H2 don't have a low power core

closed 2024-02-13 by espressif-bot