espressif/esp-idf #13102: Package compatibility

jiapei100, 2024-02-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.2

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Dependent on 
- **requests_file-2.0.0** but **NOT** **requests_file-1.5.1**
- **urllib-2.2.0** but **NOT** **urllib3-1.26.18**

### What is the actual behavior?

**ALWAYS** install **requests_file-1.5.1** and **urllib3-1.26.18** by default.

### Steps to reproduce.

Just hit `./install.sh`

### Build or installation Logs.

```plain
➜  esp-idf git:(master) ✗ ./install.sh
Detecting the Python interpreter
Checking "python3" ...
Python 3.10.12
"python3" has been detected
Checking Python compatibility
Installing ESP-IDF tools
Updating ~/.espressif/idf-env.json
Selected targets are: esp32s3, esp32c3, esp32s2, esp32c2, esp32c6, esp32h2, esp32p4, esp32c5, esp32
Current system platform: linux-amd64
Installing tools: xtensa-esp-elf-gdb, riscv32-esp-elf-gdb, xtensa-esp-elf, riscv32-esp-elf, esp32ulp-elf, openocd-esp32, esp-rom-elfs
Skipping xtensa-esp-elf-gdb@12.1_20231023 (already installed)
Skipping riscv32-esp-elf-gdb@12.1_20231023 (already installed)
Skipping xtensa-esp-elf@esp-13.2.0_20240109 (already installed)
Skipping riscv32-esp-elf@esp-13.2.0_20240109 (already installed)
Skipping esp32ulp-elf@2.38_20240113 (already installed)
Skipping openocd-esp32@v0.12.0-esp32-20230921 (already installed)
Skipping esp-rom-elfs@20230320 (already installed)
Installing Python environment and packages
Python 3.10.12
pip 24.0 from ~/.local/lib/python3.10/site-packages/pip (python 3.10)
Upgrading pip and setuptools...
Requirement already satisfied: pip in ~/.local/lib/python3.10/site-packages (24.0)
Requirement already satisfied: setuptools in ~/.local/lib/python3.10/site-packages (69.0.3)
Skipping the download of ~/.espressif/espidf.constraints.v5.3.txt because it was downloaded recently.
Installing Python packages
 Constraint file: ~/.espressif/espidf.constraints.v5.3.txt
 Requirement files:
  - ....../esp-idf/tools/requirements/requirements.core.txt
Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
Ignoring importlib_metadata: markers 'python_version < "3.8"' don't match your environment
[... 56 lines cut ...]
Requirement already satisfied: mdurl~=0.1 in ~/.local/lib/python3.10/site-packages (from markdown-it-py>=2.2.0->rich->esp-idf-size->-r ....../esp-idf/tools/requirements/requirements.core.txt (line 18)) (0.1.2)
Using cached idf_component_manager-1.4.2-py2.py3-none-any.whl (140 kB)
Using cached esp_idf_monitor-1.4.0-py3-none-any.whl (44 kB)
Using cached esp_idf_size-1.1.0-py3-none-any.whl (44 kB)
Installing collected packages: urllib3, esp-idf-panic-decoder, esp-idf-kconfig, requests-file, esptool, esp-idf-size, esp-idf-nvs-partition-gen, esp-coredump, idf-component-manager, esp-idf-monitor
  Attempting uninstall: urllib3
    Found existing installation: urllib3 2.2.0
    Not uninstalling urllib3 at ~/.local/lib/python3.10/site-packages, outside environment ~/.espressif/python_env/idf5.3_py3.10_env
    Can't uninstall 'urllib3'. No files were found to uninstall.
  Attempting uninstall: requests-file
    Found existing installation: requests-file 2.0.0
    Not uninstalling requests-file at ~/.local/lib/python3.10/site-packages, outside environment ~/.espressif/python_env/idf5.3_py3.10_env
    Can't uninstall 'requests-file'. No files were found to uninstall.
Successfully installed esp-coredump-1.10.0 esp-idf-kconfig-2.1.0 esp-idf-monitor-1.4.0 esp-idf-nvs-partition-gen-0.1.1 esp-idf-panic-decoder-1.0.1 esp-idf-size-1.1.0 esptool-4.8.dev1 idf-component-manager-1.4.2 requests-file-1.5.1 urllib3-1.26.18
All done! You can now run:

  . ./export.sh

➜  esp-idf git:(master) ✗
```

### More Information.

- OS: Ubuntu 22.04.3
- Python: Python 3.10.12

jiapei100, 2024-02-04:
Some more info:

```bash
➜  esp-idf git:(master) ✗ idf.py --version
The following Python requirements are not satisfied:
Requirement 'urllib3<2' was not met. Installed version: 2.2.0
Requirement 'requests-file<2' was not met. Installed version: 2.0.0
To install the missing packages, please run "install.sh"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: ~/.espressif/python_env/idf5.3_py3.10_env
    Python interpreter used: ~/.espressif/python_env/idf5.3_py3.10_env/bin/python
Constraint file: ~/.espressif/espidf.constraints.v5.3.txt
Requirement files:
 - ....../esp-idf/tools/requirements/requirements.core.txt
Python being checked: ~/.espressif/python_env/idf5.3_py3.10_env/bin/python

ESP-IDF v5.3-dev-1724-ga5b261f699-dirty
```

```bash
➜  site-packages pwd
~/.espressif/python_env/idf5.3_py3.10_env/lib/python3.10/site-packages
➜  site-packages ls
esp_coredump                     esp_idf_monitor-1.4.0.dist-info            esp_idf_size-1.1.0.dist-info           idf_component_tools            requests_file.py
esp_coredump-1.10.0.dist-info    esp_idf_nvs_partition_gen                  espsecure                              kconfcheck                     test
espefuse                         esp_idf_nvs_partition_gen-0.1.1.dist-info  esptool                                kconfgen                       tests
esp_idf_kconfig                  esp_idf_panic_decoder                      esptool-4.8.dev1.dist-info             kconfserver                    urllib3
esp_idf_kconfig-2.1.0.dist-info  esp_idf_panic_decoder-1.0.1.dist-info      idf_component_manager                  __pycache__                    urllib3-1.26.18.dist-info
esp_idf_monitor                  esp_idf_size                               idf_component_manager-1.4.2.dist-info  requests_file-1.5.1.dist-info
➜  site-packages 
```

jiapei100, 2024-02-05:
Sorry that I have to re-open this issue, because I now met the same issue as [https://github.com/espressif/esp-idf/issues/10823](https://github.com/espressif/esp-idf/issues/10823), and my **error** message is:

```bash
[2/2] idf (5.3.0)
-- Project sdkconfig file ....../esp-who/examples/human_face_detection/lcd/sdkconfig
~/.espressif/python_env/idf5.3_py3.10_env/bin/python: No module named kconfgen.__main__; 'kconfgen' is a package and cannot be directly executed
CMake Error at /opt/Espressif/esp-idf/tools/cmake/kconfig.cmake:209 (message):
  Failed to run kconfgen
  (~/.espressif/python_env/idf5.3_py3.10_env/bin/python;-m;kconfgen;--list-separator=semicolon;--kconfig;/opt/Espressif/esp-idf/Kconfig;--sdkconfig-rename;/opt/Espressif/esp-idf/sdkconfig.rename;--config;....../esp-who/examples/human_face_detection/lcd/sdkconfig;--defaults;....../esp-who/examples/human_face_detection/lcd/sdkconfig.defaults;--defaults;....../esp-who/examples/human_face_detection/lcd/sdkconfig.defaults.esp32s3;--env-file;....../esp-who/examples/human_face_detection/lcd/build/config.env).
  Error 1
Call Stack (most recent call first):
  /opt/Espressif/esp-idf/tools/cmake/build.cmake:615 (__kconfig_generate_config)
  /opt/Espressif/esp-idf/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!
```

dobairoland, 2024-02-05:
I don't remember any similar issue when packages installed in `~/.local/` would block installing different versions in the virtual environment. Did you create `~/.espressif/python_env/idf5.3_py3.10_env` on your own? Could you please remove it and re-create it by re-running `install.sh`?

-----
This is just an irrelevant side note. Your ESP-IDF version is not 5.1.2 but the the upcoming v5.3 (master branch).

dobairoland, 2024-02-05:
Have you set any of the ESP-IDF environment variables permanently in your system (e.g. `IDF_PYTHON_ENV_PATH`)? The output of `install.sh` doesn't seem right to me. It looks as if the Python environment would be redirected to `~/.local/lib/python3.10`. After that, the export script switches to the ESP-IDF environment (based on your `idf.py --version`) correctly. That is why you getting these issues. The install script updates a different environment.

jiapei100, 2024-02-05:
My bad... I got 2 versions of **kconfgen** installed, one under `~/.local`, the other under `~/.espressif` .

Thank you

closed 2024-02-05 by jiapei100