espressif/esp-idf #13104: lwip_recv_tcp Invalid Copylen Error during File Upload in Async HTTP Web Server

Luca96, 2024-02-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-beta1

### Espressif SoC revision.

ESP32-D0WD-V3 (revision v3.1)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-WROOM-32

### Power Supply used.

USB

### What is the expected behavior?

The expected behavior is to handle file upload requests by the server, asynchronously. The `httpd_req_recv` function should receive a chunk of bytes without failing, such that the received bytes can be then written on an SD card. The whole thing handled by FreeRTOS's tasks. 

### What is the actual behavior?

When the task is invoked by the FreeRTOS sheduler, the file upload handler manages to receive a few chunks but then crashes suddenly, apparently at random points during the upload. The error I get is: 

> assert failed: **lwip_recv_tcp** /IDF/components/lwip/lwip/src/api/sockets.c:1001 **(invalid copylen)**

and is triggered by the `httpd_req_recv` function.

### Steps to reproduce.

Basically my code merges the [async_handlers](https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/async_handlers) example with the [file_serving](https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/file_serving) one. The example code is:

```c
#include <stdio.h>
#include <string.h>
#include <sys/param.h>
#include <errno.h>

#include "freertos/FreeRTOS.h"
#include "freertos/semphr.h"
#include <esp_wifi.h>
#include <esp_event.h>
#include <esp_log.h>
#include <esp_system.h>
#include <nvs_flash.h>
#include <sys/param.h>
#include "nvs_flash.h"
#include "esp_netif.h"
#include "esp_eth.h"
#include "esp_tls_crypto.h"
#include <esp_http_server.h>

[... 530 lines cut ...]
0x40110c33: lwip_recvfrom at C:/Espressif/frameworks/esp-idf-v5.2-beta1/components/lwip/lwip/src/api/sockets.c:1218
0x40110d02: lwip_recv at C:/Espressif/frameworks/esp-idf-v5.2-beta1/components/lwip/lwip/src/api/sockets.c:1283
0x400e045c: recv at C:/Espressif/frameworks/esp-idf-v5.2-beta1/components/lwip/include/lwip/sockets.h:38
 (inlined by) httpd_default_recv at C:/Espressif/frameworks/esp-idf-v5.2-beta1/components/esp_http_server/src/httpd_txrx.c:660
0x400dfdba: httpd_recv_with_opt at C:/Espressif/frameworks/esp-idf-v5.2-beta1/components/esp_http_server/src/httpd_txrx.c:121
0x400dfdea: httpd_recv at C:/Espressif/frameworks/esp-idf-v5.2-beta1/components/esp_http_server/src/httpd_txrx.c:142
0x400e035c: httpd_req_recv at C:/Espressif/frameworks/esp-idf-v5.2-beta1/components/esp_http_server/src/httpd_txrx.c:542
0x400d683a: upload_post_handler at C:/Espressif/frameworks/esp-idf-v5.2-beta1/examples/protocols/http_server/async_handlers/main/main.c:286
0x400d655b: async_req_worker_task at C:/Espressif/frameworks/esp-idf-v5.2-beta1/examples/protocols/http_server/async_handlers/main/main.c:181
0x40089b4d: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf-v5.2-beta1/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134
```

### More Information.

From the `menuconfig`:
* `CONFIG_EXAMPLE_MAX_ASYNC_REQUESTS=2`.

My `CMakeLists.txt` is:
```CMake
idf_component_register(SRCS "main.c" "wifi_ap.c"
                       EMBED_FILES "page.html" 
                       INCLUDE_DIRS "." "include")
```

I also say that the whole code works fine without the async part, i.e., without using the FreeRTOS's tasks.

hmalpani, 2024-02-28:
Hello @Luca96 
It seems like the issue is because the stack size is small for the async handler task:
`#define ASYNC_WORKER_TASK_STACK_SIZE    2048`
Can you try increasing the stack size and test it again. Please let me know if it helps.

Thanks!

Luca96, 2024-03-01:
Hi @hmalpani , thanks for the help but increasing the task stack size still not works: I increased it to 16k but now I get a different error:
```
I (111177) file_server: invoking /upload/file.png
I (111187) file_server: Receiving file : /filr.png...
I (111187) file_server: Remaining size : 16221
I (111207) file_server: Remaining size : 11517
I (111227) file_server: Remaining size : 6973
I (111237) file_server: Remaining size : 4480

W (116227) httpd_txrx: httpd_sock_err: error in recv : 11
W (116227) httpd_txrx: httpd_sock_err: error in recv : 128
E (116297) file_server: File reception failed! Error type: -1 (Socket is not connected)
W (116307) httpd_txrx: httpd_resp_send_err: error calling setsockopt : 9
W (116317) httpd_txrx: httpd_sock_err: error in send : 9
```
which is about the underlying socket being disconnected.

I've also noticed that the async task fails, then it gets resubmitted but fails again. Moreover, I tried to upload another file, getting the usual error:
```
assert failed: lwip_recv_tcp /IDF/components/lwip/lwip/src/api/sockets.c:1001 (invalid copylen)
```

Should I change the HTTP server config too? Like the `core_id`, `task_priority` and `stack_size`? Thanks

kayzen9, 2024-03-27:
@chipweinberger maybe you know . ..

chipweinberger, 2024-03-27:
not sure of a reason. 

seems lwip related

AxelLin, 2024-06-05:
Try enable LWIP_DEBUG to check the values of p->tot_len, recv_left, recvd and copylen before the assertion.

david-cermak, 2024-10-10:
I was able to reproduce the reported problems and confirm that there's a race condition in LWIP.

Thanks @Luca96 for sharing all these details and steps to reproduce it. Apologies for not handling this sooner, looks like a pretty nasty bug in lwip (socket owned cached pbuf not protected):

https://github.com/espressif/esp-lwip/blob/a45be9e438f6cf9c54ec150581819c3b95d5af6b/src/api/sockets.c#L1001

It was failing on v5.2, but working on the latest IDF, yet probably just because of the way worker threads are run in the latest http server. 

I'm not sure how to fix the issue correctly at the moment, but will come up with some workaround at least in the following days.

david-cermak, 2024-10-21:
Hi @Luca96 

Could you please try to cherry-pick 4a7f37105659383e154d38d89fbc89cc26e29d9a and test again?

I was chasing this bug in lwip and indeed the socket internal `lastdata.pbuf` was not thread-safe (had a concrete idea already on fixing it), but the fix would (only) prevent the code from crashing. it wouldn't work, since the http-server is not supposed to handle the same request (=socket) by multiple workers, IMO

Luca96, 2024-11-04:
Hello @david-cermak, thank you very much for your help! 

I finally tested the changes you suggested, and I can say that the code works without any crashes. 

In addition, I tried to upload two files at the same time from two different sessions (one on PC, and another on my smartphone) and that seemed to work too! From the logs, I can see that two different sockets were used to handle the file transfer, indeed I have to allocate enough workers.

david-cermak, 2024-11-05:
@Luca96 Thank you for the confirmation! I'll backport the fix to the release branches.

david-cermak, 2024-11-14:
Closing as https://github.com/espressif/esp-idf/commit/4a7f37105659383e154d38d89fbc89cc26e29d9a has been backported to v5.2 in https://github.com/espressif/esp-idf/commit/83216d64e4e1002f424a753e72c873e3ab0a7c82

Also documented the lwIP limitation in https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/lwip.html#lwip-limitations

PS: holding off the lwip fix for now, as it's really impractical and would kill performance for most usecases. I'll think if it'd be possible to add an assert with a meaningful error message.