espressif/esp-idf #13105: ESP IDF isn't beginner friendly

4e576rt8uh9ij9okp, 2024-02-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm a fullstack webdev and most of my projects are private and deployed on the web.

I used Arduinos before and it was pretty easy to get started, I bought my first ESP32 S3 and I have issues following tutorials.

I'm on VS Code + ESP-IDF extension
I wanted to try the usb_host_lib example and I have connected the breakout USB Port to the GPIO pins and the issue starts with the first project step.

`idf.py menuconfig`
resulting with this output here
```
PS C:\Users\user\Documents\CODING\ESP32-S3\tinyusb> idf.py menuconfig
idf.py : The term 'idf.py' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the spelling of the name, or if a path was included, verify that the path is correct and try again.
At line:1 char:1
+ idf.py menuconfig
+ ~~~~~~
    + CategoryInfo          : ObjectNotFound: (idf.py:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException
```

I know that something isn't setup but even tho I looked at the installation of ESP-IDF and I installed it successfully but sadly I'm unable to follow any example as `idf.py` doesn't exist.

I don't blame anyone but my own lack of knowledge, I'm writing this from a POV of a total beginner who only used the Arduino IDE and did some small stuff with Arduinos.

TzionCastillo1, 2024-02-04:
Have you sourced esp-idf? See step 4 of this link: https://docs.espressif.com/projects/esp-idf/en/v5.1.2/esp32/get-started/linux-macos-setup.html

nopnop2002, 2024-02-05:
>I'm on VS Code + ESP-IDF extension

This is my personal opinion, but I don't think this choice is for beginners.

When problems arise, they become more complex to solve.

pedrominatel, 2024-02-05:
Hi @4e576rt8uh9ij9okp,
Make sure you have your environment variables exported on your current terminal.
You can run the `export.bat` located in the ESP-IDF install folder.

[Using the command prompt](https://docs.espressif.com/projects/esp-idf/en/v5.1.2/esp32s3/get-started/windows-setup.html#using-the-command-prompt)

If you are using the ESP-IDF Extension on VS Code, you can try:

- Press F1 to open the command search.
- Type "Open ESP-IDF Terminal"

On this new terminal, you can use `idf.py`.

dobairoland, 2024-02-05:
Hi @4e576rt8uh9ij9okp. Can you share your way of getting to installing these tools? Like how did you proceed? Did you have vscode then you found the vscode plugin and then you found some guide (which?), or you did it in a completely different order? Describing this process could be very useful for us.

ESP-IDF is not ESP-IDF + Vscode plugin from the setup point of view. For ESP-IDF + Vscode plugin here is the guide to follow: https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/install.md. Did you follow this one? I've noticed that you try to run `idf.py menuconfig` which is not recommended in this setup. The Vscode plugin has a GUI for this: SDK configuration editor. I guess that either the setup was unsuccessful or you've opened a regular PowerShell window where `idf.py` is not visible. The tools are not visible globally, only in Vscode, Eclipse or terminal ESP-IDF environment.

For ESP-IDF on Windows the installer from https://dl.espressif.cn/dl/esp-idf/ will set up everything. You will get ESP-IDF shortcuts on your desktop where everything is ready to use. This way you can use ESP-IDF without Vscode.

4e576rt8uh9ij9okp, 2024-02-05:
> Hi @4e576rt8uh9ij9okp. Can you share your way of getting to installing these tools? Like how did you proceed? Did you have vscode then you found the vscode plugin and then you found some guide [...]

Ok so this is what I did, I'm a webdev and I had already VS Code installed same as NodeJS and NPM.
Then I've installed the VS Code extension named "ESP-IDF"

Then it opened a GUI and I clicked Express Install... it started throwing errors that things are missing.
So I have installed the ESP-IDF (thing) from this website here:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html
and following this doc:
https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/install.md

This is also the doc that was shown to me when I googled and there was no mention of installing (ESP-IDF for Windows). Some could say, "if we didn't mention it, there is no need" but then we go back to the issue where the plugin throws errors when pressing "install".

After installing the ESP-IDF via windows installer I was able to complete the VS Code ESP-IDF installer GUI.

After that I have added a Path to my Path variable:
`C:\Espressif\frameworks\esp-idf-v5.1.2\tools`

I've opened a Example from "Show Examples" and pick the Hello World one.
Clicked "Select Serial Port" being the COM8
Clicked "Build" everything was build
Now I pressed "Flash" and it throwed an error that COM8 access denied.

So I used the ESP Flash tool, selected the bootloader, firmware etc and successfully flashed, sadly it said that the CPU ID is wrong so it didn't work...

I tried the Arduino IDE way before with a simple `Serial.print("hello");` and it worked without any issues and logged it into the Serial Monitor.

The Board is ALWAYS COM8 as it's the only Type-C USB port on my Mainboard.

dobairoland, 2024-02-05:
Thanks, this is very helpful feedback to us!

-----

Did you managed to solve your issue? Since you used the Windows installer, there should be on the Desktop and/or in the Start menu several ESP-IDF shortcuts (one for each terminal program). You should be able to run `idf.py` commands from it.

dobairoland, 2024-02-05:
> "ESP-IDF 5.1 CMD"
> "ESP-IDF 5.1 PowerShell"

Both should work whether you prefer Windows command line or Power Shell.

italocjs, 2024-02-07:
While i do agree that esp-idf if by no means beginner friendly, its possible to get it running and there are some really good videos on youtube with detailed process.  I think you issue is that you are trying to call those commands from windows terminal, and they have not been registered to the path (nor need).  you can use the vscode embedded terminal to call those commands.

I had some similar issues trying to install espidf with the offline installer and then using on vscode, using the express installer all worked fine. to be honest if you dont need to change stuff on menuconfig, you are better off using platformio extension + arduino core, its so much easier and stuff just works, as opposite to idf which requires some extra work (ex even adding an lib is harder on idf, as it requires you to edit the cmake files. on platformio it just works "magically" (there is no magic, just code running under the hood).

4e576rt8uh9ij9okp, 2024-02-09:
> While i do agree that esp-idf if by no means beginner friendly, its possible to get it running and there are some really good videos on youtube with detailed process. I think you issue is that you [...]

I understand, but is there a overhead when comparing VScode + eps idf vs Arduino with ESP32 lib?
I'm not talking about the "under the hood" magic, but the output of the compiled code, is the size and performance the same?

nopnop2002, 2024-02-09:
>the output of the compiled code, is the size and performance the same?

When you build the ESP32 sketch in the Arduino environment, this code will be executed as the main function.
UNICORE's ESP32 generates a 50ms wait every 2000ms.

https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/main.cpp

nopnop2002, 2024-02-09:
When you build the ESP32 sketch in the Arduino environment, this code will be executed as the main function.
UNICORE's ESP32 generates a 50ms wait every 2000ms.

When you build the ESP32 sketch in the Arduino environment, this code will be executed as the main function.
UNICORE's ESP32 generates a 50ms wait every 2000ms.

https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/main.cpp

4e576rt8uh9ij9okp, 2024-02-10:
> > is there a overhead when comparing VScode + eps idf vs Arduino with ESP32 lib?
> 
> When you build the ESP32 sketch in the Arduino environment, this code will be executed as the main function. UNICORE's ESP32 incurs a 50ms wait every 2000ms.
> 
> By default, 8192 bytes of STACK are consumed regardless of the processing content of Loop().
> 
> https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/main.cpp

So from what you say, Arduino env has no delays, unless the delay function is being used?
I'm asking because I have a decent goal, but I'm making baby steps.
I have used Arduinos to like understand it...
Now I thought using the VSCODE + esp idf is good as the next step...
The end goal is programming myself MCUs on a gaming mouse, so I have a way to go to achieve that.

italocjs, 2024-02-11:
> I understand, but is there a overhead when comparing VScode + eps idf vs Arduino with ESP32 lib? I'm not talking about the "under the hood" magic, but the output of the compiled code, is the size [...]

Compiled size will be a bit bigger and you will have a little less flexibility (no easy access to menuconfig unless you build your own arduino lib).  but this usually is fine, most stuff just works. if needed You can still use most ESP-IDF functions directly.  never used a single core esp32 so never seen the 50ms delay,  on dual core it works without any delay and i can get microsseconds sampling on interrupt easily.

italocjs, 2024-02-11:
> So from what you say, Arduino env has no delays, unless the delay function is being used? I'm asking because I have a decent goal, but I'm making baby steps. I have used Arduinos to like understand [...]

If you absolutely must you can kill the loop by calling vtaskdelete(null).  i've been doing this for a while since loop was not useful for me and all other stuff ran in other smaller tasks / callbacks.

jgardona, 2024-05-07:
Agreed. Esp idf is very confusing due to the lack of beginner documentation. No examples, no step by step in developmnt, only in setup they have. It gets hard to start, and the example codes are very "dirty", with lots of ifdef macros. Guys need to do documentations for  beginners.

b55585wy, 2024-08-18:
Agreed. The documentation is hard to understand for beginners. Some configuration steps are not necessary and should be included in embedded software. It lacks basic collision detection capabilities.

diplfranzhoepfinger, 2024-10-15:
> > I'm on VS Code + ESP-IDF extension
> 
> This is my personal opinion, but I don't think this choice is for beginners.
> 
> When problems arise, they become more complex to solve.

I share this opinion. Start with Espressif IDE. I think much more work goes into Espressif IDE than in vscode. So things work easier there.

dobairoland, 2025-03-10:
I think this is a useful feedback thread but nothing we should keep it open. I'm closing it for inactivity.