espressif/esp-idf #13106: Multiple partition tables

AchimPieters, 2024-02-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello All,

I'm trying to add support for multiple ESP32 modules to a project, but when compiling, it does not select the right partition.csv file. This is what I did:

```
/led
  /main
    - main.c
  /partitions
    - partitions_esp32.csv
    - partitions_esp32s2.csv
    - partitions_esp32c3.csv
    - ...
  - CMakeLists.txt
```
Added this to main.c
```
#ifdef CONFIG_IDF_TARGET_ESP32
  // ESP32-specific code
#elif defined(CONFIG_IDF_TARGET_ESP32S2)
  // ESP32-S2-specific code
#elif defined(CONFIG_IDF_TARGET_ESP32C3)
  // ESP32-C3-specific code
#elif defined(CONFIG_IDF_TARGET_ESP32S3)
  // ESP32-S3-specific code
#elif defined(CONFIG_IDF_TARGET_ESP32C2)
  // ESP32-C2-specific code
#elif defined(CONFIG_IDF_TARGET_ESP32C6)
  // ESP32-C6-specific code
#elif defined(CONFIG_IDF_TARGET_ESP32H2)
  // ESP32-H2-specific code
#elif defined(CONFIG_IDF_TARGET_ESP32P4)
  // ESP32-P4-specific code
#elif defined(CONFIG_IDF_TARGET_ESP32C5)
  // ESP32-C5-specific code
#endif
```
Added this to CMakeLists.txt
```
message(STATUS "Configuring for ESP32 variant: ${CONFIG_IDF_TARGET}")
if(CONFIG_IDF_TARGET STREQUAL "esp32")
  message(STATUS "Using partition table for ESP32")
  set(PARTITION_TABLE_DEFAULT "partitions/partitions_esp32.csv" CACHE FILE "Default partition table file")
elseif(CONFIG_IDF_TARGET STREQUAL "esp32s2")
  message(STATUS "Using partition table for ESP32S2")
  set(PARTITION_TABLE_DEFAULT "partitions/partitions_esp32s2.csv" CACHE FILE "Default partition table file")
elseif(CONFIG_IDF_TARGET STREQUAL "esp32c3")
  message(STATUS "Using partition table for ESP32C3")
  set(PARTITION_TABLE_DEFAULT "partitions/partitions_esp32c3.csv" CACHE FILE "Default partition table file")
elseif(CONFIG_IDF_TARGET STREQUAL "esp32s3")
  message(STATUS "Using partition table for ESP32S3")
  set(PARTITION_TABLE_DEFAULT "partitions/partitions_esp32s3.csv" CACHE FILE "Default partition table file")
elseif(CONFIG_IDF_TARGET STREQUAL "esp32c2")
  message(STATUS "Using partition table for ESP32C2")
  set(PARTITION_TABLE_DEFAULT "partitions/partitions_esp32c2.csv" CACHE FILE "Default partition table file")
elseif(CONFIG_IDF_TARGET STREQUAL "esp32c6")
  message(STATUS "Using partition table for ESP32C6")
  set(PARTITION_TABLE_DEFAULT "partitions/partitions_esp32c6.csv" CACHE FILE "Default partition table file")
elseif(CONFIG_IDF_TARGET STREQUAL "esp32h2")
  message(STATUS "Using partition table for ESP32H2")
  set(PARTITION_TABLE_DEFAULT "partitions/partitions_esp32h2.csv" CACHE FILE "Default partition table file")
elseif(CONFIG_IDF_TARGET STREQUAL "esp32p4")
  message(STATUS "Using partition table for ESP32P4")
  set(PARTITION_TABLE_DEFAULT "partitions/partitions_esp32p4.csv" CACHE FILE "Default partition table file")
elseif(CONFIG_IDF_TARGET STREQUAL "esp32c5")
  message(STATUS "Using partition table for ESP32C5")
  set(PARTITION_TABLE_DEFAULT "partitions/partitions_esp32c5.csv" CACHE FILE "Default partition table file")
endif()
message(STATUS "Partition table file: ${PARTITION_TABLE_DEFAULT}")
```
In `menuconfig`, didn't change anything left default.

Steps:

1. idf.py set-target esp32c3
2. idf.py build
3. Logfile: [log.txt](https://github.com/espressif/esp-idf/files/14156901/log.txt)

nopnop2002, 2024-02-05:
Specific partition settings must be written in sdkconfig.defaults.

This will help.

https://github.com/espressif/esp-idf/tree/master/examples/storage/spiffs

AchimPieters, 2024-02-05:
@nopnop2002 Thank you for your reply.  But with the solution you provided, it doesn't fix my problem, because I define only one of the *.csv files and need to change the `sdkconfig.defaults` every time I want to compile for another ESP module?

nopnop2002, 2024-02-05:
sdkconfig.default can be created for each target.

https://github.com/espressif/esp-idf/tree/master/examples/wifi/iperf

AchimPieters, 2024-02-05:
@nopnop2002 Thank you that did the trick!

closed 2024-02-05 by AchimPieters