espressif/esp-idf #13107: Restoring pre-5.0 component require behavior (all components publicly requirable)

amaschas, 2024-02-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

So from a developer usability standpoint I feel like the features detailed here is a bit of a solution search for a problem:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.0/build-system.html#specify-component-requirements-explicitly

I'm sure there was a good reason for it, but my builds always worked previously, and the documentation seems to indicate that there is a way to enable the previous dependency behavior, but I've not been able to determine what that might be. Specifically the phrase "these components are no longer added as public requirements by default" would indicate that there is a non-default option. Given the amount of interdependency between my components, and the amount of work needed to update them to work with version above 5.0, I would be extremely interested in such an option, or at the very least a more detailed elaboration as to why this additional work is required on my part, given that (as I mentioned) this appeared to work fine prior to the "fix".

dobairoland, 2024-02-05:
I'm not aware that any such backward compatible option would exists. I hope someone will correct me if I'm wrong.

If I remember it correctly then the motivation behind this breaking change was to pull in less component dependencies, simplify the dependency tree (find dependency issues easily) and to reduce build time.

o-marshmallow, 2024-02-12:
Hello @amaschas ,

I confirm what @dobairoland said, the goal of these changes was to reduce the build time by only compiling the required dependencies. There is no way to restore the previous behavior via a configuration option or anything like that.

I understand the confusion with the sentence `In ESP-IDF v5.0, this behavior is fixed and these components are no longer added as public requirements by default` from the documentation. In fact, it means that the components were added to the build system by default, so the `by default` was referring to the fact that CMake would pull them in the dependency list and you could override this behavior and only keep a few of them. It does **not** mean that this behavior is the default one now.

Anyway, if you cannot modify the `REQUIRES` to pull in only the required components you need, one workaround is to pick one of your component (or make a new one) that I am calling `A`, that would depend on all the ones specified in https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.0/build-system.html#specify-component-requirements-explicitly
Then all your other components would need to rely on `A`, with a `REQUIRES`. As such, `A` would pull in all these IDF dependencies when compiled.

o-marshmallow, 2024-04-10:
Closing this issue

closed 2024-04-10 by o-marshmallow