espressif/esp-idf #13109: how to control "Allow Potentially Insecure Options" in websocket example?

PES-SARJU, 2024-02-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi! 
     We are using WebSocket example from SDK idf-v4.4, we are able to connect with the WebSocket server with Server side certificate , while "Allow Potentially Insecure Options" is un-checked in SDK configuration. which make sense. but on the other hand we want to connect without sever certificate as well, but in that case "Allow Potentially Insecure Options" should we checked. And in this case also it is able to connect with the server. we gave name to both the cases, profile 1 and profile 2 respectively. which can be opted from our local we page at run time. Now the question is how we can select or unselect this option "Allow Potentially Insecure Options" in order to connect with WebSocket Server? In both the cases.

Thanks in advance,
sarju

gabsuren, 2024-04-17:
@PES-SARJU 
if you don’t want to verify the server, you can set the confirmation below
 ```
 CONFIG_ESP_TLS_INSECURE=y
 CONFIG_ESP_TLS_SKIP_SERVER_CERT_VERIFY=y
 CONFIG_WS_OVER_TLS_SKIP_COMMON_NAME_CHECK=y
 ```
 
 Regards,
 Suren

gabsuren, 2024-05-29:
@PES-SARJU closing the issue. Please feel free if you have other questions or face any issues.

closed 2024-05-29 by gabsuren