espressif/esp-idf #13110: [ModBus] Allow sending request with custom function code

mattiaVDM, 2024-02-05:
### Is your feature request related to a problem?

I need to communicate with a custom device that accept customized ModBus request to update it's firmware. I have an issue while trying to send ModBus requests with custom function code, that does not belong to the standard ModBus protocol. 

`
esp_err_t mbc_master_send_request(mb_param_request_t* request, void* data_ptr);
`

I'm trying to use the send request function to send a custom request with function code: (0x0D), but the function does not accept custom function codes. 

### Describe the solution you'd like.

I would like to be able to send requests with custom function codes, that does not belong to the standard ModBus documentiation.

alisitsyn, 2024-02-06:
Hi @mattiaVDM,

You can send the custom request using the [mbc_master_send_request() function](https://docs.espressif.com/projects/esp-modbus/en/latest/esp32/master_api_overview.html#master-communication). However, it is not possible for now to support all possible custom commands and you need to write your custom request function and response handler for your function code.
I have a plan to extend the interface API to add custom functions handlers later. This will be done in extension of [v2.0.0-beta](https://github.com/espressif/esp-modbus/discussions/45)

For example the steps to do this:
 
1. Move your esp-modbus component from `managed_components` into `components` folder of your project root then you can modify the component.  

2. The request function for write multiple holding register is [here](https://github.com/espressif/esp-modbus/blob/master/freemodbus/modbus/functions/mbfuncholding_m.c#L108)

3. The response check function for write holding register is [here](https://github.com/espressif/esp-modbus/blob/master/freemodbus/modbus/functions/mbfuncholding_m.c#L132). Similar handling function you need to write for your  custom function code in a separate source file located in [command handlers folder](https://github.com/espressif/esp-modbus/tree/master/freemodbus/modbus/functions). 

4. Once this is done please add your custom handler function into table [here](https://github.com/espressif/esp-modbus/blob/master/freemodbus/modbus/mb_m.c#L122).

Once this is completed you can use the [mbc_master_send_request() function](https://docs.espressif.com/projects/esp-modbus/en/latest/esp32/master_api_overview.html#master-communication) to send your custom request.
Let me know if you have any questions.

alisitsyn, 2024-05-24:
[the esp-mobus documentation is updated to address this issue](https://docs.espressif.com/projects/esp-modbus/en/latest/esp32/master_api_overview.html?highlight=supported%20communication%20options#master-communication-options). The issue is closed.

alisitsyn, 2025-04-07:
@mattiaVDM, the support for new commands is merged with commit [cbe96056f2bcb44d4580355126b7255af332c3ba](https://github.com/espressif/esp-modbus/commit/cbe96056f2bcb44d4580355126b7255af332c3ba). This allows to add or override the custom handler for a function or get existing one for each Slave or Master object. If you have time, please take a look and verify this functionality. Thanks.

The issue will be close as completed.

closed 2025-04-07 by alisitsyn